Koenkk/zigbee2mqtt #5666: Eurotronic Spirit - Error: Cluster 'hvacThermostat' has no attribute '16387'

Robert904, 2021-01-10:
First, thanks a lot for this great software, I love it !
I'm afraid my skills do not allow me to track down this issue myself...

## What happened
After replacing my CC2531 with a CC2652RB and while repairing my eurotronic spirit radiator valves, I get the following error, during the painring and "onZigbeeEvent" and "onMQTTConnected" callbacks :

`
Zigbee2MQTT:error 2021-01-10 10:36:27: Failed to call 'Bridge' 'onMQTTConnected' (Error: Cluster 'hvacThermostat' has no attribute '16387'                                                             │
    at Object.getAttribute (/local/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl/utils.js:84:19)                                                                                                   │
    at _configuredReportings.map (/local/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:87:36)                                                                             │
    at Array.map (<anonymous>)                                                                                                                                                                         │
    at Endpoint.get configuredReportings [as configuredReportings] (/local/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:83:43)                                           │
    at devices.zigbee.getDevices.map (/local/zigbee2mqtt/lib/extension/bridge.js:512:60)                                                                                                               │
    at Array.map (<anonymous>)                                                                                                                                                                         │
    at Bridge.publishDevices (/local/zigbee2mqtt/lib/extension/bridge.js:484:50)                                                                                                                       │
    at Bridge.onMQTTConnected (/local/zigbee2mqtt/lib/extension/bridge.js:47:20))                                                                                                                      │
`

While I can still use the devices normally, I can't receive anymore the list of devices in the zigbee2mqtt web interface ("Loading devices takes too long time.")

## What did you expect to happen
No error messages, and list of devices in the web interface

## How to reproduce it (minimal and precise)
Not sure, I had no such error with my CC2531 adapter.  I also tried deleting and repairing the devices with the same outcome

## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC2652RB
Adapter firmware version: 20201026

Koenkk, 2021-01-10:
Fixed!
Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-10 by Koenkk

Robert904, 2021-01-10:
Working perfectly, many thanks!