Koenkk/zigbee2mqtt #5681: GL-S-003Z more broken

Tiitan, 2021-01-10:
## What happened
My GL-S-003Z is now more broken than ever with the 1.17.0 update, it already didn't have white  because my fix was not merged (https://github.com/Koenkk/zigbee2mqtt/pull/3828).
But now, no messages can be sent at all with the error:
Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'constructor' of undefined at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.17.0/lib/extension/publish.js:183:52) at Controller.callExtensionMethod (/zigbee2mqtt-1.17.0/lib/controller.js:381:44) at runMicrotasks () at processTicksAndRejections (internal/process/task_queues.js:97:5)).
I removed it from the network and paired again twice, but still crashing

## How to reproduce it (minimal and precise)
pair a GL-S-003Z, send any message

## Debug info
Zigbee2MQTT version: 1.17.0.1
Adapter hardware: CC2531
Adapter firmware version: early 2020

Koenkk, 2021-01-10:
#3823 was never merged because I didn't get an answer to: https://github.com/Koenkk/zigbee2mqtt/pull/3828#issuecomment-651884412

Can you provide the debug log of this error?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Tiitan, 2021-01-11:
I'm affraid there is not much more info in debug, but pairing info might help:
modelId:"GL-S-003Z", manufacturerName:"GLEDOPTO", powerSource:1, zclVersion:1, appVersion:1, stackVersion:2, hwVersion:1, dateCode:"20131206, swBuildId:"2.0.0"
full log for 0x00124b001f248627 device from pairing to error:
https://gist.github.com/Tiitan/84659b34f3f5ecb5df8ff441b5da631a

Koenkk, 2021-01-11:
Can you provide the data/database.db entry of this device?

Tiitan, 2021-01-11:
https://gist.github.com/Tiitan/a2d51891619a1f58d57871b75ad98d74

Koenkk, 2021-01-11:
Should be fixe now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-18 by github-actions[bot]