Koenkk/zigbee2mqtt #5687: Since 1.17.0 part of the installation is DEAD (?)

njordan77, 2021-01-10:
## What happened
After upgrading to 1.17.0 (from 1.16.2) part of my installation (mostly Tradfri switches/remotes) some of them do not react anymore. Very strange, spent already 2 nights, did downgrade, did not help. tried to include new Tradfri's does recognize and add to the YAML, but after that, no log entries anymore. BUT some of the Tradfri's do enter the logs, strange as well, its not a specific model, completely mixed up, some Tradfri Remotes work, some do not...same for the 0/1 Switch.

What can i do next. i'm out of ideas and it takes lots of my spare time to try-and-error. Any hint or idea would be highly appreciated. It feels like it all started with 1.17.0 upgrade.....but downgrade does not fix it as well...so maybe even not.

Adapter hardware: CC2531, 
Adapter firmware version: the June 2019...the classic i guess.

jonastep, 2021-01-12:
I have the same issue. 
Tradfi repeaters are unstable and some devices after being re-paired (because they disappeared) are not exposing the identities. 
 Config: 
Adapter hardware: CC2531.
Adapter firmware version: the June 2019.

njordan77, 2021-01-13:
@Koenkk could this be any issue or our installations a problem? Is there a way to clean start and copy back the DB + config. or will this still keep the problem due to these files....most likely the DB. I tried reverting back to 16.2...but the issue is kept.

Koenkk, 2021-01-13:
@njordan77 can you provide the debug log when pressing the switch?

njordan77, 2021-01-13:
sure. What i already did was to change the info setting to debug, and look at the systemctl -f logs but nothing is shown.
Is there anything else you mean with more engineering level of debug, otherwise i have to tell you that all these switches (that do not work) are not showing anything in these logs.

njordan77, 2021-01-13:
I've logged a complete shutdown and restart of the service.
https://1drv.ms/t/s!AqVcwrl1pLMG1ktbBsAYYB6yjwBn?e=uwWeSJ    (onedrive)

What looks strange to me and this never was like this before in the startup logs...is the long long list of items shown. Any idea where this comes from, if this is part of the problem? and how to fix it? Regards

Koenkk, 2021-01-13:
@njordan77 what if you force remove a tradfri remote and pair it again? https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestdeviceremove

njordan77, 2021-01-13:
I already tried this for the Tradfri_Remote05 before....only error messages. What i totally do not understand, that this singe Remote05 device pairing could kill part of my 18 device installation. Can this happen?

Jan 13 22:32:37 SmartPi npm[28743]: Zigbee2MQTT:info  2021-01-13 22:32:37: MQTT publish: topic '/z2m/bridge/response/dev                                                                                                                     ice/remove', payload '{"data":{},"error":"Failed to remove device 'Tradfri_Switch05' (Error: AREQ - ZDO - mgmtLeaveRsp a                                                                                                                     fter 10000ms)","status":"error"}'

But is this big big list normal? I do not remember to have ever seen this "messy" huge log before....or is this due to debug logging?
...
...
...
...
58d00044bf190":{"friendly_name":"XiaomiAqara_WaterLeak02"},"0x0017880106735eea":{"friendly_name":"PhilipsHue_Remote01","retain":false},"0x0017880106e84949":{"friendly_name":"PhilipsHue_Remote02","retain":false},"0x14b457fffe3e0198":{"friendly_name":"Tradfri_Remote04"},"0x14b457fffe507412":{"friendly_name":"Tradfri_Remote02","retain":false},"0x14b457fffe5d44dc":{"friendly_name":"Tradfri_Switch05"},"0x14b457fffe672935":{"friendly_name":"Tradfri_Switch04"},"0x14b457fffe686b52":{"friendly_name":"Tradfri_Motion01","retain":false},"0x14b457fffe75f847":{"friendly_name":"Tradfri_Switch03","retain":
...
...
...

sjorge, 2021-01-13:
@Koenkk this looks similar to the single event I saw with the test firmware for the NWK_TABLE_FULL issue a while ago

njordan77, 2021-01-13:
@sjorge OK thanks for joining. What does it mean? How to fix? 18 devices from my memory is far from the limit a single stick can handle. Cheers Norbert

sjorge, 2021-01-13:
It mostly some extra info that hopefully will ring a bell for Koen, I had a similar every device got kicked event myself.

At the time we (koen and me) assumed it might have been due to running a highly experimental firmware, so my solution of switching back to the old firmware won’t help here.

Koenkk, 2021-01-15:
`{"data":{},"error":"Failed to remove device 'Tradfri_Switch05' (Error: AREQ - ZDO - mgmtLeaveRsp a fter 10000ms)","status":"error"}'` this happens when the device is sleeping. You can force remove it instead. Please read https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestdeviceremove carefully.

njordan77, 2021-01-16:
force removing even several devices did not improve anything. From your experience @Koenkk do these cc2531 sticks sometimes mess up (when close to the paired clients limit). Is there a way to hard rest the stick and join all devices again and see if it works again afterwards. I already bought a conbee2 stick as the ZZH! is sold out right now....this should long term make things better....

Koenkk, 2021-01-17:
@njordan77 you can do that by reflashing the cc2531

closed 2021-02-24 by github-actions[bot]