Koenkk/zigbee2mqtt #5690: CR Smart Home TS0202 motion sensor false event issue

Cportelli, 2021-01-11:
## What happened
As per manual it says that when continuous motion is detected, it won't send an event until no motion is detected again for 1 minute. After that it will restore no motion status and will go into "working mode".

I can see both events, the first motion detected and the other no motion that triggers after 1 minute of last motion detected. However the false event implemented in [fix 2818](https://github.com/Koenkk/zigbee2mqtt/issues/2818)(triggered after 90) is throwing of the automation if the room is occupied for more then 90 seconds.

## What did you expect to happen
I cannot replicate the issue reported on https://github.com/Koenkk/zigbee2mqtt/issues/2818 where the "status is always occupied in Home Assistant even after leaving the device stay in an empty room for a day" and I'm getting occupied false after 60s on non motion as expected.

## How to reproduce it (minimal and precise)
1. trigger the motion sensor. The occupancy: true  will be reported accordingly.
2. keep movement in front of sensor for >90 seconds. the false event is triggered as [per issue fix 2818](https://github.com/Koenkk/zigbee2mqtt/issues/2818). This will trigger any lights to turn off incorrectly in automations.
3. stop the movement and after 60s the sensor will report correctly occupancy: false as expected.

## Debug info
_Zigbee2MQTT:info 2021-01-10 22:24:22: MQTT publish: topic 'zigbee2mqtt/Motion PIR Sensor', payload '{"battery":100,"battery_low":false,"linkquality":84,"occupancy":true,"tamper":false}' ---Trigger motion sensor
Zigbee2MQTT:info 2021-01-10 22:25:52: MQTT publish: topic 'zigbee2mqtt/Motion PIR Sensor', payload '{"battery":100,"battery_low":false,"linkquality":84,"occupancy":false,"tamper":false}' ---FALSE EVENT 90s after first event
Zigbee2MQTT:info 2021-01-10 22:27:10: MQTT publish: topic 'zigbee2mqtt/Motion PIR Sensor', payload '{"battery":100,"battery_low":false,"linkquality":78,"occupancy":false,"tamper":false}' ---Actual 1min after no motion

Zigbee2MQTT version: 1.17.0
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608

Koenkk, 2021-01-11:
- Can you try updating the definition in devices.js to

```js
{
    zigbeeModel: ['TS0202'],
    model: 'TS0202',
    vendor: 'CR Smart Home',
    description: 'Motion sensor',
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery()],
},
```

Does this fixes it?

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

- Also, is your device really a CR smart home one or did you buy a different whitelabel (form e.g. aliExpress), if yes please provide a link to it
- Can you share the data/database.db entry of this device?

70MM13, 2021-01-11:
Hello,
sorry for jumping in but got triggered by the title as I recently bought this:  https://nl.aliexpress.com/item/1005001895670295.html 
I can't find a brand anywhere, it identifies as a CR smart home TS0202 but  the behavior is different than stated above. I've been waving at it for a couple of minutes and then it keeps triggering occupancy:true about every 30 seconds. When movement stops, occupancy:false comes after 5 seconds.

this is the database.db entry:
```
{"id":9,"type":"EndDevice","ieeeAddr":"0xec1bbdfffee22e6b","nwkAddr":43095,"manufId":4098,"manufName":"_TYZB01_ef5xlc9q","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,1280,3],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TYZB01_ef5xlc9q","powerSource":3,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0x00212effff069cbb","zoneState":1,"currentZoneSensitivityLevel":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1610384978217}
```

Cportelli, 2021-01-11:
> * Can you try updating the definition in devices.js to
>
> ```js
> {
>     zigbeeModel: ['TS0202'],
>     model: 'TS0202',
>     vendor: 'CR Smart Home',
>     description: 'Motion sensor',
>     fromZigbee: [...]

First of all thanks for the help i'm new to this and really enjoying learning more!
This is indeed a aliexpress product (https://www.aliexpress.com/item/4001135645873.html?spm=a2g0s.9042311.0.0.5e474c4dQmJPtA)

if can you guide me how to get that data for you il be happy to share.

Cportelli, 2021-01-11:
So I modified the devices.js as suggested and it works! thanks a lot! 💯 
Also found where the database is but I cant decrypt it :(

Koenkk, 2021-01-12:
@Cportelli you can open the data/database.db with any text editor

Cportelli, 2021-01-13:
@Koenkk 
{"id":2,"type":"EndDevice","ieeeAddr":"0xbc33acfffe3314de","nwkAddr":11772,"manufId":0,"manufName":"_TYZB01_jytabjkb","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,61185],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TYZB01_jytabjkb","powerSource":3,"zclVersion":3,"appVersion":68}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001949ae29","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1610547908585}

Koenkk, 2021-01-13:
Thanks, fixed

Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-13 by Koenkk

groenmarsmannetje, 2021-02-09:
See my comment in #2818