Koenkk/zigbee2mqtt #5691: Setting "filtered_attributes" is ignored

m0wlheld, 2021-01-11:
## What happened

Added a [device](https://shop.codm.de/bundles/zigbee/router/25/zigbee-cc2530-cc2592-long-range-router-bundle-9cm/2.5dbi?c=7) that acts as router only, based on CC2530 + CC2592. It's supported using [ptvo.info](https://www.zigbee2mqtt.io/devices/ptvo.switch.html) firmware.

As the SoC supports inputs/output (which are not connected), tried to filter the matching attributes to prevent them appearing in Home Assistant. However, it seems like filtered_attributes is ignored. It might be worth mentioning, that I added the filtered_attributes setting after the device have been initially reported to Home Assistant.

See https://pastebin.com/KgfcG8G5 for the current MQTT publish payload.

## What did you expect to happen

Sensors and switches "I1" to "I8" and others should not be reported via MQTT to Home Assistant. 

## How to reproduce it (minimal and precise)

Join an appropriate device to your network. Adjust configuration.yaml afterwards to filter unwanted attributes:

```
devices:
  '0x0000000000001cb9':
    friendly_name: 'routing device'
    filtered_attributes:
      - action
      - humidity
      - l1
      - l2
      - l3
      - l4
      - l5
      - l6
      - l7
      - l8
      - pressure
    legacy: false

```
## Debug info

Zigbee2MQTT version: Zigbee2MQTT version 1.17.0 (commit #07cdc9d, "latest" Docker image)
Adapter hardware: Raspbee II (reported as type "ConBee2/RaspBee2")
Adapter firmware version: 0x26690700

Koenkk, 2021-01-11:
`filtered_attributes` will filter attributes from the payload, to prevent switches from being discovered use

```yaml
devices:
  '0x0000000000001cb9':
    friendly_name: 'routing device'
    homeassistant:
      switch_l1: null
      switch_l2: null
      # etc...
```

closed 2021-01-11 by Koenkk

m0wlheld, 2021-01-12:
Sorry, but doing so causes an error in ZigbeeMQTT (see below). Should setting entities (switches and sensors) to null be enough or is it required to use filtered_attributes additionally?

```
Zigbee2MQTT:error 2021-01-12 11:54:38: Failed to call 'HomeAssistant' 'onMQTTConnected' (TypeError: Cannot convert undefined or null to object
    at Function.keys (<anonymous>)
    at add (/app/lib/extension/homeassistant.js:757:28)
    at /app/lib/extension/homeassistant.js:775:21
    at Array.forEach (<anonymous>)
    at HomeAssistant.discover (/app/lib/extension/homeassistant.js:608:41)
    at HomeAssistant.onMQTTConnected (/app/lib/extension/homeassistant.js:541:18)
    at Controller.callExtensionMethod (/app/lib/controller.js:381:44)
    at async Controller.start (/app/lib/controller.js:167:9))
```

configuration.yaml (excerpt):

```
devices:
  '0x00000000209d1cb9':
    friendly_name: '0x00000000209d1cb9'
    homeassistant:
      switch_l1: null
```

Koenkk, 2021-01-12:
Ah yes, that has already been fixed in the (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

ajlunis, 2022-07-04:
Is there any way to prevent the 'effect' attribute from being discovered for lights?  Sorry, I'm just coming from ST and trying to figure out how things work, but the effects don't work on any of my Sengled bulbs as stated and there's a few other people with the same problem, so I'd just like to get rid of them entirely.  filtered_attributes doesn't seem to work though, and I've tried to adapt your switch config to this but can't quite figure it out. Thanks!