Koenkk/zigbee2mqtt #5694: Changing Zigbee 3.0 cc2652rb to Zigbee 3.0 CC2538 killed all devices and entities

wojciechczyz, 2021-01-11:
## What happened

All devices and entities got deleted after I have changed Zigbee 3.0 stick type to another one. According to documentation re-pairing is not needed{

"Requires repairing"
...
"Except when switching between adapters with the following chips: CC2652, CC1352, CC253* (only when running Zigbee 3.0 firmware)"

In my case I am using Zigbee 3.0 firmware on my previous cc2652rb and replacement CC2538.

I see in log that zigbee2mqtt is killing all entities publishing null:

Zigbee2MQTT:info  2021-01-11 11:26:38: MQTT publish: topic 'homeassistant/sensor/0x00158d00042425a7/battery/config', payload 'null'
Zigbee2MQTT:info  2021-01-11 11:26:38: MQTT publish: topic 'homeassistant/sensor/0x00158d00042425a7/linkquality/config', payload 'null'
Zigbee2MQTT:info  2021-01-11 11:26:38: MQTT publish: topic 'homeassistant/sensor/0x00158d00042425a7/illuminance_lux/config', payload 'null'
Zigbee2MQTT:info  2021-01-11 11:26:38: MQTT publish: topic 'homeassistant/sensor/0x00158d0004512286/illuminance/config', payload 'null'
Zigbee2MQTT:info  2021-01-11 11:26:38: MQTT publish: topic 'homeassistant/sensor/0x00158d0004512286/battery/config', payload 'null'
Z

## What did you expect to happen

I would expect that I can switch Zigbee 3.0 coordinator from one type to another keeping all configuration.

## How to reproduce it (minimal and precise)

My steps were:
1. I have stopped latest zigbee2mqtt 1.17.0.1
2. I have changed ONLY serial connection from:
  /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CB_F3_CD-if00-port0
to:
/dev/serial/by-id/usb-Texas_Instruments_CC2538_USB_CDC-if00
both are Zigbee 3.0 firmware coordinators

3. Started zigbee2mqtt,

## Debug info
Zigbee2MQTT version: 1.17.0.1
Adapter hardware: previous cc2652rb, replacement CC2538.
Adapter firmware version: latest

Koenkk, 2021-01-11:
Can you reproduce this again by:
1. Start with CC2652
2. Pair one device
3. Stop, replace with CC2538 
4. Start

And provide the herdsman debug logging of step 4.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

wojciechczyz, 2021-01-11:
Hi Koenkk,

I have restored paired devices in Zigbee2Mqtt and then entities and devices in Home Assistant so my system is running again.
If you would see value in reproducing it again for diagnostic purpose and to improve zigbee2mqtt let me know and I will do this.

Thank you,
Wojciech

Koenkk, 2021-01-12:
> If you would see value in reproducing it again for diagnostic purpose and to improve zigbee2mqtt let me know and I will do this.

Yes so (if you are also interested in this), otherwise close it.

wojciechczyz, 2021-01-12:
I will close for now, as I already flashed CC2538 as router. Thank you.

closed 2021-01-12 by wojciechczyz