Koenkk/zigbee2mqtt #5696: > Which Zigbee 3.0 firmware would work on the CC2652P (with amplifier)?

Gio-dot, 2021-01-11:
> Which Zigbee 3.0 firmware would work on the CC2652P (with amplifier)?
> 
> CC1352P has amp (amp enable registers and sub-1 GHz, not used) but the CC2652R has no amp and no sub-1 GHz registers.
> 
> One of the modules I ordered has a built-in amp: https://www.aliexpress.com/item/1005001265532720.html
> 
> My plan is to connect then thru a CP2104 or FT232H serial to USB and use it as a coordinator.
> 
> Thank you

See: 
https://github.com/egony/cc2652p_E72-2G4M20S1E/tree/master/firmware/coordinator
and:
https://github.com/egony/cc2652p_E72-2G4M20S1E/wiki

_Originally posted by @ProASV in https://github.com/Koenkk/zigbee2mqtt/issues/1429#issuecomment-741661570_

Gio-dot, 2021-01-11:
Hi, i read this post some time ago, but now i see that CC2652 support is also in koenkk firmware (https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_other_20210107.zip). Which is best to use at the moment? I suppose Koenkk fw, am i right?

znanev, 2021-01-11:
@Gio-dot What hardware are you using? I've built egony's board for E72-2GM20S1E module with external antenna and for now I've flashed his firmware (https://github.com/egony/cc2652p_E72-2G4M20S1E/tree/master/firmware/coordinator). Haven't tried Koenkk's yet, but I suppose it terms of Zigbee functionality it shouldn't be much different than egony's.

Gio-dot, 2021-01-11:
> @Gio-dot What hardware are you using? I've built egony's board for E72-2GM20S1E module with external antenna and for now I've flashed his firmware [...]

I would like to use the same module.

Koenkk, 2021-01-11:
These firmwares can be used: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin (and are recommended). Firmwares should be similar (or at least almost) as those guys helped me to built this one.

closed 2021-01-11 by Koenkk

Gio-dot, 2021-01-12:
> @Gio-dot What hardware are you using? I've built egony's board for E72-2GM20S1E module with external antenna and for now I've flashed his firmware [...]

Did you try other modules before (like cc2538+2592)? How is this compared?

znanev, 2021-01-12:
> Did you try other modules before (like cc2538+2592)? How is this compared?

Yes, I've also built one of these:

https://github.com/egony/MODKAM-STICK-V3/wiki

The module that I have is the "alternative" one.

In terms of range, both devices are quite on par, but I have strong preferences towards the E72-based ones. On one hand I was able to buy the E72 modules quite cheaply from Aliexpress (less than $5 shipped a piece), compared to at least $8 shipped for the cc2538+2592 based modules; on the other hand the E72-based modules use newer MCU - CC2652P, so I thought of them to be more future proof. So in terms of performance both should be fine, I'd go with whatever will be cheaper to build :)

Gio-dot, 2021-01-14:
@Koenkk sorry, what are differences between https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_20210107.zip
and
https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_other_20210107.zip
?