Koenkk/zigbee2mqtt #5701: Ledvance Smart+ Motion Sensor

CBismark, 2021-01-11:
Hello everyboy,

I tried to pair a Ledvance Smart+ Motion Sensor, but that didn't work. I was under the impression that this device is basically an OSRAM Smart+ Motion sensor, which is supported. However, that didn't turn out to be the truth.

Below the database entry generated by zigbee2mqtt.

`{"id":22,"type":"EndDevice","ieeeAddr":"0x000d6f000bb30004","nwkAddr":29656,"manufId":4174,"epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1026,1280,2821],"outClusterList":[25],"clusters":{},"binds":[],"meta":{}},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1610388791828}`

As you cansee, the interview doesn't complete.

Any chance this device can be supported?

Kind Regards,

Christoph

Koenkk, 2021-01-11:
Can you provide the herdsman debug log when pairing it?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

CBismark, 2021-01-11:
From the point where I started the pairing process:

`zigbee-herdsman:controller:log Device '0x000d6f000bb30004' joined +22s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms
  zigbee-herdsman:controller:log Device '0x000d6f000bb30004' accepted by handler +0ms
  zigbee-herdsman:controller:log New device '0x000d6f000bb30004' joined +0ms
  zigbee-herdsman:controller:log Creating device '0x000d6f000bb30004' +1ms
Zigbee2MQTT:info  2021-01-11 20:07:41: Device '0x000d6f000bb30004' joined
  zigbee-herdsman:controller:log Interview '0x000d6f000bb30004' start +39ms
Zigbee2MQTT:info  2021-01-11 20:07:41: Starting interview of '0x000d6f000bb30004'
  zigbee-herdsman:controller:device:log Interview - start device '0x000d6f000bb30004' +0ms
Zigbee2MQTT:info  2021-01-11 20:07:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x000d6f000bb30004"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-01-11 20:07:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x000d6f000bb30004"},"type":"pairing"}'
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x89af EP:0 SeqNr. 37 request id: 7 +97ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 7 status: 0 +99ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +2ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +584ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +583ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110 +7ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +7ms
  zigbee-herdsman:deconz:driver query aps data confirm +9ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 38 +100ms
  zigbee-herdsman:deconz:driver query aps data indication +0ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x89af request id: 7 confirm status: 0 +111ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +2ms
  zigbee-herdsman:deconz:driver query aps data indication +1ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 39 +97ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 39 srcAddr: 0x89af destAddr: 0x0 profile id: 0x0 cluster id: 0x8002 +100ms
  zigbee-herdsman:deconz:frameParser response payload: 7,0,175,137,2,64,128,78,16,82,82,0,0,0,82,0,0 +0ms
  zigbee-herdsman:deconz:adapter resolve data request without a transSeq Nr. +22s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms
  zigbee-herdsman:deconz:adapter RECEIVING NODE_DESCRIPTOR - addr: 0x89af type: EndDevice manufacturer: 0x104e +1ms
  zigbee-herdsman:controller:device:log Interview - got node descriptor for device '0x000d6f000bb30004' +866ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x89af EP:0 SeqNr. 41 request id: 8 +97ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 40 +0ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 8 status: 0 +100ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +2ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr.: 40 status: 5 +3ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +639ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +642ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110 +50ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +50ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110 +6ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +6ms
  zigbee-herdsman:deconz:driver query aps data confirm +3ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 42 +98ms
  zigbee-herdsman:deconz:driver query aps data indication +2ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x89af request id: 8 confirm status: 0 +103ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +1ms
  zigbee-herdsman:deconz:driver query aps data indication +0ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 43 +97ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 43 srcAddr: 0x89af destAddr: 0xfffd profile id: 0x0 cluster id: 0x6 +101ms
  zigbee-herdsman:deconz:frameParser response payload: 2,253,255,4,1,0,1,0,5 +0ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +4ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 44 +97ms
  zigbee-herdsman:deconz:driver query aps data indication +2ms
[... cut ...]
  zigbee-herdsman:deconz:driver query aps data indication +89ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 64 +99ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 64 srcAddr: 0xa384 destAddr: 0xfffd profile id: 0x0 cluster id: 0x6 +191ms
  zigbee-herdsman:deconz:frameParser response payload: 8,253,255,4,1,0,1,0,5 +0ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms`

Koenkk, 2021-01-12:
Can you provide it till interview failed?

CBismark, 2021-01-12:
I added the full log. But if I unterstand you correctly, you wanted the log until an interview failed happens and, well that doesn't really happen.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5804742/log.txt)

As you can see, repeatedly the device rejoins,gets an updated network address and then it says: not interviewing:

`  zigbee-herdsman:controller:log Device '0x000d6f000bb30004' joined +2s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms
  zigbee-herdsman:controller:log Device '0x000d6f000bb30004' accepted by handler +0ms
  zigbee-herdsman:controller:log Device '0x000d6f000bb30004' is already in database with different networkAddress, updating networkAddress +1ms
  zigbee-herdsman:controller:log Not interviewing '0x000d6f000bb30004', completed 'false', in progress 'true' +2ms`

Not sure whats going on there. I also can't seem to get the device into the state from the beginning, no matter how long I leave the battery out and how long or short I press the button during inserting the battery again.
Tried it the way it says for the OSRAM sensor: https://www.zigbee2mqtt.io/devices/AC01353010G.html

Kind Regards,

Christoph

Koenkk, 2021-01-12:
Very strange, I'm afraid we have to sniff the traffic while joining to get more info: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

CBismark, 2021-01-12:
Hm okay sounds doable. I got some extra CC2531 lying around here, but no programmer to flash them. Maybe I can get one and then try that. I'll be in contact once I got results!

CBismark, 2021-01-14:
I bought a CCDebugger and flashed an old CC2531 with it, sniffed while powering on the motion sensor. Here is the result as a Wireshark Capture File. Where it says 'Beacon' is when I put the battery in the sensor.
[sniff.zip](https://github.com/Koenkk/zigbee2mqtt/files/5816601/sniff.zip)

Koenkk, 2021-01-15:
I think the devices keeps resetting because the `Match descriptor request` is never answered. This device expects cluster `0x0500` as an output cluster.  @chrishae can you add this cluster

[image]

ChrisHae, 2021-01-18:
There is already the Cluster 0x500 in the deCONZ adapter.
@CBismark Which FW version do you use? You should updat to the latest: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually
You can also try to re-pair the motion sensor when you delete it from the z2mqtt database first.
@Koenkk Shouldn't the IAS Zone enroll Request be independent of the match descriptor request?

CBismark, 2021-01-18:
Did a Firmware update and the problem persists. Sniffed again, maybe you can make something out of it:
@ChrisHae I usually delete from configuration and database (while zigbee2mqtt is stopped), then try to pair.
[sniff2.zip](https://github.com/Koenkk/zigbee2mqtt/files/5830423/sniff2.zip)

ChrisHae, 2021-01-19:
@CBismark Do you have deCONZ installed? You could already check if the cluster 0x500 is missing. The Z2mqtt code is coming soon.
[image]

CBismark, 2021-01-19:
@ChrisHae I don't have a Linux with a GUI installed, only running OpenHAB on the standard openHabian (Raspian). If I would install deconz as a service, can I also check for the Cluster?

ChrisHae, 2021-01-20:
Ok, I have added the simple descriptor and cluster check to zigbee-herdsman code. Wait until new z2mqtt release and then try it again. It should be visible in the debug log during startup. If this does not solve the problem, then it must be something else.

Koenkk, 2021-01-20:
@ChrisHae your fix is available in the latest dev now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

CBismark, 2021-01-20:
Thank you both a lot for the quick help. With dev branch interviewing and pairing worked and the device is now recognized as a Motion Sensor-A.
I'll close this now.

closed 2021-01-20 by CBismark