Koenkk/zigbee2mqtt #5706: Support for Aqara motion sensor T1 (RTCGQ12LM; lumi.motion.agl02)

Otnow, 2021-01-11:
## Information about the device + link

Aqara motion sensor T1 (RTCGQ12LM; lumi.motion.agl02)

It has 1 parameter for customization:
- Detection period (2-199s, step: 1s) - although stated 1-200s

[image]

- https://www.aqara.com/cn/productDetail/d17 (https://www.aqara.com/cn/productInner/d17)
- https://fccid.io/2AKIT-AS011
- https://www.reddit.com/r/Aqara/comments/kuskin/quick_test_aqara_motion_sensor_t1_rtcgq12lm/
- https://aliexpress.com/item/1005001889397376.html?&sku_id=12000018069572598

## data/database.db entry of the device

{"id":46,"type":"EndDevice","ieeeAddr":"0x04cf8cdf3c7fd37e","nwkAddr":56004,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.motion.agl02","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1030,3,1],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.motion.agl02","manufacturerName":"LUMI","powerSource":3,"zclVersion":3,"appVersion":29,"stackVersion":2,"hwVersion":1,"dateCode":"20200327","swBuildId":"2019\u0000www."}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":29,"stackVersion":2,"hwVersion":1,"dateCode":"20200327","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1610396769681}

## Additional info

<details>
  <summary>Zigbee2mqtt pairing log</summary>

- Zigbee2MQTT:debug 2021-01-11 23:25:46: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
- Zigbee2MQTT:info  2021-01-11 23:25:46: Zigbee: allowing new devices to join.
- Zigbee2MQTT:info  2021-01-11 23:25:47: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"1.17.0","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":104,"product":0,"revision":"0x26680700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.17.0"}'
- Zigbee2MQTT:info  2021-01-11 23:25:57: Device '0x04cf8cdf3c7fd37e' joined
- Zigbee2MQTT:info  2021-01-11 23:25:57: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_joined"}'
- Zigbee2MQTT:info  2021-01-11 23:25:57: Starting interview of '0x04cf8cdf3c7fd37e'
- Zigbee2MQTT:info  2021-01-11 23:25:57: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e","status":"started"},"type":"device_interview"}'
- Zigbee2MQTT:info  2021-01-11 23:25:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_connected"}'
- Zigbee2MQTT:info  2021-01-11 23:25:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"pairing"}'
- Zigbee2MQTT:debug 2021-01-11 23:26:01: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.motion.agl02"}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-11 23:26:01: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2021-01-11 23:26:02: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-11 23:26:02: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2021-01-11 23:26:03: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-11 23:26:03: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2021-01-11 23:26:04: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-11 23:26:04: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2021-01-11 23:26:05: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"appVersion":29}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-11 23:26:05: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2021-01-11 23:26:06: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-11 23:26:06: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2021-01-11 23:26:07: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-11 23:26:07: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2021-01-11 23:26:08: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20200327"}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-11 23:26:08: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2021-01-11 23:26:09: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2019\u0000www."}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-11 23:26:09: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:info  2021-01-11 23:26:09: Successfully interviewed '0x04cf8cdf3c7fd37e', device has successfully been paired
- Zigbee2MQTT:warn  2021-01-11 23:26:09: Device '0x04cf8cdf3c7fd37e' with Zigbee model 'lumi.motion.agl02' and manufacturer name 'LUMI' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
- Zigbee2MQTT:info  2021-01-11 23:26:09: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e","status":"successful","supported":false},"type":"device_interview"}'
- Zigbee2MQTT:info  2021-01-11 23:26:09: MQTT publish: topic 'zigbee2mqtt/bridge/devices', payload '**[{"date_code":"20200327","definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":["genBasic","msOccupancySensing","genIdentify","genPowerCfg"],"output":["genIdentify","genOta"]},"configured_reportings":[]}},"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e","interview_completed":true,"interviewing":false,"model_id":"lumi.motion.agl02","network_address":56004,"power_source":"Battery","software_build_id":"2019\u0000www.","supported":false,"type":"EndDevice"}]**'
- Zigbee2MQTT:info  2021-01-11 23:26:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x04cf8cdf3c7fd37e","supported":false},"type":"pairing"}'

</details>

<details>
  <summary>Dump zigbee-packets with ZBOSS Sniffer and Wireshark</summary>

The dump was made according to [this instruction](https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html):
- [RTCGQ12LM.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/5798137/RTCGQ12LM.pcapng.gz) - encrypted pcapng-dump (_Network Key for decryption entries: 21724f48ad8ebbc4f96ae7000506d3fc_)
- [RTCGQ12LM.txt.zip](https://github.com/Koenkk/zigbee2mqtt/files/5798141/RTCGQ12LM.txt.zip) - decrypted plain text dump

**Recorded events in the dump:**
- sensor pairing
- changing sensor settings
	- _Detection period = 199s_ - dump entry №121
	- _Detection period = 2s_ - dump entry №183
- motion detections

**Environment:**
- Sensor firmware version: _0.0.0_0029_
- Aqara Hub M1S (ZHWG15LM) firmware version: _3.1.0_0005.0517_
- Aqara Home app version: _2.2.4_

</details>

sjorge, 2021-01-11:
Try with adding to devices.js

```
    {
        zigbeeModel: ['lumi.motion.agl02'],
        model: 'RTCGQ12LM',
        vendor: 'Xiaomi',
        description: 'Aqara human body movement and illuminance sensor',
        meta: {battery: {voltageToPercentage: '3V_2100'}},
        fromZigbee: [fz.xiaomi_battery, fz.occupancy_with_timeout, fz.RTCGQ11LM_illuminance, fz.RTCGQ11LM_interval],
        toZigbee: [],
        exposes: [e.battery(), e.occupancy(), e.illuminance_lux().withProperty('illuminance'),
            e.illuminance().withUnit('lx').withDescription('Measured illuminance in lux')],
    },
```

Otnow, 2021-01-11:
> Try with adding to devices.js

<details>
  <summary>Zigbee2mqtt pairing log with modify devices.js</summary>

- Zigbee2MQTT:debug 2021-01-12 00:30:03: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
- Zigbee2MQTT:info  2021-01-12 00:30:03: Zigbee: allowing new devices to join.
- Zigbee2MQTT:info  2021-01-12 00:30:04: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"1.17.0","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":104,"product":0,"revision":"0x26680700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.17.0"}'
- Zigbee2MQTT:info  2021-01-12 00:30:13: Device '0x04cf8cdf3c7fd37e' joined
- Zigbee2MQTT:info  2021-01-12 00:30:13: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_joined"}'
- Zigbee2MQTT:info  2021-01-12 00:30:13: Starting interview of '0x04cf8cdf3c7fd37e'
- Zigbee2MQTT:info  2021-01-12 00:30:13: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e","status":"started"},"type":"device_interview"}'
- Zigbee2MQTT:info  2021-01-12 00:30:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_connected"}'
- Zigbee2MQTT:info  2021-01-12 00:30:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"pairing"}'
- Zigbee2MQTT:debug 2021-01-12 00:30:19: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.motion.agl02"}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-12 00:30:20: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-12 00:30:21: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-12 00:30:22: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-12 00:30:23: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"appVersion":29}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-12 00:30:24: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-12 00:30:25: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-12 00:30:26: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20200327"}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2021-01-12 00:30:27: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2019\u0000www."}' from endpoint 1 with groupID null
- Zigbee2MQTT:info  2021-01-12 00:30:27: Successfully interviewed '0x04cf8cdf3c7fd37e', device has successfully been paired
- Zigbee2MQTT:info  2021-01-12 00:30:27: Device '0x04cf8cdf3c7fd37e' is supported, identified as: Xiaomi Aqara T1 human body movement and illuminance sensor (RTCGQ12LM)
- Zigbee2MQTT:info  2021-01-12 00:30:27: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara T1 human body movement and illuminance sensor","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Indicates whether the device detected occupancy","name":"occupancy","property":"occupancy","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Measured illuminance in lux","name":"illuminance_lux","property":"illuminance","type":"numeric","unit":"lx"},{"access":1,"description":"Measured illuminance in lux","name":"illuminance","property":"illuminance","type":"numeric","unit":"lx"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"RTCGQ12LM","vendor":"Xiaomi"},"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e","status":"successful","supported":true},"type":"device_interview"}'
- Zigbee2MQTT:info  2021-01-12 00:30:27: MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c7fd37e/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c7fd37e"],"manufacturer":"Xiaomi","model":"Aqara T1 human body movement and illuminance sensor (RTCGQ12LM)","name":"0x04cf8cdf3c7fd37e","sw_version":"Zigbee2MQTT 1.17.0"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c7fd37e","name":"0x04cf8cdf3c7fd37e battery","state_topic":"zigbee2mqtt/0x04cf8cdf3c7fd37e","unique_id":"0x04cf8cdf3c7fd37e_battery_zigbee2mqtt","unit_of_mea [... cut ...] TT 1.17.0"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c7fd37e","name":"0x04cf8cdf3c7fd37e linkquality","state_topic":"zigbee2mqtt/0x04cf8cdf3c7fd37e","unique_id":"0x04cf8cdf3c7fd37e_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'

</details>

After pairing, no sensor events appear.

Sensor exposes in Web-interface:

_battery - N/A %
occupancy - N/A 
illuminance_lux - N/A lx
illuminance - N/A lx
linkquality - 47 lqi_

sjorge, 2021-01-11:
If you remove all entries from fromZigbee, restart z2m... do any messages show up for missing converters?

Otnow, 2021-01-11:
> If you remove all entries from fromZigbee, restart z2m... do any messages show up for missing converters?

<details>
  <summary>Only such events appeared</summary>

- Zigbee2MQTT:debug 2021-01-12 01:27:45: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"lumi.motion.agl02"}'
- Zigbee2MQTT:debug 2021-01-12 01:27:46: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"LUMI"}'
- Zigbee2MQTT:debug 2021-01-12 01:27:47: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
- Zigbee2MQTT:debug 2021-01-12 01:27:48: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
- Zigbee2MQTT:debug 2021-01-12 01:27:49: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":29}'
- Zigbee2MQTT:debug 2021-01-12 01:27:50: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":2}'
- Zigbee2MQTT:debug 2021-01-12 01:27:51: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
- Zigbee2MQTT:debug 2021-01-12 01:27:52: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20200327"}'
- Zigbee2MQTT:debug 2021-01-12 01:27:53: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"swBuildId":"2019\u0000www."}'

</details>

Moreover, after listening to the zigbee-traffic, I see that the sensor does not respond to movement (there are no corresponding events - "Report Attributes") after pairing with the zigbee-stick.

Koenkk, 2021-01-12:
@Otnow I think you have to sniff the traffic with the original gateway as we need to figure out how to configure the detection period. So start sniffing, pair it with the gateway, change detection period. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Otnow, 2021-01-12:
> @Otnow I think you have to sniff the traffic with the original gateway as we need to figure out how to configure the detection period. So start sniffing, pair it with the gateway, change detection period. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

I already made such a dump in advance and placed it in the initial post in the **Additional info** section under the spoiler **Dump zigbee-packets with ZBOSS Sniffer and Wireshark** )

Koenkk, 2021-01-13:
@Otnow sorry for being blind. I've created an initial pr, it only allows setting the detection period: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2079/files 

Seems that something non standard is used for occupancy. Can you provide the packet numbers of occupancy: true and false (please provide multiple)

Otnow, 2021-01-13:
> Seems that something non standard is used for occupancy. Can you provide the packet numbers of occupancy: true and false (please provide multiple)

When motion is detected, only 1 event is generated: _dump entries № 214, 220, 226, 232, 238, 244, 251, 257, 263_ (events were generated every 2 seconds, because _detection period = 2s_).
There are no events at the end of the detection period.

Otnow, 2021-01-13:
By analogy with the _Aqara T1 temperature, humidity and pressure sensor (WSDCGQ12LM)_, this code:

	{
        zigbeeModel: ['lumi.motion.agl02'],
        model: 'RTCGQ12LM',
        vendor: 'Xiaomi',
        description: 'Aqara T1 human body movement and illuminance sensor',
        fromZigbee: [],
        toZigbee: [],
        meta: {},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            const binds = ['msOccupancySensing'];			
            await reporting.bind(endpoint, coordinatorEndpoint, binds);
        },
		exposes: [],
    },

allows you to generate a motion detection event in the log:

- Zigbee2MQTT:debug 2021-01-13 22:16:02: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
- Zigbee2MQTT:debug 2021-01-13 22:16:02: No converter available for 'RTCGQ12LM' with cluster 'msOccupancySensing' and type 'attributeReport' and data '{"occupancy":1}'
- Zigbee2MQTT:debug 2021-01-13 22:18:00: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 1 with groupID 0
- Zigbee2MQTT:debug 2021-01-13 22:18:00: No converter available for 'RTCGQ12LM' with cluster 'msOccupancySensing' and type 'attributeReport' and data '{"occupancy":0}'

Koenkk, 2021-01-13:
Can you check updates of https://github.com/Koenkk/zigbee-herdsman-converters/pull/2079/files?

Following should work now: occupancy, illuminance and battery (please check carefully in the logs if configure is succesfull, you can force it via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestdeviceconfigure)

Otnow, 2021-01-13:
> Can you check updates of https://github.com/Koenkk/zigbee-herdsman-converters/pull/2079/files?

The test showed that motion events appear, but the set detection period is not observed, although the parameter is set normally. Also, illuminance and battery power are not displayed.

<details>
  <summary>Log</summary>

- debug 2021-01-14 00:55:18: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
- info  2021-01-14 00:55:18: Zigbee: allowing new devices to join.
- info  2021-01-14 00:55:18: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"1.17.0","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":15,"extendedPanID":"0x00124b0014d9f126","panID":6754},"permit_join":true,"version":"1.17.0"}'
- info  2021-01-14 00:55:30: Device '0x04cf8cdf3c7fd37e' joined
- info  2021-01-14 00:55:30: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_joined"}'
- info  2021-01-14 00:55:30: Starting interview of '0x04cf8cdf3c7fd37e'
- info  2021-01-14 00:55:30: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e","status":"started"},"type":"device_interview"}'
- info  2021-01-14 00:55:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_connected"}'
- info  2021-01-14 00:55:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"pairing"}'
- debug 2021-01-14 00:55:31: Device '0x04cf8cdf3c7fd37e' announced itself
- info  2021-01-14 00:55:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_announce"}'
- info  2021-01-14 00:55:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_announced"}'
- debug 2021-01-14 00:55:34: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.motion.agl02"}' from endpoint 1 with groupID 0
- debug 2021-01-14 00:55:34: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"lumi.motion.agl02"}'
- debug 2021-01-14 00:55:35: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
- debug 2021-01-14 00:55:35: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"LUMI"}'
- debug 2021-01-14 00:55:38: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
- debug 2021-01-14 00:55:38: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
- debug 2021-01-14 00:55:39: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
- debug 2021-01-14 00:55:39: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
- debug 2021-01-14 00:55:40: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"appVersion":29}' from endpoint 1 with groupID 0
- debug 2021-01-14 00:55:40: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":29}'
- debug 2021-01-14 00:55:41: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
- debug 2021-01-14 00:55:41: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":2}'
- debug 2021-01-14 00:55:42: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
- debug 2021-01-14 00:55:42: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
- debug 2021-01-14 00:55:43: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20200327"}' from endpoint 1 with groupID 0
- debug 2021-01-14 00:55:43: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20200327"}'
- debug 2021-01-14 00:55:44: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2019\u0000www."}' from endpoint 1 with groupID 0
- debug 2021-01-14 00:55:44: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"swBuildId":"2019\u0000www."}'
- info  2021-01-14 00:55:44: MQTT publish: topic 'homeassistant/binary_sensor/0x04cf8cdf3c7fd37e/occupancy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c7fd37e"],"manufacturer":"Xiaomi","model":"Aqara T1 human body movement and illumin [... cut ...]
- info  2021-01-14 01:11:01: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"detection_period":3,"linkquality":115,"occupancy":true,"voltage":3100}'
- debug 2021-01-14 01:12:57: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 1 with groupID 0
- info  2021-01-14 01:12:57: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"detection_period":3,"linkquality":157,"occupancy":false,"voltage":3100}'

</details>

Koenkk, 2021-01-15:
Made some more updates to https://github.com/Koenkk/zigbee-herdsman-converters/pull/2079/files, battery and illuminance should work after this 

> , but the set detection period is not observed, although the parameter is set normally.

I don't understand what you mean with this, can you explain it further?

Otnow, 2021-01-15:
> Made some more updates to https://github.com/Koenkk/zigbee-herdsman-converters/pull/2079/files, battery and illuminance should work after this

The battery power is now displayed, but the illuminance is still not:

```
- Zigbee2MQTT:error 2021-01-15 20:58:32: Failed to configure '0x04cf8cdf3c7fd37e', attempt 1 (Error: ConfigureReporting 0x04cf8cdf3c7fd37e/1 msIlluminanceMeasurement([{"attribute":"measuredValue","minimumReportInterval":15,"maximumReportInterval":3600,"reportableChange":500}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 9879 - 1 - 14 - 1024 - 7 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
```

<details>
  <summary>Here is this error in the form of 2 zigbee packets (command and response)</summary>

```
No.     Time                          Source                Destination           Protocol Length Dectination Port Info
   1214 2021-01-15 20:59:14.927202    0x0000                0x2697                ZigBee HA 69                      ZCL: Configure Reporting, Seq: 18

Frame 1214: 69 bytes on wire (552 bits), 69 bytes captured (552 bits) on interface \\.\pipe\zboss_sniffer, id 0
    Interface id: 0 (\\.\pipe\zboss_sniffer)
        Interface name: \\.\pipe\zboss_sniffer
    Encapsulation type: IEEE 802.15.4 Wireless PAN (104)
    Arrival Time: Jan 15, 2021 20:59:14.927202000 RTZ 2 (зима)
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1610733554.927202000 seconds
    [Time delta from previous captured frame: 0.007457000 seconds]
    [Time delta from previous displayed frame: 0.007457000 seconds]
    [Time since reference or first frame: 970.230916000 seconds]
    Frame Number: 1214
    Frame Length: 69 bytes (552 bits)
    Capture Length: 69 bytes (552 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: wpan:zbee_nwk:zbee_aps:zbee_zcl]
ZBOSS dump, IN, page 0, channel 15
    .... ...0 = ZBOSS Direction: IN (0x0)
    0000 000. = Page: 2.4 GHz (0)
    Channel: 15
    Trace number: 0
IEEE 802.15.4 Data, Dst: 0x2697, Src: 0x0000
    Frame Control Field: 0x8861, Frame Type: Data, Acknowledge Request, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
        .... .... .... .001 = Frame Type: Data (0x1)
        .... .... .... 0... = Security Enabled: False
        .... .... ...0 .... = Frame Pending: False
        .... .... ..1. .... = Acknowledge Request: True
        .... .... .1.. .... = PAN ID Compression: True
        .... .... 0... .... = Reserved: False
        .... ...0 .... .... = Sequence Number Suppression: False
        .... ..0. .... .... = Information Elements Present: False
        .... 10.. .... .... = Destination Addressing Mode: Short/16-bit (0x2)
        ..00 .... .... .... = Frame Version: IEEE Std 802.15.4-2003 (0)
        10.. .... .... .... = Source Addressing Mode: Short/16-bit (0x2)
    Sequence Number: 175
    Destination PAN: 0x1a62
    Destination: 0x2697
    Source: 0x0000
    [Extended Source: TexasIns_00:14:d9:f1:26 (00:12:4b:00:14:d9:f1:26)]
    [Origin: 1]
    TI CC24xx-format metadata: FCS OK
        FCS Valid: True
        RSSI: -30 dB
        LQI Correlation Value: 108
ZigBee Network Layer Data, Dst: 0x2697, Src: 0x0000
    Frame Control Field: 0x0248, Frame Type: Data, Discover Route: Enable, Security Data
        .... .... .... ..00 = Frame Type: Data (0x0)
        .... .... ..00 10.. = Protocol Version: 2
        .... .... 01.. .... = Discover Route: Enable (0x1)
        .... ...0 .... .... = Multicast: False
        .... ..1. .... .... = Security: True
        .... .0.. .... .... = Source Route: False
[... 227 lines cut ...]
- Zigbee2MQTT:debug 2021-01-15 20:58:56: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
- Zigbee2MQTT:info  2021-01-15 20:58:56: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"linkquality":105,"occupancy":true,"voltage":3100}'
- Zigbee2MQTT:info  2021-01-15 20:58:56: Configuring '0x04cf8cdf3c7fd37e'
- Zigbee2MQTT:debug 2021-01-15 20:59:05: Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c7fd37e/set' with data '{"detection_period":5}'
- Zigbee2MQTT:debug 2021-01-15 20:59:05: Publishing 'set' 'detection_period' to '0x04cf8cdf3c7fd37e'
- Zigbee2MQTT:error 2021-01-15 20:59:25: Failed to configure '0x04cf8cdf3c7fd37e', attempt 2 (Error: ConfigureReporting 0x04cf8cdf3c7fd37e/1 msIlluminanceMeasurement([{"attribute":"measuredValue","minimumReportInterval":15,"maximumReportInterval":3600,"reportableChange":500}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 9879 - 1 - 18 - 1024 - 7 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
[... cut ...]
- Zigbee2MQTT:info  2021-01-15 21:02:22: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"detection_period":5,"linkquality":136,"occupancy":true,"voltage":3100}'
- Zigbee2MQTT:debug 2021-01-15 21:04:08: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 1 with groupID 0
- Zigbee2MQTT:info  2021-01-15 21:04:08: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"detection_period":5,"linkquality":136,"occupancy":false,"voltage":3100}'
```

</details>

Koenkk, 2021-01-16:
@Otnow can you provide a sniff with the Xiaomi gateway from pairing till a illuminance report is send?

Otnow, 2021-01-16:
> @Otnow can you provide a sniff with the Xiaomi gateway from pairing till a illuminance report is send?

[RTCGQ12LM_2.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/5825041/RTCGQ12LM_2.pcapng.gz) - encrypted pcapng-dump (Network Key for decryption entries: 21724f48ad8ebbc4f96ae7000506d3fc)
[RTCGQ12LM_2.zip](https://github.com/Koenkk/zigbee2mqtt/files/5825043/RTCGQ12LM_2.zip) - decrypted plain text dump
 
Recorded events in the dump:

- sensor pairing
- №75 - gateway wrote attribute _0x00ff_ (?)
- №79 - sensor reported attribute _0x00ff_ (?)
- №83 - sensor reported attribute _0x0020_ (Battery Voltage)
- №85 - sensor reported attribute _0x00f7_ (?)
- №93 - sensor reported attribute _0x00f7_ (?)
- №136 - gateway wrote attribute _0x0102 = 2_ (I set in the app: _detection period = 2s_)
- №233 - sensor reported attribute _0x0112 = 65543_ (in the app is displayed as _motion event with 7 Lux_)
- №299 - sensor reported attribute _0x0112 = 65603_ (in the app is displayed as _motion event with 67 Lux_)
- №384 - sensor reported attribute _0x0112 = 65543_ (in the app is displayed as _motion event with 7 Lux_)
- №392 - sensor reported attribute _0x0112 = 65595_ (in the app is displayed as _motion event with 59 Lux_)

Koenkk, 2021-01-16:
Can you update to the latest zigbee2mqtt dev branch first (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html), apply latest changes from https://github.com/Koenkk/zigbee-herdsman-converters/pull/2079 and check if illuminance is reported now?

Can you also provide a sniff when pairing it with Zigbee2MQTT?

Otnow, 2021-01-18:
> Can you update to the latest zigbee2mqtt dev branch first (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html), apply latest changes from [Koenkk/zigbee-herdsman-converters#2079](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2079) and check if illuminance is reported now?
> Can you also provide a sniff when pairing it with Zigbee2MQTT?

After pairing the sensor with a zigbee-stick, no motion events are generated in zigbee-traffic. Accordingly, they are not in the Zigbee2MQTT log either, as are the illuminance values. The battery power value is not detected again. It was only possible to set the value of the detection period.

<details>
  <summary>Zigbee2mqtt log</summary>

```
- info  2021-01-18 03:41:36: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"2ef2c54","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":15,"extendedPanID":"0x00124b0014d9f126","panID":6754},"permit_join":false,"version":"1.17.0-dev"}'
- debug 2021-01-18 03:41:57: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"transaction":"f3bml-2","value":true}'
- info  2021-01-18 03:41:57: Zigbee: allowing new devices to join.
- info  2021-01-18 03:41:57: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":true},"status":"ok","transaction":"f3bml-2"}'
- info  2021-01-18 03:42:09: Device '0x04cf8cdf3c7fd37e' joined
- info  2021-01-18 03:42:09: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_joined"}'
- info  2021-01-18 03:42:09: Starting interview of '0x04cf8cdf3c7fd37e'
- info  2021-01-18 03:42:09: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e","status":"started"},"type":"device_interview"}'
- info  2021-01-18 03:42:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_connected"}'
- info  2021-01-18 03:42:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"pairing"}'
- debug 2021-01-18 03:42:09: Device '0x04cf8cdf3c7fd37e' announced itself
- info  2021-01-18 03:42:09: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_announce"}'
- info  2021-01-18 03:42:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_announced"}'
- debug 2021-01-18 03:42:12: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.motion.agl02"}' from endpoint 1 with groupID 0
- debug 2021-01-18 03:42:12: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"lumi.motion.agl02"}'
- debug 2021-01-18 03:42:13: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
- debug 2021-01-18 03:42:13: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"LUMI"}'
- debug 2021-01-18 03:42:16: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
- debug 2021-01-18 03:42:16: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
- debug 2021-01-18 03:42:17: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
- debug 2021-01-18 03:42:17: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
- debug 2021-01-18 03:42:18: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"appVersion":35}' from endpoint 1 with groupID 0
- debug 2021-01-18 03:42:18: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":35}'
- debug 2021-01-18 03:42:19: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
- debug 2021-01-18 03:42:19: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":2}'
- debug 2021-01-18 03:42:20: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
- debug 2021-01-18 03:42:20: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
- debug 2021-01-18 03:42:21: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20201113"}' from endpoint 1 with groupID 0
- debug 2021-01-18 03:42:21: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20201113"}'
[... cut ...]
**Recorded events in the dump**:

- sensor pairing
- №181 - coordinator wrote attribute _0x0102 = 2_ (I set in the web-interface Zigbee2mqtt: _detection period = 2s_)
- Next, I make movements in front of the sensor, but it does not respond to them (there are no corresponding events "Report Attributes", as when working with the original gateway)

**Environment:**

- Sensor firmware version: _0.0.0_0035_
- Ebyte E18-2G4U04B (CC2531) firmware version: _zStack30x (20190425)_
- Zigbee2mqtt version: _1.17.0-dev (2ef2c54)_

Koenkk, 2021-01-18:
Updated https://github.com/Koenkk/zigbee-herdsman-converters/pull/2079/files again, can you check what works and provide a sniff?

Otnow, 2021-01-19:
> Updated https://github.com/Koenkk/zigbee-herdsman-converters/pull/2079/files again, can you check what works and provide a sniff?

As before after pairing the sensor with a zigbee-stick, no motion events are generated in zigbee-traffic. Accordingly, they are not in the Zigbee2MQTT log either, as are the illuminance values. _But the battery power value is detected again._

<details>
  <summary>Zigbee2mqtt log</summary>

```
- info  2021-01-19 05:08:10: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"2ef2c54","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":15,"extendedPanID":"0x00124b0014d9f126","panID":6754},"permit_join":false,"version":"1.17.0-dev"}'
- debug 2021-01-19 05:10:22: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"transaction":"qgakc-1","value":true}'
- info  2021-01-19 05:10:22: Zigbee: allowing new devices to join.
- info  2021-01-19 05:10:22: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":true},"status":"ok","transaction":"qgakc-1"}'
- info  2021-01-19 05:10:36: Device '0x04cf8cdf3c7fd37e' joined
- info  2021-01-19 05:10:36: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_joined"}'
- info  2021-01-19 05:10:36: Starting interview of '0x04cf8cdf3c7fd37e'
- info  2021-01-19 05:10:36: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e","status":"started"},"type":"device_interview"}'
- info  2021-01-19 05:10:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_connected"}'
- info  2021-01-19 05:10:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"pairing"}'
- debug 2021-01-19 05:10:37: Device '0x04cf8cdf3c7fd37e' announced itself
- info  2021-01-19 05:10:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_announce"}'
- info  2021-01-19 05:10:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_announced"}'
- debug 2021-01-19 05:10:40: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.motion.agl02"}' from endpoint 1 with groupID 0
- debug 2021-01-19 05:10:40: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"lumi.motion.agl02"}'
- debug 2021-01-19 05:10:41: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
- debug 2021-01-19 05:10:41: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"LUMI"}'
- debug 2021-01-19 05:10:44: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
- debug 2021-01-19 05:10:44: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
- debug 2021-01-19 05:10:45: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
- debug 2021-01-19 05:10:45: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
- debug 2021-01-19 05:10:46: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"appVersion":35}' from endpoint 1 with groupID 0
- debug 2021-01-19 05:10:46: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":35}'
- debug 2021-01-19 05:10:47: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
- debug 2021-01-19 05:10:47: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":2}'
- debug 2021-01-19 05:10:48: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
- debug 2021-01-19 05:10:48: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
- debug 2021-01-19 05:10:49: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20201113"}' from endpoint 1 with groupID 0
- debug 2021-01-19 05:10:49: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20201113"}'
- debug 2021-01-19 05:10:50: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2019\u0000www."}' from endpoint 1 with groupID 0
[... cut ...]
- №258 - coordinator wrote attribute _0x0102 = 2_ (I set in the web-interface Zigbee2mqtt: _detection period = 2s_)
- №262 - sensor reported attribute _0x0020 (Battery Voltage)_
- Next, I make movements in front of the sensor, but it does not respond to them (there are no corresponding events "Report Attributes", as when working with the original gateway)

**Environment:**

- Sensor firmware version: _0.0.0_0035_
- Ebyte E18-2G4U04B (CC2531) firmware version: _zStack30x (20190425)_
- Zigbee2mqtt version: _1.17.0-dev (2ef2c54)_

Koenkk, 2021-01-19:
Don't understand why no occupancy is reported. Can you try again with: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2079/commits/79c3986231d25f4a636921e4842f9ed091170c79 with this you reported that occupancy worked.

Otnow, 2021-01-19:
> Don't understand why no occupancy is reported. Can you try again with: [Koenkk/zigbee-herdsman-converters@79c3986](https://github.com/Koenkk/zigbee-herdsman-converters/commit/79c3986231d25f4a636921e4842f9ed091170c79) with this you reported that occupancy worked.

After pairing the sensor with a zigbee-stick, no motion events are generated in zigbee-traffic. Accordingly, they are not in the Zigbee2MQTT log either, as are the illuminance values. The battery power value is not detected again. It was only possible to set the value of the detection period.

<details>
  <summary>Zigbee2mqtt log</summary>

```
- info  2021-01-19 21:29:34: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"2ef2c54","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":15,"extendedPanID":"0x00124b0014d9f126","panID":6754},"permit_join":false,"version":"1.17.0-dev"}'
- debug 2021-01-19 21:30:24: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"transaction":"obb4b-1","value":true}'
- info  2021-01-19 21:30:24: Zigbee: allowing new devices to join.
- info  2021-01-19 21:30:24: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":true},"status":"ok","transaction":"obb4b-1"}'
- info  2021-01-19 21:30:37: Device '0x04cf8cdf3c7fd37e' joined
- info  2021-01-19 21:30:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_joined"}'
- info  2021-01-19 21:30:37: Starting interview of '0x04cf8cdf3c7fd37e'
- info  2021-01-19 21:30:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e","status":"started"},"type":"device_interview"}'
- info  2021-01-19 21:30:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_connected"}'
- info  2021-01-19 21:30:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"pairing"}'
- debug 2021-01-19 21:30:37: Device '0x04cf8cdf3c7fd37e' announced itself
- info  2021-01-19 21:30:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_announce"}'
- info  2021-01-19 21:30:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_announced"}'
- debug 2021-01-19 21:30:41: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.motion.agl02"}' from endpoint 1 with groupID 0
- debug 2021-01-19 21:30:41: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"lumi.motion.agl02"}'
- debug 2021-01-19 21:30:44: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
- debug 2021-01-19 21:30:44: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"LUMI"}'
- debug 2021-01-19 21:30:45: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
- debug 2021-01-19 21:30:45: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
- debug 2021-01-19 21:30:46: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
- debug 2021-01-19 21:30:46: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
- debug 2021-01-19 21:30:47: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"appVersion":35}' from endpoint 1 with groupID 0
- debug 2021-01-19 21:30:47: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":35}'
- debug 2021-01-19 21:30:48: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
- debug 2021-01-19 21:30:48: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":2}'
- debug 2021-01-19 21:30:49: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
- debug 2021-01-19 21:30:49: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
- debug 2021-01-19 21:30:50: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20201113"}' from endpoint 1 with groupID 0
- debug 2021-01-19 21:30:50: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20201113"}'
[... cut ...]
- №275 - sensor reported attribute _0x0020 (Battery Voltage)_
- №317 - sensor responds that attribute _Battery Percentage Remaining (0x0021)_ is not supported for reporting
- Next, I make movements in front of the sensor, but it does not respond to them (there are no corresponding events "Report Attributes", as when working with the original gateway)

**Environment:**

- Sensor firmware version: _0.0.0_0035_
- Ebyte E18-2G4U04B (CC2531) firmware version: _zStack30x (20190425)_
- Zigbee2mqtt version: _1.17.0-dev (2ef2c54)_

Otnow, 2021-01-19:
In this configuration:

```
    {
        zigbeeModel: ['lumi.motion.agl02'],
        model: 'RTCGQ12LM',
        vendor: 'Xiaomi',
        description: 'Aqara T1 human body movement and illuminance sensor',
        fromZigbee: [fz.occupancy, fz.battery],
        toZigbee: [tz.RTCGQ12LM_detection_period],
        exposes: [e.occupancy(), e.battery(),
            exposes.numeric('detection_period', exposes.access.STATE_SET).withValueMin(2).withValueMax(199).withUnit('s')
                .withDescription('Time in seconds till occupancy goes to false')],
        meta: {configureKey: 1, battery: {voltageToPercentage: '3V_2100'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'msOccupancySensing']);            
	        await reporting.occupancy(endpoint);
            await reporting.batteryVoltage(endpoint);
        },
    },
```

battery power is displayed and motion events are generated (_no illuminance_), but the set detection period is not taken into account (_actually 116s instead of the given 2s_).

<details>
  <summary>Zigbee2mqtt log</summary>

```
- debug 2021-01-19 23:05:45: Received MQTT message on 'zigbee2mqtt/bridge/config' with data '{"commit":"2ef2c54","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":15,"extendedPanID":"0x00124b0014d9f126","panID":6754},"permit_join":false,"version":"1.17.0-dev"}'
- info  2021-01-19 23:05:46: MQTT publish: topic 'zigbee2mqtt/bridge/devices', payload '[{"definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"10":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"11":{"bindings":[],"clusters":{"input":["ssIasAce"],"output":["ssIasZone","ssIasWd"]},"configured_reportings":[]},"110":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"12":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"13":{"bindings":[],"clusters":{"input":["genOta"],"output":[]},"configured_reportings":[]},"2":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"242":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"3":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"4":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"47":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"5":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"6":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"8":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]}},"friendly_name":"Coordinator","ieee_address":"0x00124b0014d9f126","interview_completed":true,"interviewing":false,"network_address":0,"supported":false,"type":"Coordinator"}]'
- info  2021-01-19 23:05:46: MQTT publish: topic 'zigbee2mqtt/bridge/groups', payload '[]'
- info  2021-01-19 23:05:46: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"2ef2c54","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":15,"extendedPanID":"0x00124b0014d9f126","panID":6754},"permit_join":false,"version":"1.17.0-dev"}'
- debug 2021-01-19 23:06:19: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"transaction":"hkt26-1","value":true}'
- info  2021-01-19 23:06:19: Zigbee: allowing new devices to join.
- info  2021-01-19 23:06:19: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":true},"status":"ok","transaction":"hkt26-1"}'
- info  2021-01-19 23:06:32: Device '0x04cf8cdf3c7fd37e' joined
- info  2021-01-19 23:06:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_joined"}'
- info  2021-01-19 23:06:32: Starting interview of '0x04cf8cdf3c7fd37e'
- info  2021-01-19 23:06:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e","status":"started"},"type":"device_interview"}'
- info  2021-01-19 23:06:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_connected"}'
- info  2021-01-19 23:06:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"pairing"}'
- debug 2021-01-19 23:06:32: Device '0x04cf8cdf3c7fd37e' announced itself
- info  2021-01-19 23:06:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_announce"}'
- info  2021-01-19 23:06:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_announced"}'
- debug 2021-01-19 23:06:35: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.motion.agl02"}' from endpoint 1 with groupID 0
[... cut ...]
- №473 - sensor reported attribute _0x0000 = 1_ (Occupied=True)
- №570 - sensor reported attribute _0x0000 = 0_ (Occupied=False) - _after 116s instead of given 2s_
- №572 - sensor reported attribute _0x0000 = 1_ (Occupied=True)
- №581 - sensor reported attribute _0x0000 = 0_ (Occupied=False) - _after 116s instead of given 2s_

**Environment:**

- Sensor firmware version: _0.0.0_0035_
- Ebyte E18-2G4U04B (CC2531) firmware version: _zStack30x (20190425)_
- Zigbee2mqtt version: _1.17.0-dev (2ef2c54)_

Otnow, 2021-01-19:
I tried by analogy with:

```
    occupancy: async (endpoint, overrides) => {
        const p = payload('occupancy', 0, repInterval.HOUR, 0, overrides);
        await endpoint.configureReporting('msOccupancySensing', p);
    },
```

set up reporting in the _reporting.js_ for _0x0112_ attribute from _aqaraOpple_ cluster:

```
    RTCGQ12LM_occupancy: async (endpoint, overrides) => {
        const p = payload('illuminance', 0, repInterval.HOUR, 0, overrides);
        await endpoint.configureReporting('aqaraOpple', p);
    },
```

with _devices.js_:

```
    {
        zigbeeModel: ['lumi.motion.agl02'],
        model: 'RTCGQ12LM',
        vendor: 'Xiaomi',
        description: 'Aqara T1 human body movement and illuminance sensor',
        fromZigbee: [fz.xiaomi_opple_illuminance, fz.battery],
        toZigbee: [tz.RTCGQ12LM_detection_period],
        exposes: [e.illuminance(), e.battery(),
            exposes.numeric('detection_period', exposes.access.STATE_SET).withValueMin(2).withValueMax(199).withUnit('s')
                .withDescription('Time in seconds till occupancy goes to false')],
        meta: {configureKey: 1, battery: {voltageToPercentage: '3V_2100'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'aqaraOpple']);            
	        await reporting.RTCGQ12LM_occupancy(endpoint);
            await reporting.batteryVoltage(endpoint);
        },
    },
```

**but sensor responds that attribute _0x0112_ is not supported for reporting:**

<details>
  <summary>Error log</summary>

```
- error 2021-01-20 00:27:39: Failed to configure '0x04cf8cdf3c7fd37e', attempt 3 (Error: ConfigureReporting 0x04cf8cdf3c7fd37e/1 aqaraOpple([{"attribute":"illuminance","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:196:23)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:419:26)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))

No.     Time                          Source                Destination           Protocol Length Dectination Port Info
    616 2021-01-20 00:27:37.934306    0x0000                0x5c4b                ZigBee HA 71                      ZCL: Configure Reporting, Seq: 16

Frame 616: 71 bytes on wire (568 bits), 71 bytes captured (568 bits) on interface \\.\pipe\zboss_sniffer, id 0
    Interface id: 0 (\\.\pipe\zboss_sniffer)
        Interface name: \\.\pipe\zboss_sniffer
    Encapsulation type: IEEE 802.15.4 Wireless PAN (104)
    Arrival Time: Jan 20, 2021 00:27:37.934306000 RTZ 2 (зима)
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1611091657.934306000 seconds
    [Time delta from previous captured frame: 0.006482000 seconds]
    [Time delta from previous displayed frame: 1.011611000 seconds]
    [Time since reference or first frame: 341.488951000 seconds]
    Frame Number: 616
    Frame Length: 71 bytes (568 bits)
    Capture Length: 71 bytes (568 bits)
    [Frame is marked: False]
[... 169 lines cut ...]
    Source Endpoint: 1
    Counter: 31
ZigBee Cluster Library Frame, Command: Configure Reporting Response, Seq: 16
    Frame Control Field: Profile-wide (0x18)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 1... = Direction: Server to Client
        ...1 .... = Disable Default Response: True
    Sequence Number: 16
    Command: Configure Reporting Response (0x07)
    Attribute Status Record
        Status: Unsupported Attribute (0x86)
        Direction: Reported (0x00)
        Attribute: 0x0112
```

</details>

Accordingly, there are no motion events.

**Environment:**

- Sensor firmware version: _0.0.0_0035_
- Ebyte E18-2G4U04B (CC2531) firmware version: _zStack30x (20190425)_
- Zigbee2mqtt version: _1.17.0-dev (2ef2c54)_

Otnow, 2021-01-20:
By analogy with the _Philips Hue motion sensor (SML001)_, this code:

```
    {
        zigbeeModel: ['lumi.motion.agl02'],
        model: 'RTCGQ12LM',
        vendor: 'Xiaomi',
        description: 'Aqara T1 human body movement and illuminance sensor',
        fromZigbee: [fz.occupancy, fz.battery],
        toZigbee: [tz.occupancy_timeout],
        exposes: [e.occupancy(), e.battery(),
            exposes.numeric('occupancy_timeout', exposes.access.ALL).withValueMin(0).withValueMax(65535).withUnit('s')
                .withDescription('Time in seconds till occupancy goes to false')],
        meta: {configureKey: 1, battery: {voltageToPercentage: '3V_2100'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'msOccupancySensing']);            
	    await reporting.occupancy(endpoint);
            await reporting.batteryVoltage(endpoint);
        },
    },
```

allows you to set the detection period (any, not just 2-199s) and generate a motion detection event based on it:

<details>
  <summary>Zigbee2mqtt log</summary>

```
- debug 2021-01-20 04:05:56: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
- info  2021-01-20 04:05:56: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"linkquality":110,"occupancy":true}'
- debug 2021-01-20 04:06:19: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryVoltage":31}' from endpoint 1 with groupID 0
- info  2021-01-20 04:06:19: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"linkquality":81,"occupancy":true,"voltage":3100}'
- debug 2021-01-20 04:06:20: Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c7fd37e/set' with data '{"occupancy_timeout":0}'
- debug 2021-01-20 04:06:20: Publishing 'set' 'occupancy_timeout' to '0x04cf8cdf3c7fd37e'
- info  2021-01-20 04:06:21: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"linkquality":86,"occupancy":true,"occupancy_timeout":0,"voltage":3100}'
- debug 2021-01-20 04:06:23: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 1 with groupID 0
- info  2021-01-20 04:06:23: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"linkquality":94,"occupancy":false,"occupancy_timeout":0,"voltage":3100}'
- debug 2021-01-20 04:06:30: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
- info  2021-01-20 04:06:30: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"linkquality":84,"occupancy":true,"occupancy_timeout":0,"voltage":3100}'
- debug 2021-01-20 04:06:38: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 1 with groupID 0
- info  2021-01-20 04:06:38: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"linkquality":102,"occupancy":false,"occupancy_timeout":0,"voltage":3100}'
- debug 2021-01-20 04:06:43: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
- info  2021-01-20 04:06:43: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"linkquality":73,"occupancy":true,"occupancy_timeout":0,"voltage":3100}'
- debug 2021-01-20 04:06:48: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 1 with groupID 0
- info  2021-01-20 04:06:48: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"linkquality":102,"occupancy":false,"occupancy_timeout":0,"voltage":3100}'
- debug 2021-01-20 04:06:52: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
- info  2021-01-20 04:06:52: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"linkquality":86,"occupancy":true,"occupancy_timeout":0,"voltage":3100}'
- debug 2021-01-20 04:06:57: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 1 with groupID 0
- info  2021-01-20 04:06:57: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"linkquality":115,"occupancy":false,"occupancy_timeout":0,"voltage":3100}'
- debug 2021-01-20 04:07:00: Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c7fd37e/set' with data '{"occupancy_timeout":0}'
- debug 2021-01-20 04:07:00: Publishing 'set' 'occupancy_timeout' to '0x04cf8cdf3c7fd37e'
- info  2021-01-20 04:07:00: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"linkquality":115,"occupancy":false,"occupancy_timeout":0,"voltage":3100}'
- debug 2021-01-20 04:07:05: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
[... cut ...]
- debug 2021-01-20 04:22:39: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 1 with groupID 0
- info  2021-01-20 04:22:39: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"linkquality":162,"occupancy":false,"occupancy_timeout":300,"voltage":3100}'
```

</details>

In previous configuration attempts, the static delay between start and end of detection in 116s was due to the default attribute _pirOToUDelay_ = 120s.

It remains to understand where to get the value of illuminance.

Koenkk, 2021-01-20:
Really like the config from https://github.com/Koenkk/zigbee2mqtt/issues/5706#issuecomment-763271991 , if I understand correctly everything works except the illuminance? 

Do you see illuminance reports when adding  `msIlluminanceMeasurement` to `reporting.bind`? Make sure to reconfigure the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestdeviceconfigure after changing the configure() (otherwise it won't be applied)

Otnow, 2021-01-21:
> Really like the config from [#5706 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/5706#issuecomment-763271991) , if I understand correctly everything works except the illuminance?

Yes.

> Do you see illuminance reports when adding `msIlluminanceMeasurement` to `reporting.bind`? Make sure to reconfigure the device via [...]

Added `msIlluminanceMeasurement` to `reporting.bind`, but the _illuminance parameter is still not visible_:

<details>
  <summary>Zigbee2mqtt log</summary>

```
- info  2021-01-21 02:29:45: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"2ef2c54","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":15,"extendedPanID":"0x00124b0014d9f126","panID":6754},"permit_join":false,"version":"1.17.0-dev"}'
- debug 2021-01-21 02:30:19: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"transaction":"lnpf2-1","value":true}'
- info  2021-01-21 02:30:19: Zigbee: allowing new devices to join.
- info  2021-01-21 02:30:19: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":true},"status":"ok","transaction":"lnpf2-1"}'
- info  2021-01-21 02:30:34: Device '0x04cf8cdf3c7fd37e' joined
- info  2021-01-21 02:30:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_joined"}'
- info  2021-01-21 02:30:34: Starting interview of '0x04cf8cdf3c7fd37e'
- info  2021-01-21 02:30:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e","status":"started"},"type":"device_interview"}'
- info  2021-01-21 02:30:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_connected"}'
- info  2021-01-21 02:30:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"pairing"}'
- debug 2021-01-21 02:30:34: Device '0x04cf8cdf3c7fd37e' announced itself
- info  2021-01-21 02:30:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_announce"}'
- info  2021-01-21 02:30:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_announced"}'
- debug 2021-01-21 02:30:37: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.motion.agl02"}' from endpoint 1 with groupID 0
- debug 2021-01-21 02:30:37: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"lumi.motion.agl02"}'
- debug 2021-01-21 02:30:38: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
- debug 2021-01-21 02:30:38: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"LUMI"}'
- debug 2021-01-21 02:30:41: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
- debug 2021-01-21 02:30:41: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
- debug 2021-01-21 02:30:42: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
- debug 2021-01-21 02:30:42: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
- debug 2021-01-21 02:30:43: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"appVersion":35}' from endpoint 1 with groupID 0
- debug 2021-01-21 02:30:43: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":35}'
- debug 2021-01-21 02:30:44: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
- debug 2021-01-21 02:30:44: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":2}'
- debug 2021-01-21 02:30:45: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
- debug 2021-01-21 02:30:45: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
- debug 2021-01-21 02:30:46: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20201113"}' from endpoint 1 with groupID 0
- debug 2021-01-21 02:30:46: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20201113"}'
- debug 2021-01-21 02:30:47: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2019\u0000www."}' from endpoint 1 with groupID 0
[... cut ...]
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 1... = Direction: Server to Client
        ...1 .... = Disable Default Response: True
    Sequence Number: 5
    Command: Default Response (0x0b)
    Response to Command: 0x06
    Status: Unsupported Cluster (0xc3)
```

</details>

**Environment:**

- Sensor firmware version: _0.0.0_0035_
- Ebyte E18-2G4U04B (CC2531) firmware version: _zStack30x (20190425)_
- Zigbee2mqtt version: _1.17.0-dev (2ef2c54)_

Koenkk, 2021-01-21:
What if we bind the `aqaraOpple` cluster? Updated: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2079/files

Otnow, 2021-01-21:
> What if we bind the `aqaraOpple` cluster? Updated: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2079/files

Added `aqaraOpple` to `reporting.bind`, but the _illuminance parameter is still not visible_:

<details>
  <summary>Zigbee2mqtt log</summary>

```
- info  2021-01-22 01:27:03: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"2ef2c54","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":15,"extendedPanID":"0x00124b0014d9f126","panID":6754},"permit_join":false,"version":"1.17.0-dev"}'
- debug 2021-01-22 01:27:21: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"transaction":"lhu8n-1","value":true}'
- info  2021-01-22 01:27:21: Zigbee: allowing new devices to join.
- info  2021-01-22 01:27:21: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":true},"status":"ok","transaction":"lhu8n-1"}'
- info  2021-01-22 01:27:32: Device '0x04cf8cdf3c7fd37e' joined
- info  2021-01-22 01:27:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_joined"}'
- info  2021-01-22 01:27:32: Starting interview of '0x04cf8cdf3c7fd37e'
- info  2021-01-22 01:27:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e","status":"started"},"type":"device_interview"}'
- info  2021-01-22 01:27:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_connected"}'
- info  2021-01-22 01:27:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"pairing"}'
- debug 2021-01-22 01:27:32: Device '0x04cf8cdf3c7fd37e' announced itself
- info  2021-01-22 01:27:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_announce"}'
- info  2021-01-22 01:27:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_announced"}'
- debug 2021-01-22 01:27:35: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.motion.agl02"}' from endpoint 1 with groupID 0
- debug 2021-01-22 01:27:35: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"lumi.motion.agl02"}'
- debug 2021-01-22 01:27:36: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
- debug 2021-01-22 01:27:36: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"LUMI"}'
- debug 2021-01-22 01:27:39: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
- debug 2021-01-22 01:27:39: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
- debug 2021-01-22 01:27:40: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
- debug 2021-01-22 01:27:40: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
- debug 2021-01-22 01:27:41: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"appVersion":35}' from endpoint 1 with groupID 0
- debug 2021-01-22 01:27:41: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":35}'
- debug 2021-01-22 01:27:42: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
- debug 2021-01-22 01:27:42: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":2}'
- debug 2021-01-22 01:27:43: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
- debug 2021-01-22 01:27:43: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
- debug 2021-01-22 01:27:44: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20201113"}' from endpoint 1 with groupID 0
- debug 2021-01-22 01:27:44: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20201113"}'
- debug 2021-01-22 01:27:45: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2019\u0000www."}' from endpoint 1 with groupID 0
- debug 2021-01-22 01:27:45: No converter available for 'RTCGQ12LM' with cluster 'genBasic' and type 'readResponse' and data '{"swBuildId":"2019\u0000www."}'
[... cut ...]
- №261 - sensor reported attribute _0x0000 = 1_ (Occupied=True)
- №269 - sensor reported attribute _0x0000 = 0_ (Occupied=False)
- №281 - sensor reported attribute _0x0000 = 1_ (Occupied=True)
- №291 - sensor reported attribute _0x0000 = 0_ (Occupied=False)
- №299 - sensor reported attribute _0x0000 = 1_ (Occupied=True)
- №307 - sensor reported attribute _0x0000 = 0_ (Occupied=False)

**Environment:**

- Sensor firmware version: _0.0.0_0035_
- Ebyte E18-2G4U04B (CC2531) firmware version: _zStack30x (20190425)_
- Zigbee2mqtt version: _1.17.0-dev (2ef2c54)_

Koenkk, 2021-01-22:
TBH I'm currently out of ideas how to get the illuminance working. I think the magic is in the write of 0x00FF (https://github.com/Koenkk/zigbee2mqtt/issues/5706#issuecomment-761622061) but looking at your various Xiaomi gateway sniff the values is different every time (so don't know how to reconstruct it). I will merge the current MR so at least occupancy is supported for now.

Otnow, 2021-01-24:
> I think the magic is in the write of 0x00FF (#5706 (comment)) but looking at your various Xiaomi gateway sniff the values is different every time (so don't know how to reconstruct it).

Yes, I think so too. You have correctly noticed that each time the gateway writes the 0x00FF attribute with a different value. I tried to write the 0x00FF attribute with the value from the dump, but the illuminance did not appear, because there seems to be some special algorithm for its generation. I also wanted to check for a possible vendor lock by MAC address, but for some reason the value of the configuration parameter `ext_pan_id: [84, 239, 68, 16, 0, 2, 212, 160]` is not applied:

```
- info  2021-01-22 05:35:30: Starting Zigbee2MQTT version 1.17.0-dev (commit #2ef2c54)
- info  2021-01-22 05:35:30: Starting zigbee-herdsman (0.13.57)
- debug 2021-01-22 05:35:30: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[25],"extendedPanID":[84,239,68,16,0,2,212,160],"networkKey":"HIDDEN","panID":22441},"serialPort":{"path":"/dev/ttyACM0"}}'
- info  2021-01-22 05:35:34: zigbee-herdsman started
- info  2021-01-22 05:35:34: Coordinator firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"}'
- debug 2021-01-22 05:35:34: Zigbee network parameters: {"channel":25,"extendedPanID":"0x00124b0014d9f126","panID":22441}
```

What can be wrong?

> I will merge the current MR so at least occupancy is supported for now.

Ok, let it be as a temporary solution for now. Thank you.

> Where did you buy this sensor? Interested in buying one myself.

I bought this sensor on Aliexpress from [this seller](https://aliexpress.com/item/1005001889397376.html?&sku_id=12000018069572598) ([my purchase history](https://www.reddit.com/r/Aqara/comments/kid7vc/where_to_buy_t1_sensors/gh38kxz)), but judging by the [comments under my review of the sensor on Reddit](https://www.reddit.com/r/Aqara/comments/kuskin/quick_test_aqara_motion_sensor_t1_rtcgq12lm/giz69h9), it quickly ran out and now this seller is sending the old version (not T1). I think after some time this sensor should again appear on sale on Aliexpress and it is better to ask the seller in advance for the name of the model before buying, so as not to be mistaken.

Koenkk, 2021-01-24:
About the ext_pan_id, can you provide the herdsman debug log when starting with this? 

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Otnow, 2021-01-24:
> About the ext_pan_id, can you provide the herdsman debug log when starting with this?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

<details>
  <summary>Herdsman debug log</summary>

```
$ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.17.0-dev start
> node index.js

Zigbee2MQTT:debug 2021-01-24 21:23:48: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2021-01-24 21:23:48: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-01-24.21-23-45' filename: log.txt
Zigbee2MQTT:info  2021-01-24 21:23:48: Starting Zigbee2MQTT version 1.17.0-dev (commit #2ef2c54)
Zigbee2MQTT:info  2021-01-24 21:23:48: Starting zigbee-herdsman (0.13.57)
Zigbee2MQTT:debug 2021-01-24 21:23:48: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[25],"extendedPanID":[84,239,68,16,0,2,212,160],"networkKey":"HIDDEN","panID":22441},"serialPort":{"path":"/dev/ttyACM0"}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[33,114,79,72,173,142,187,196,249,106,231,0,5,6,211,252],"panID":22441,"extendedPanID":[84,239,68,16,0,2,212,160],"channelList":[25]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +20ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +473ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,121,7,28] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,121,7,28] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [121,7] - 28 +3ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1913} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +12ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +39ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +34ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,2,2,7,2,217,20,52,1,2,0,0,0,0,144] +14ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,2,2,7,2,217,20,52,1,2,0,0,0,0,144] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,2,2,7,2,217,20,52,1,2,0,0,0,0] - 144 +2ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425} +31ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +9ms
  zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2 +0ms
  zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack30x' ({"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}) +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +34ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74] +37ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +24ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +2ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +31ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +27ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74] +25ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +16ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +23ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +23ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":132,"len":4,"offset":0,"value":{"type":"Buffer","data":[0,0,0,2]}} +22ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,132,0,0,174] +22ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,97,8,0,4,0,0,0,2,105] +16ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 3 - 5 - 80 - [0,0,9,169,87,0,0,38,241,217,20,0,75,18,0,0,0,0,0,0,0,0,0,25] - 128 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extNwkInfo - {"shortaddr":0,"devstate":9,"panid":22441,"parentaddr":0,"extendedpanid":"0x00124b0014d9f126","parentextaddr":"0x0000000000000000","channel":25} +748ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
Zigbee2MQTT:debug 2021-01-24 21:23:53: Zigbee network parameters: {"channel":25,"extendedPanID":"0x00124b0014d9f126","panID":22441}
```

</details>

Koenkk, 2021-01-24:
Ah, this is a bug in the Zstack 3.0.x firmware (probably will not be fixed since this firmware is not maintained anymore by TI). 

Here you see Zigbee2MQTT checks the extended pan id, which matches yours:

```
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":8,"value":{"type":"Buffer","data":[84,239,68,16,0,2,212,160]}} +38ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'extendedPanID' is valid +39ms
```

However when z2m reads the extended pan id with another request it returns the `ieeeAddr` instead of the extended pan id.

```js
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extNwkInfo - {"shortaddr":0,"devstate":9,"panid":22441,"parentaddr":0,"extendedpanid":"0x00124b0014d9f126","parentextaddr":"0x0000000000000000","channel":25} +748ms
```

This probably is not the issue why the illuminance doesn't work (since other stuff works fine)

Otnow, 2021-02-27:
Experimentally, it was possible to find out that the value of the attribute 0x00FF, which the gateway transmits to the sensor during pairing, changes approximately once every 5 minutes:

```
21:20:33 - 30:53:24:44:22:15:63:24:12:55:14:34:29:22:84:10
21:21:20 - 30:53:24:44:22:15:63:24:12:55:14:34:29:22:84:10
21:22:18 - 30:53:24:44:22:15:63:24:12:55:14:34:29:22:84:10
21:23:17 - 30:53:24:44:22:15:63:24:12:55:14:34:29:22:84:10
21:24:16 - 30:53:24:44:22:15:63:24:12:55:14:34:29:22:84:10
21:25:16 - 30:53:24:44:22:15:63:24:12:55:14:34:29:22:84:10

21:26:16 - 95:47:57:88:77:34:10:69:61:74:79:54:40:78:18:18
21:27:17 - 95:47:57:88:77:34:10:69:61:74:79:54:40:78:18:18
21:28:15 - 95:47:57:88:77:34:10:69:61:74:79:54:40:78:18:18
21:29:17 - 95:47:57:88:77:34:10:69:61:74:79:54:40:78:18:18
21:30:16 - 95:47:57:88:77:34:10:69:61:74:79:54:40:78:18:18

21:31:16 - 13:41:03:54:26:81:18:82:21:26:24:94:14:78:81:83
21:32:15 - 13:41:03:54:26:81:18:82:21:26:24:94:14:78:81:83
21:33:48 - 13:41:03:54:26:81:18:82:21:26:24:94:14:78:81:83
21:34:19 - 13:41:03:54:26:81:18:82:21:26:24:94:14:78:81:83
21:35:15 - 13:41:03:54:26:81:18:82:21:26:24:94:14:78:81:83

21:36:24 - 18:23:20:31:56:54:56:74:94:08:16:32:42:00:96:53
21:37:17 - 18:23:20:31:56:54:56:74:94:08:16:32:42:00:96:53
21:38:15 - 18:23:20:31:56:54:56:74:94:08:16:32:42:00:96:53
21:39:52 - 18:23:20:31:56:54:56:74:94:08:16:32:42:00:96:53
21:40:21 - 18:23:20:31:56:54:56:74:94:08:16:32:42:00:96:53
21:41:17 - 18:23:20:31:56:54:56:74:94:08:16:32:42:00:96:53

21:42:16 - 97:89:23:89:20:68:35:37:87:57:99:95:19:14:44:15
21:43:15 - 97:89:23:89:20:68:35:37:87:57:99:95:19:14:44:15
```

[RTCGQ12LM_9.zip](https://github.com/Koenkk/zigbee2mqtt/files/6055597/RTCGQ12LM_9.zip) - decrypted plain text dump

Using the intercepted value of the 0x00FF attribute for 5 minutes, I repeated its transmission to the sensor in the test environment, but the sensor did not start transmitting the illuminance value.

**Environment:**

- Sensor firmware version: _0.0.0_0035_
- Ebyte E18-2G4U04B (CC2531) firmware version: _zStack30x (20190425)_
- Zigbee2mqtt version: _1.17.0-dev (2ef2c54)_

dmulcahey, 2021-11-17:
@Koenkk while working on supporting the E1 devices in ZHA I noticed something interesting. When we do the same thing that you did in the firmware to support the E1 devices (respond w/ Xiamoi's MFG code when pairing) this device acts differently. I haven't had a ton of time to dig into this much yet but I am getting reports now from cluster 0xfcc0 which isn't in the device signature.

Otnow, 2021-11-17:
> @Koenkk while working on supporting the E1 devices in ZHA I noticed something interesting. When we do the same thing that you did in the firmware to support the E1 devices (respond w/ Xiamoi's MFG [...]

Yes, thanks, I am planning to test the operation of this motion sensor with a modified coordinator firmware (respond w/ Xiamoi's MFG code when pairing) by analogy with the e1 sensors and with a high probability I hope to get the illuminance data.

dmulcahey, 2021-11-17:
> Yes, thanks, I am planning to test the operation of this motion sensor with a modified coordinator firmware (respond w/ Xiamoi's MFG code when pairing) by analogy with the e1 sensors and with a high [...]

From a quick test… the default occupancy sensing cluster functionality now appears broken. I see traffic on the mfg cluster. There are 2 reports that I have noticed so far. One looks like the typical Xiaomi byte string and the other looks like it’s the occupancy event. I’m not at the logs atm so I can’t provide specifics

Koenkk, 2021-11-17:
@dmulcahey interesting! (and somehow disappointing that Xiaomi doesn't follow the spec, I guess they just don't want their device to work with other gateways) 

@Otnow could you provide the z2m debug log when running with the new fw and triggering occupancy on the sensor?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

dmulcahey, 2021-11-17:
This isn't from Z2M but it is what the sensor sends when occupancy is triggered: 
Endpoint 1, Cluster Id: 0xfcc0 
ZCL request 0x000a: [[Attribute(attrid=274, value=<TypeValue type=uint32_t, value=65540>)]]

if you press the button on the device 1x you also get the Xiaomi bytes report:

Endpoint 1, Cluster Id: 0xfcc0 
ZCL request 0x000a: [[Attribute(attrid=247, value=<TypeValue type=LVBytes, value=b'\x01!?\x0c\x03(\x19\x04!\xa8C\x05!\x18\x00\x06$\x03\x00\x00\x00\x00\x08!#\x01\n!\x00\x00\x0c \x01d\x10\x00e!\x00\x00f \x03g \x01h!\xa8\x00'>)]]

Otnow, 2021-11-18:
> @dmulcahey interesting! (and somehow disappointing that Xiaomi doesn't follow the spec, I guess they just don't want their device to work with other gateways)
>
> @Otnow could you provide the z2m debug log when running with the new fw and triggering occupancy on the sensor?

Yes, unfortunately Aqara is trying to prevent its sensors from being used in alternative systems.

This motion sensor, when paired with a coordinator on a modified firmware, completely ceases to use standard clusters (_msOccupancySensing_ no longer gives occupancy data and is occupied all the time) and uses only its own (_aqaraOpple - 0xFCC0_).

Now, in fact, there is only a motion start event containing illuminance data - _cluster 0xFCC0: attribute 0x0112 (type: 0x23), manufacturerCode: 0x115f_ - starting point: _65536 = 0 Lux_:

```
Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'aqaraOpple', data '{"illuminance":65540}' from endpoint 1 with groupID 0
```

The occupancy timeout is now contained in parameter -  _cluster 0xFCC0: attribute 0x0102 (type: 0x20), manufacturerCode: 0x115f_ - default: _60 sec_:

 ```
Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'aqaraOpple', data '{"258":60}' from endpoint 1 with groupID 0
```

<details>
  <summary>Zigbee2mqtt debug log</summary>

```
- info  2021-11-18 01:39:48: Device '0x04cf8cdf3c7fd37e' joined
- info  2021-11-18 01:39:48: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_joined"}'
- info  2021-11-18 01:39:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_connected"}'
- info  2021-11-18 01:39:48: Starting interview of '0x04cf8cdf3c7fd37e'
- info  2021-11-18 01:39:48: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e","status":"started"},"type":"device_interview"}'
- info  2021-11-18 01:39:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"pairing"}'
- debug 2021-11-18 01:39:49: Device '0x04cf8cdf3c7fd37e' announced itself
- info  2021-11-18 01:39:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c7fd37e","ieee_address":"0x04cf8cdf3c7fd37e"},"type":"device_announce"}'
- info  2021-11-18 01:39:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x04cf8cdf3c7fd37e"},"type":"device_announced"}'
- debug 2021-11-18 01:39:54: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'genBasic', data '{"appVersion":35,"modelId":"lumi.motion.agl02"}' from endpoint 1 with groupID 0
- debug 2021-11-18 01:39:54: Skipping message, definition is undefined and still interviewing
- debug 2021-11-18 01:39:57: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,243,11,3,40,25,4,33,168,1,5,33,112,2,6,36,4,0,0,0,0,8,33,35,1,10,33,0,0,12,32,1,100,16,0,101,33,5,0,102,32,3,103,32,1,104,33,168,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
- debug 2021-11-18 01:39:57: Skipping message, definition is undefined and still interviewing
- debug 2021-11-18 01:40:01: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,227,11,3,40,25,4,33,168,1,5,33,112,2,6,36,6,0,0,0,0,8,33,35,1,10,33,0,0,12,32,1,100,16,0,101,33,5,0,102,32,3,103,32,1,104,33,168,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
- debug 2021-11-18 01:40:01: Skipping message, definition is undefined and still interviewing
- debug 2021-11-18 01:40:02: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
- debug 2021-11-18 01:40:02: Skipping message, definition is undefined and still interviewing
- debug 2021-11-18 01:40:03: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
- debug 2021-11-18 01:40:03: Skipping message, definition is undefined and still interviewing
- debug 2021-11-18 01:40:04: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'aqaraOpple', data '{"illuminance":65541}' from endpoint 1 with groupID 0
- debug 2021-11-18 01:40:04: Skipping message, definition is undefined and still interviewing
- debug 2021-11-18 01:40:04: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
- debug 2021-11-18 01:40:04: Skipping message, definition is undefined and still interviewing
- debug 2021-11-18 01:40:05: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
- debug 2021-11-18 01:40:05: Skipping message, definition is undefined and still interviewing
- debug 2021-11-18 01:40:06: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
- debug 2021-11-18 01:40:06: Skipping message, definition is undefined and still interviewing
[... cut ...]
- debug 2021-11-18 01:41:12: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'aqaraOpple', data '{"illuminance":65540}' from endpoint 1 with groupID 0
- debug 2021-11-18 01:41:12: No converter available for 'RTCGQ12LM' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"illuminance":65540}'
```

</details>

**Environment:**

- Sensor firmware version: _0.0.0_0035_
- Ebyte E18-2G4U04B (CC2531) firmware version: _[CC2530_CC2531_XIAOMI_E1_SOURCE_ROUTING](https://github.com/Koenkk/zigbee2mqtt/files/7468825/CC2530_CC2531_XIAOMI_E1_SOURCE_ROUTING.zip)_
- Zigbee2mqtt version: _1.22.0-master (945b72b)_

Koenkk, 2021-11-18:
@Otnow could you try with this external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const {calibrateAndPrecisionRoundOptions} = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    occupancy_illuminance: {
        cluster: 'aqaraOpple',
        type: ['attributeReport', 'readResponse'],
        options: [exposes.options.precision('illuminance'), exposes.options.calibration('illuminance', 'percentual'), 
            exposes.options.occupancy_timeout()],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('illuminance')) {
                // The occupancy sensor only sends a message when motion detected.
                // Therefore we need to publish the no_motion detected by ourselves.
                const timeout = options && options.hasOwnProperty('occupancy_timeout') ? options.occupancy_timeout : 90;

                // Stop existing timers because motion is detected and set a new one.
                globalStore.getValue(msg.endpoint, 'timers', []).forEach((t) => clearTimeout(t));
                globalStore.putValue(msg.endpoint, 'timers', []);

                if (timeout !== 0) {
                    const timer = setTimeout(() => {
                        publish({occupancy: false});
                    }, timeout * 1000);

                    globalStore.getValue(msg.endpoint, 'timers').push(timer);
                }

                const illuminance = msg.data['illuminance'] - 65536;
                return {occupancy: true, illuminance: calibrateAndPrecisionRoundOptions(illuminance, options, 'illuminance')}
            }
        },
    }
}

const definition = {
    zigbeeModel: ['lumi.motion.agl02'],
    model: 'RTCGQ12LM',
    vendor: 'Xiaomi',
    description: 'Aqara T1 human body movement and illuminance sensor',
    fromZigbee: [fzLocal.occupancy_illuminance, fz.aqara_opple_report],
    toZigbee: [],
    exposes: [e.illuminance().withUnit('lx').withDescription('Measured illuminance in lux'), e.occupancy(), e.battery(), e.battery_voltage()],
};

module.exports = definition;

```

See https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#instructions how to use it

Questions:
- Does battery/voltage work? Could take 1 day to show up
- I would expect that the sensor also sends illuminance values on an interval, could you catch such message? I hope it doesn't use the same message as on occupancy because the `occupancy: true` would be set when none is detected

> The occupancy timeout is now contained in parameter - cluster 0xFCC0: attribute 0x0102 (type: 0x20), manufacturerCode: 0x115f - default: 60 sec:
- I don't understand why the sensor does this when it doesn't send an occupancy false event itself.

dmulcahey, 2021-11-18:
> > The occupancy timeout is now contained in parameter - cluster 0xFCC0: attribute 0x0102 (type: 0x20), manufacturerCode: 0x115f - default: 60 sec:
> 
> * I don't understand why the sensor does this when it doesn't send an occupancy false event itself.

Maybe it's writeable that way software can use this to reset it? similar to on w/ timed off from the OnOff cluster? I don't understand why the entire device needs to function completely differently...

Given what we are seeing I am starting to go through the rest of the Aqara Zigbee 3.0 devices that I have managed to acquire... The Aqara TVOC device now has the Opple cluster as well when pairing w/ this change. Unlike the T1 motion sensor the device continues to function as it did before the change. So much for any form of consistency. The Opple cluster is reporting the Xiaomi data but I have no clue what any of it is for. 

It may make sense to go through all Aqara Zigbee 3.0 devices to ensure they function still before widely recommending this firmware change to support the E1 devices.

Koenkk, 2021-11-19:
> Maybe it's writeable that way software can use this to reset it? similar to on w/ timed off from the OnOff cluster? I don't understand why the entire device needs to function completely differently...

That would only make sense if the occupancy sensor also can produce an occupancy: false event (like e.g. the Hue occupancy sensors)

> It may make sense to go through all Aqara Zigbee 3.0 devices to ensure they function still before widely recommending this firmware change to support the E1 devices.

True, @z-x-f you have added support for many new Xiaomi devices, do you see any differences with the new fw?

dmulcahey, 2021-11-19:
> True, @z-x-f you have added support for many new Xiaomi devices, do you see any differences with the new fw?

You need to remove them and re-pair them to test just FYI.

z-x-f, 2021-11-19:
@Koenkk My smart outlet, smart plug and smart wall switch are no different with the new fw (after removing, restart z2m, and then re-pair).

Otnow, 2021-11-20:
> @Otnow could you try with this external converter:

I slightly modified the converter to use the _occupancy_timeout_ parameter:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const {calibrateAndPrecisionRoundOptions} = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    occupancy_illuminance: {
        cluster: 'aqaraOpple',
        type: ['attributeReport', 'readResponse'],
        options: [exposes.options.precision('illuminance'), exposes.options.calibration('illuminance', 'percentual')],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('illuminance')) {
                // The occupancy sensor only sends a message when motion detected.
                // Therefore we need to publish the no_motion detected by ourselves.
                const timeout = meta && meta.state && meta.state.hasOwnProperty('occupancy_timeout') ? meta.state.occupancy_timeout : 60;

                // Stop existing timers because motion is detected and set a new one.
                globalStore.getValue(msg.endpoint, 'timers', []).forEach((t) => clearTimeout(t));
                globalStore.putValue(msg.endpoint, 'timers', []);

                if (timeout !== 0) {
                    const timer = setTimeout(() => {
                        publish({occupancy: false});
                    }, timeout * 1000);

                    globalStore.getValue(msg.endpoint, 'timers').push(timer);
                }

                const illuminance = msg.data['illuminance'] - 65536;
                return {occupancy: true, illuminance: calibrateAndPrecisionRoundOptions(illuminance, options, 'illuminance')}
            }
        },
    },
    occupancy_timeout: {
        cluster: 'aqaraOpple',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty(0x0102)) {
                return {occupancy_timeout: msg.data[0x0102]};
            }
        },
    }
}

const tzLocal = {
    occupancy_timeout: {
        key: ['occupancy_timeout'],
        convertSet: async (entity, key, value, meta) => {
            value *= 1;
            await entity.write('aqaraOpple', {0x0102: {value: [value], type: 0x20}}, {manufacturerCode: 0x115f});
            return {state: {occupancy_timeout: value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('aqaraOpple', [0x0102], {manufacturerCode: 0x115f});
        },
    }
}

const definition = {
    zigbeeModel: ['lumi.motion.agl02'],
[... 163 lines cut ...]
- debug 2021-11-19 18:57:22: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,30,12,3,40,25,4,33,168,19,5,33,163,2,6,36,50,0,0,0,0,8,33,35,1,10,33,0,0,12,32,1,100,16,0,101,33,9,0,102,32,3,103,32,1,104,33,168,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
- info  2021-11-19 18:57:22: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"illuminance":15,"linkquality":76,"occupancy":false,"voltage":3102}'
- debug 2021-11-19 19:52:13: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,29,12,3,40,25,4,33,168,19,5,33,163,2,6,36,51,0,0,0,0,8,33,35,1,10,33,0,0,12,32,1,100,16,0,101,33,9,0,102,32,3,103,32,1,104,33,168,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
- info  2021-11-19 19:52:13: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"illuminance":15,"linkquality":78,"occupancy":false,"voltage":3101}'
- debug 2021-11-19 20:47:42: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,31,12,3,40,25,4,33,168,19,5,33,163,2,6,36,52,0,0,0,0,8,33,35,1,10,33,0,0,12,32,1,100,16,0,101,33,9,0,102,32,3,103,32,1,104,33,168,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
- info  2021-11-19 20:47:42: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"illuminance":15,"linkquality":70,"occupancy":false,"voltage":3103}'
- debug 2021-11-19 21:42:48: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,31,12,3,40,25,4,33,168,19,5,33,163,2,6,36,53,0,0,0,0,8,33,35,1,10,33,0,0,12,32,1,100,16,0,101,33,9,0,102,32,3,103,32,1,104,33,168,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
- info  2021-11-19 21:42:48: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"illuminance":15,"linkquality":70,"occupancy":false,"voltage":3103}'
- debug 2021-11-19 22:38:07: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,30,12,3,40,25,4,33,168,19,5,33,163,2,6,36,54,0,0,0,0,8,33,35,1,10,33,0,0,12,32,1,100,16,0,101,33,8,0,102,32,3,103,32,1,104,33,168,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
[... cut ...]
According to my observations in this service array of data, the _value # 39_ (if you count from 1) presumably contains the illumination value, since when, for the sake of experiment, I pointed the lamp at the sensor, this value increased significantly, which can be seen in the last log entry (_9 -> 242_). 

**Environment:**

- Sensor firmware version: _0.0.0_0035_
- Ebyte E18-2G4U04B (CC2531) firmware version: _[CC2530_CC2531_XIAOMI_E1_SOURCE_ROUTING](https://github.com/Koenkk/zigbee2mqtt/files/7468825/CC2530_CC2531_XIAOMI_E1_SOURCE_ROUTING.zip)_
- Zigbee2mqtt version: _1.22.0-master (945b72b)_

Koenkk, 2021-11-20:
> I left the motion sensor alone for a day and it turned out that about every 55 minutes it broadcasts this kind of service information:

Could you add `fz.xiaomi_switch_opple_basic` to `fromZigbee` and provide the debug logging again?

Otnow, 2021-11-21:
> Could you add `fz.xiaomi_switch_opple_basic` to `fromZigbee` and provide the debug logging again?

<details>
  <summary>Zigbee2mqtt debug log</summary>

```javascript
- debug 2021-11-21 01:56:10: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'aqaraOpple', data '{"illuminance":65547}' from endpoint 1 with groupID 0
- info  2021-11-21 01:56:10: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"illuminance":11,"linkquality":170,"occupancy":true,"occupancy_timeout":60}'
- info  2021-11-21 01:57:10: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"illuminance":11,"linkquality":170,"occupancy":false,"occupancy_timeout":60}'
- debug 2021-11-21 02:44:29: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'aqaraOpple', data '{"illuminance":65539}' from endpoint 1 with groupID 0
- info  2021-11-21 02:44:29: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"illuminance":3,"linkquality":157,"occupancy":true,"occupancy_timeout":60}'
- info  2021-11-21 02:45:29: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"illuminance":3,"linkquality":157,"occupancy":false,"occupancy_timeout":60}'
- debug 2021-11-21 02:54:06: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,25,12,3,40,25,4,33,168,19,5,33,200,2,6,36,19,0,0,0,0,8,33,35,1,10,33,0,0,12,32,1,100,16,0,101,33,2,0,102,32,3,103,32,1,104,33,168,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 1 with value 3097
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 4 with value 5032
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 5 with value 712
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown vtype=36, pos=16
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 6 with value null
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown vtype=19, pos=17
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 36 with value null
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown vtype=0, pos=18
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 19 with value null
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown vtype=0, pos=19
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 0 with value null
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown vtype=0, pos=20
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 0 with value null
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown vtype=0, pos=21
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 0 with value null
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown vtype=8, pos=22
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 0 with value null
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 8 with value 291
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 10 with value 0
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 12 with value 1
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 101 with value 2
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 102 with value 3
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 103 with value 1
- debug 2021-11-21 02:54:06: lumi.motion.agl02: unknown index 104 with value 168
- info  2021-11-21 02:54:06: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"illuminance":3,"linkquality":170,"occupancy":false,"occupancy_timeout":60,"state":"OFF","temperature":25,"voltage":3097}'
- debug 2021-11-21 03:49:05: Received Zigbee message from '0x04cf8cdf3c7fd37e', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,25,12,3,40,25,4,33,168,19,5,33,200,2,6,36,20,0,0,0,0,8,33,35,1,10,33,0,0,12,32,1,100,16,0,101,33,4,0,102,32,3,103,32,1,104,33,168,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown index 1 with value 3097
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown index 4 with value 5032
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown index 5 with value 712
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown vtype=36, pos=16
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown index 6 with value null
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown vtype=20, pos=17
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown index 36 with value null
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown vtype=0, pos=18
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown index 20 with value null
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown vtype=0, pos=19
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown index 0 with value null
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown vtype=0, pos=20
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown index 0 with value null
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown vtype=0, pos=21
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown index 0 with value null
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown vtype=8, pos=22
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown index 0 with value null
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown index 8 with value 291
- debug 2021-11-21 03:49:05: lumi.motion.agl02: unknown index 10 with value 0
[... cut ...]
- debug 2021-11-22 00:02:10: lumi.motion.agl02: unknown index 101 with value 9
- debug 2021-11-22 00:02:10: lumi.motion.agl02: unknown index 102 with value 3
- debug 2021-11-22 00:02:10: lumi.motion.agl02: unknown index 103 with value 1
- debug 2021-11-22 00:02:10: lumi.motion.agl02: unknown index 104 with value 168
- info  2021-11-22 00:02:10: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c7fd37e', payload '{"battery":100,"illuminance":7,"linkquality":113,"occupancy":false,"occupancy_timeout":60,"state":"OFF","temperature":25,"voltage":3093}'
```

</details>

Koenkk, 2021-11-22:
@Otnow does any of the logged `unknown index` values contain the illuminance?

Otnow, 2021-11-24:
> @Otnow does any of the logged `unknown index` values contain the illuminance?

I carried out an additional check of the operation of the motion sensor with the original gateway and I can say with confidence that `unknown index 101` contains the illuminance value in Lux.

z-x-f, 2021-11-24:
> > @Otnow does any of the logged `unknown index` values contain the illuminance?
> 
> I carried out an additional check of the operation of the motion sensor with the original gateway and I can say with confidence that `unknown index 101` contains the illuminance value in Lux.

Index 101 in a smart switch usually represents the state from endpoint 2.

Wh1terat, 2021-11-25:
[image]

From the aqara developer docs. Interesting that illuminance is shown twice, once for movement explicitly and again for movement *or* heartbeat.

I'm only seeing 0x0112 on motion. value - 0xffff seems to match what is seen in the aqara app for the decimal lux value.

dmulcahey, 2021-11-25:
@Wh1terat where did you get that document? Is that something you can share?

Wh1terat, 2021-11-25:
@dmulcahey 
It's from [https://developer.aqara.com/](https://developer.aqara.com/) - it's for the cloud api service ("aiot" not "miot") rather than zigbee of course but perhaps gives _some_ insight into the device operation.

Wh1terat, 2021-11-29:
Been looking at the custom lumi attrs:

```
Frame 32: 104 bytes on wire (832 bits), 104 bytes captured (832 bits) on interface -, id 0
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x7205
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x7205
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Manufacturer Specific (0xfcc0)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 14
ZigBee Cluster Library Frame, Mfr: Lumi United Techology, Ltd Shenzhen (0x115f), Command: Report Attributes, Seq: 244
    Frame Control Field: Profile-wide (0x1c)
    Manufacturer Code: Lumi United Techology, Ltd Shenzhen (0x115f)
    Sequence Number: 244
    Command: Report Attributes (0x0a)
    Attribute Field, Octets: 01:21:75:0c:03:28:19:04:21:a8:01:05:21:14:00:06:24:05:00:00:00:00:08:21:…
        Attribute: 0x00f7
        Data Type: Octet String (0x41)
        Octet String: 01:21:75:0c:03:28:19:04:21:a8:01:05:21:14:00:06:24:05:00:00:00:00:08:21:23:01:0a:21:00:00:0c:20:01:64:10:00:65:21:0f:00:66:20:03:67:20:01:68:21:a8:00
```

Wrote a little parser, just need to work them out. 
Battery is fairly obvious. 
I wonder if the tags are standard across all devices? Would make life easier.

```
Tag: (0x01) 1, Type: (0x21) uint16, Data: 3189
Tag: (0x03) 3, Type: (0x28) int8, Data: 25
Tag: (0x04) 4, Type: (0x21) uint16, Data: 424
Tag: (0x05) 5, Type: (0x21) uint16, Data: 20
Tag: (0x06) 6, Type: (0x24) uint40, Data: 5
Tag: (0x08) 8, Type: (0x21) uint16, Data: 291
Tag: (0x0a) 10, Type: (0x21) uint16, Data: 0
Tag: (0x0c) 12, Type: (0x20) uint8, Data: 1
Tag: (0x64) 100, Type: (0x10) bool, Data: False
Tag: (0x65) 101, Type: (0x21) uint16, Data: 15
Tag: (0x66) 102, Type: (0x20) uint8, Data: 3
Tag: (0x67) 103, Type: (0x20) uint8, Data: 1
Tag: (0x68) 104, Type: (0x21) uint16, Data: 168
```

Koenkk, 2021-11-29:
@Wh1terat tags are probably standard, you can find a converter here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/5f54ef94ac2a3d99a3424e64baec9af188ed6eb8/converters/fromZigbee.js#L4873

Wh1terat, 2021-11-29:
@Koenkk Agreed, it does look to be standard between ^ and what I'm seeing.
I've got a bucket load of Aqara T1 sensors and seeing more and more moving away from standard attrs.

If it is indeed standardised, might be worth trying to build a database of them and a little refactor?

Wh1terat, 2021-12-30:
Bad bot @github-actions

closed 2022-01-06 by Koenkk

Otnow, 2022-01-06:
[RTCGQ12LM converter changed](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3631) (_added illuminance support_) for correct operation on coordinator firmware with support for native Aqara logic (is now available in the _[dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)_, and will appear in the _master branch_ at the beginning of next month), therefore, _for normal operation of the sensor with a new converter, it is [necessary to update the coordinator's firmware](https://www.zigbee2mqtt.io/devices/RTCGQ12LM.html#adapter-firmware)_.

On the coordinator's firmware without support for the native Aqara logic, as a workaround, it is proposed to use the previous version of the converter added in the configuration settings via the [`external_converters`](https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters) option:

- RTCGQ12LM_basic_mode.js

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.motion.agl02'],
    model: 'RTCGQ12LM',
    vendor: 'Xiaomi',
    description: 'Aqara T1 human body movement and illuminance sensor (illuminance not supported for now)',
    fromZigbee: [fz.occupancy, fz.occupancy_timeout, fz.battery],
    toZigbee: [tz.occupancy_timeout],
    exposes: [e.occupancy(), exposes.numeric('occupancy_timeout', ea.ALL).withValueMin(2).withValueMax(65535).withUnit('s')
        .withDescription('Time in seconds till occupancy goes to false'), e.battery()],
    meta: {battery: {voltageToPercentage: '3V_2100'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'msOccupancySensing']);
        await reporting.occupancy(endpoint);
        await reporting.batteryVoltage(endpoint);
        await endpoint.read('msOccupancySensing', ['pirOToUDelay']);
    },
};

module.exports = definition;
```

Otnow, 2022-03-19:
[RTCGQ12LM converter modified](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4006) to solve [latency issue](https://github.com/Koenkk/zigbee2mqtt/issues/11843) when the timeout ends (`detection_interval`), occupancy is cleared even if in fact there is still movement.

To do this, option `occupancy_timeout` has been added, which allows you to set an arbitrary _delay_ after the end of `detection_interval`, so as not to clear the occupancy if motion is detected again during the _delay_:

`occupancy_timeout` = `detection_interval` + _delay_

_**Important**: default value for `occupancy_timeout` is set as `detection_interval` + 2 seconds, so you may need to adjust it for your needs as there will be an additional delay of 2 seconds after you update zigbee2mqtt (these changes is now available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html), and will appear in the master branch at the beginning of next month)._ 

Of the additional changes: added `no_occupancy_since` option and `temperature` expose.

In zigbee2mqtt web interface, `occupancy_timeout` and `no_occupancy_since` options is located on `Settings (specific)` tab of the sensor.