Koenkk/zigbee2mqtt #5711: All devices are unresponsive on network.

jonbrunt123, 2021-01-12:
# Bug Report

@Koenkk, I still haven't worked out how to tag things as issues!!

I have just added 7 Hue Bulbs to my setup, taking me to 29 devices, the majority of which are mains powered bulbs. 

I am now getting an error whenever I attempt an action on any device.

I also have a second CC2531 flashed as a router. Is it a case that I have too big a network for the CC2531? I have ordered a more powerful co-ordinator off Tindie, however am hoping to sort the network until that arrives.

## What happened

ERROR: Publish 'set' 'state' to 'Landing Light' failed: 'Error: Command 0x00178801024bc691/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))'

## What did you expect to happen

Device to be controlled

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2021-01-12:
Looks like it's hitting the limits, using the source routing firmware might give improvements; https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

jonbrunt123, 2021-01-12:
Would a upgraded stick improve things??

Koenkk, 2021-01-12:
Yes! Also make sure to connect the adapter through a usb extension cable (so that there is some distance between your computer and the adapter)

closed 2021-02-19 by github-actions[bot]