Koenkk/zigbee2mqtt #5715: Support for Niko Zigbee Smart Plug 552-80699

fluppie, 2021-01-12:
## Information about the device + link
https://www.niko.eu/en/products/niko-home-control/products-for-traditional-wiring/smart-plug-with-pin-earthing-zigbee-productmodel-niko-59574367-4d82-576a-8142-5de4ecdad7b1

```
Friendly name
    0x5c0272fffef5f81d
Device type
    Router
Zigbee model
    Smart plug Zigbee PE
Support status

    Unsupported
IEEE address
    0x5c0272fffef5f81d
Network address
    0x68F3
Firmware build date
    20200929
Firmware version
    2.03.01
Power source
Interview completed
    Yes
```

## data/database.db entry of the device
`{"id":8,"type":"Router","ieeeAddr":"0x5c0272fffef5f81d","nwkAddr":26867,"manufId":4703,"manufName":"Niko NV","powerSource":"Mains (single phase)","modelId":"Smart plug Zigbee PE","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,2820,2821,4096,64642],"outClusterList":[3,10,25],"clusters":{"genBasic":{"attributes":{"modelId":"Smart plug Zigbee PE","manufacturerName":"Niko NV","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20200929","swBuildId":"2.03.01"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20200929","swBuildId":"2.03.01","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1610456302556}`

I would think it would behave a lot like https://www.zigbee2mqtt.io/devices/170-33505.html
Could I configure some files somewhere to use the parsers of Niko 170-33505 ?

sjorge, 2021-01-12:
Been in the market for one, found one at my local supplier it will be here friday. I'll have a look this weekend but you can probably try adding the following to devices.js

```
    {
        zigbeeModel: ['Smart plug Zigbee PE'],
        model: '552-80699',
        vendor: 'Niko',
        description: 'Smart plug with earthing',
        fromZigbee: [fz.on_off, fz.electrical_measurement],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
            await reporting.activePower(endpoint);
            await reporting.rmsCurrent(endpoint);
            await reporting.rmsVoltage(endpoint);
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage()],
    },
```

fluppie, 2021-01-12:
OK cool, it's a good? starting point.
Partially working, on/Off works. But energy data is not coming through.

In the clusters tab (Z2M webinterface in HassOS) I see this:
`    0x5c0272fffef5f81d
        Endpoint 1
            Output clusters
                genIdentify
                genTime
                genOta
            Input clusters
                genBasic
                genIdentify
                genGroups
                genScenes
                genOnOff
                seMetering
                haElectricalMeasurement
                haDiagnostic
                touchlink
                heimanSpecificInfraRedRemote
        Endpoint 242
            Output clusters
                greenPower`

sjorge, 2021-01-12:
Try this
```
    {
        zigbeeModel: ['Smart plug Zigbee PE'],
        model: '552-80699',
        vendor: 'Niko',
        description: 'Smart plug with earthing pin',
        fromZigbee: [fz.on_off, fz.metering],
        toZigbee: [tz.on_off],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
            await reporting.onOff(endpoint);
            await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.instantaneousDemand(endpoint);
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage()],
    },
```

@Koenkk first manufacture specific cluster ID collision? Having a heimanSpecificInfraRedRemote cluster seems very incorrect.

fluppie, 2021-01-12:
Nope, loses the current and voltage sensor (if they exist on this device)
[image]
Is there a way to fetch raw data from the zigbee stick (Slaesh CC2652RB stick)?

sjorge, 2021-01-12:
Is the power consumtion value at least showing change when plugging in a load?

We could try combining both.

```
    {
        zigbeeModel: ['Smart plug Zigbee PE'],
        model: '552-80699',
        vendor: 'Niko',
        description: 'Smart plug with earthing pin',
        fromZigbee: [fz.on_off, fz.electrical_measurement],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
            //await reporting.activePower(endpoint);
            await reporting.rmsCurrent(endpoint);
            await reporting.rmsVoltage(endpoint);
            await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.instantaneousDemand(endpoint);
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage()],
    },
```

sjorge, 2021-01-12:
The docs seems to indeed hint that saMetering has the bits to figure out the power:
```
CURRENT_SUMMATION_DELIVERED is marked as supported for it.
```

haElectricalMeasurement seems to have ACTIVE_POWER, RMS_CURRENT and RMS_VOLTAGE

fluppie, 2021-01-12:
Also, this "readEletricalMeasurementMultiplierDivisors" looks like a typo for readEle**C**tricalMeasurementMultiplierDivisors but the typo is consistent so no real issue I guess.
The other config for devices.js will not work either.

I have my old CC2531 stick where I can flash the Zboss Sniffer firmware into? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

fluppie, 2021-01-12:
Don't know if it can be useful. Did a lot of on/off during the sniffing. I also have one Niko 230V Outlet (inbouwstopcontact) which is sending a lot of messages on the network.

[ZbossSniffer_NIKO.pcapng.txt](https://github.com/Koenkk/zigbee2mqtt/files/5805266/ZbossSniffer_NIKO.pcapng.txt)

sjorge, 2021-01-13:
Won't have time to look more at it today, but should have a device tomorrow too. Hope to have more info then.

sjorge, 2021-01-14:
I can confirm I get an UNSUPPORTED attribute on both seMetering -> instantaneousDemand and all the haElectricaMeasurement attirbutes. Will poke around in the docs publish on the za website as IIRC they said they were supported :(

sjorge, 2021-01-14:
Progress, looks like the document is a copy of the one for the in wall plug, and they don't seem to support the same things.

Manual poking around, I think I got power working but looks like voltage and current are not available at all.

sjorge, 2021-01-14:
```[root@amethyst /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters]# mosquitto_sub -t zigbee2mqtt/0x60a423fffef83be6
{"linkquality":117,"power":0,"state":"ON"}
{"linkquality":33,"power":2,"state":"ON"}
{"linkquality":33,"power":3,"state":"ON"}
{"linkquality":27,"power":7,"state":"ON"}
{"linkquality":24,"power":13,"state":"ON"}
```

Seems to be correct ish... bulb is rated for 11 watt, but my Ubisys S1 also reads it as 12-13 watt at 100%

fluppie, 2021-01-14:
I copy/pasted the code from your pull req and can confirm Active Power is also working for me.

    {
        zigbeeModel: ['Smart plug Zigbee PE'],
        model: 'Phil 552-80699',
        vendor: 'Niko',
        description: 'Smart plug with earthing pin',
        fromZigbee: [fz.on_off, fz.electrical_measurement],
        toZigbee: [tz.on_off, tz.power_on_behavior],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            // only activePower seems to be support, although compliance document states otherwise
            await endpoint.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor']);
            await reporting.activePower(endpoint);
        },
        exposes: [
            e.switch(), e.power(),
            exposes.enum('power_on_behavior', exposes.access.STATE_SET, ['off', 'previous', 'on'])
                .withDescription('Controls the behaviour when the device is powered on'),
        ],
    },

closed 2021-02-22 by github-actions[bot]