Koenkk/zigbee2mqtt #5716: Lidl Livarno Lux smart LED light strip 2.5m (light) HG06104A not works correctly (colors and temperatrure).

sdp74, 2021-01-12:
No one one is complaining about this device but I found strange things:

## What happened
Exposes tab in Zigbee2MQTT gui:
- Color changing does'not work (always white)
- Temperature color strang behavior (not correspond what expected: 1000k is cool, 3000k is warm ..)

## Debug info
Home Assistant OS 5.10
Supervisor Version 2020.12.7 

Zigbee2MQTT:info  2021-01-12 16:35:53: Starting Zigbee2MQTT version 1.17.0 (commit #1.17.0)
Zigbee2MQTT:info  2021-01-12 16:35:53: Starting zigbee-herdsman (0.13.46)
Zigbee2MQTT:info  2021-01-12 16:35:56: zigbee-herdsman started
Zigbee2MQTT:info  2021-01-12 16:35:56: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'

## What i did
It is well knowed that the HG06104A  is selled as fully Philips Hue compatible then  in device.js in zigbee-herdsman-converter I forced the HG06104 acting as a Philips 915005106701. After repairing the device seems working correctly (color and white temperaure)

{                                                                                                                          
        fingerprint: [{modelID: 'TS0505A', manufacturerName: '_TZ3000_riwp3k79'}, {manufacturerName: '_TZ3000_riwp3k79'}],     
        model: 'HG06104A',                                                                                                     
        vendor: 'Lidl',                                                                                                        
        description: 'Livarno Lux smart LED light strip 2.5m',                                                         
        **extend: preset.hue.light_onoff_brightness_colortemp_colorxy,**                                                           
    },   

Hope it will be usefull

Koenkk, 2021-01-12:
`extend: preset.hue.light_onoff_brightness_colortemp_colorxy,` or `extend: preset.light_onoff_brightness_colortemp_colorxy,` is exactly the same except the added hue power on converters (which are not used for these commands). Can you try changing it back?

sdp74, 2021-01-12:
Your question is not clear to me. 
Actually i changed the original code from `extend: preset.light_onoff_brightness_colortemp_colorxy`  to `extend: preset.hue.light_onoff_brightness_colortemp_colorxy`. Sure I can reverse the code to the original one or pull the original docker of zigbeem2qtt for hass.io: after that what do you suggest?

Curiuosly, or not,  last hour , I had the same problem with a bulb (HG06106C Lidl Livarno Lux E27 bulb RGB light). I had yet onother same bulb paired and perfeclty working. Firstly i tried the same approach (from `extend: preset.light_onoff_brightness_colortemp_colorxy`  to `extend: preset.hue.light_onoff_brightness_colortemp_colorxy`) but this time the problem (colors and temperature issues) was persistent. So I reversed the code in device.js to the original, and retried the pairing but this time nearly my raspi3 with CC2531 and magically the bulb was working as expected!! So my conclusion (not really sure about it) that maybe is something linked to the routing during the pairing procedure (I'm not an zigbee expert actually novice to this technology). I have mainly Lidl Smart Home devices (RGB starter kit + led strip) and 1 osram+ bulb, not more than that. I would try to re_pair (!?) the led strip nearest as possible to the CC2531. I let you know soon.

sdp74, 2021-01-14:
Moved the Raspi3+CC2531 near the ledstrip (HG06104A), original device.js, no issues: the rgb led strip acts as expected. It's the problem maybe a possible low link quality during the pairing? I dont't know, I've got no errors, but for me at the moment the issue i resolved.

closed 2021-01-14 by sdp74

MrHaroldA, 2023-10-21:
I have the same problem with a Conbee II stick, but no matter how close the LED strip controller is to the coordinator, the LED strip never listens to color changes.

Is there anything else I can try?

enyineer, 2025-02-08:
@sdp74 @Koenkk sorry for the necro, but I have exactly the same device. Just as mentioned in the PR description, I can't change the strips color and the temperature scale is way off. Any easy way to fix this / readdress the device?

enyineer, 2025-02-15:
Today I learned:

After installing the latest update onto the controllers, you have to physically restart them by removing power for a few seconds. After that Color und Temperature Settings work flawless.

Rhylthus, 2026-04-21:
> Today I learned:
> 
> After installing the latest update onto the controllers, you have to physically restart them by removing power for a few seconds. After that Color und Temperature Settings work flawless.
> 
> Edit: Using Zigbee2Mqtt with an SLZB-06 Coordinator on a Raspberry Pi 5.

I have the same issue. Could you please tell me how you update the latest update on the led controler ?

Using zigbee2mqtt l'attestation version with SLZB-06 coordinatrice (latest stable version)

Thanks