Koenkk/zigbee2mqtt #5717: Lidl Livarno Lux E27 bulb CCT (HG06492C) Colour Temp issue.

P3nkiln, 2021-01-12:
Hi, I've just added a Lidl Livarno Lux E27 bulb CCT (HG06492C) which is supported but I'm having a problem with the colour temp.

Here's the log extract:

INFO: Device '0x842e14fffe3b90c1' joined
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffe3b90c1","ieee_address":"0x842e14fffe3b90c1"},"type":"device_joined"}'
INFO: Starting interview of '0x842e14fffe3b90c1'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffe3b90c1","ieee_address":"0x842e14fffe3b90c1","status":"started"},"type":"device_interview"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x842e14fffe3b90c1"},"type":"device_connected"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x842e14fffe3b90c1"},"type":"pairing"}'
INFO: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe3b90c1/availability', payload 'online'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffe3b90c1","ieee_address":"0x842e14fffe3b90c1"},"type":"device_announce"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x842e14fffe3b90c1"},"type":"device_announced"}'
INFO: Successfully interviewed '0x842e14fffe3b90c1', device has successfully been paired
INFO: Device '0x842e14fffe3b90c1' is supported, identified as: Lidl Livarno Lux E27 bulb CCT (HG06492C)
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Livarno Lux E27 bulb CCT","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","name":"color_temp","property":"color_temp","type":"numeric","unit":"mired","value_max":500,"value_min":150},{"access":7,"description":"Color temperature after cold power on of this light","name":"color_temp_startup","presets":[{"description":"Restore previous color_temp on cold power on","name":"previous","value":65535}],"property":"color_temp_startup","type":"numeric","unit":"mired","value_max":500,"value_min":150}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"HG06492C","vendor":"Lidl"},"friendly_name":"0x842e14fffe3b90c1","ieee_address":"0x842e14fffe3b90c1","status":"successful","supported":true},"type":"device_interview"}'

INFO: Configuring '0x842e14fffe3b90c1'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Livarno Lux E27 bulb CCT","friendly_name":"0x842e14fffe3b90c1","model":"HG06492C","supported":true,"vendor":"Lidl"},"type":"pairing"}'
INFO: MQTT publish: topic 'homeassistant/light/0x842e14fffe3b90c1/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x842e14fffe3b90c1/availability"}],"brightness":true,"brightness_scale":254,"color_temp":true,"command_topic":"zigbee2mqtt/0x842e14fffe3b90c1/set","device":{"identifiers":["zigbee2mqtt_0x842e14fffe3b90c1"],"manufacturer":"Lidl","model":"Livarno Lux E27 bulb CCT (HG06492C)","name":"0x842e14fffe3b90c1","sw_version":"Zigbee2MQTT 1.17.0-dev"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"hs":false,"json_attributes_topic":"zigbee2mqtt/0x842e14fffe3b90c1","name":"0x842e14fffe3b90c1","schema":"json","state_topic":"zigbee2mqtt/0x842e14fffe3b90c1","unique_id":"0x842e14fffe3b90c1_light_zigbee2mqtt","xy":false}'
INFO: MQTT publish: topic 'homeassistant/sensor/0x842e14fffe3b90c1/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x842e14fffe3b90c1/availability"}],"device":{"identifiers":["zigbee2mqtt_0x842e14fffe3b90c1"],"manufacturer":"Lidl","model":"Livarno Lux E27 bulb CCT (HG06492C)","name":"0x842e14fffe3b90c1","sw_version":"Zigbee2MQTT 1.17.0-dev"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x842e14fffe3b90c1","name":"0x842e14fffe3b90c1 linkquality","state_topic":"zigbee2mqtt/0x842e14fffe3b90c1","unique_id":"0x842e14fffe3b90c1_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
ERROR: Failed to configure '0x842e14fffe3b90c1', attempt 1 (Error: Read 0x842e14fffe3b90c1/1 lightingColorCtrl(["colorCapabilities"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:185:23) at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:240:26) at Generator.next (<anonymous>) at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))
WARNING: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x842e14fffe3b90c1!

Koenkk, 2021-01-12:
Yes agree about the try/catch, I've also marked the logging as debug instead of warn to avoid confusion. 

Issue should be fixed now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-19 by github-actions[bot]

WhiteAss, 2023-02-20:
Problem is still there.
256 seems the warmest temperature 
500 the coolest
beneath 256 it is all cool.