Koenkk/zigbee2mqtt #5725: Weird behavior with group states

TebbeUbben, 2021-01-12:
## What happened
I am experiencing a weird behavior when it comes to updating the state of a ZigBee group: When I set the state of a group to OFF (so that all the lights turn off), the first state update of a single light in that group is not reflected. Further updates do have an effect on the group state.

## What did you expect to happen
The group state should also take the first update into account.

## How to reproduce it (minimal and precise)
1. Create a group and add some lights to.
2. Turn off all lights using the group endpoint:
[image]
3. Turn on a single light:
[image]
The group state remains OFF. I'd expect it to be ON.
4. Turn the light off again:
[image]
5. ...and on again:
[image]
Now the group state is updated as I'd expect it.

The screenshots are from Home Assistant, but the logs show that Zigbee2MQTT is in fact not publishing a new group state.

## Debug info
Zigbee2MQTT version: 1.17.0 commit: 07cdc9d
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2021-01-13:
Can you provide the debug log of this?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

TebbeUbben, 2021-01-13:
Here you are:

```
Attaching to smarthome_zigbee2mqtt_1
[36mzigbee2mqtt_1    |[0m [34mZigbee2MQTT:debug[39m 2021-01-13 21:40:19: Received MQTT message on 'zigbee2mqtt/ceiling/set' with data '{"state": "OFF"}'
[36mzigbee2mqtt_1    |[0m [34mZigbee2MQTT:debug[39m 2021-01-13 21:40:19: Publishing 'set' 'state' to 'ceiling'
[36mzigbee2mqtt_1    |[0m [32mZigbee2MQTT:info [39m 2021-01-13 21:40:19: MQTT publish: topic 'zigbee2mqtt/ceiling_1', payload '{"brightness":254,"color":{"x":0.459040904539081,"y":0.410413186446054},"color_temp":368,"linkquality":52,"state":"OFF","update":{"state":"idle"},"update_available":false}'
[36mzigbee2mqtt_1    |[0m [32mZigbee2MQTT:info [39m 2021-01-13 21:40:19: MQTT publish: topic 'zigbee2mqtt/ceiling', payload '{"brightness":254,"color":{"x":0.459040904539081,"y":0.410413186446054},"color_temp":368,"state":"OFF"}'
[36mzigbee2mqtt_1    |[0m [32mZigbee2MQTT:info [39m 2021-01-13 21:40:19: MQTT publish: topic 'zigbee2mqtt/ceiling_2', payload '{"brightness":254,"color":{"x":0.459040904539081,"y":0.410413186446054},"color_temp":368,"state":"OFF","update":{"state":"idle"},"update_available":false}'
[36mzigbee2mqtt_1    |[0m [32mZigbee2MQTT:info [39m 2021-01-13 21:40:19: MQTT publish: topic 'zigbee2mqtt/ceiling_3', payload '{"brightness":254,"color":{"x":0.459040904539081,"y":0.410413186446054},"color_temp":368,"state":"OFF","update":{"state":"idle"},"update_available":false}'
[36mzigbee2mqtt_1    |[0m [32mZigbee2MQTT:info [39m 2021-01-13 21:40:19: MQTT publish: topic 'zigbee2mqtt/ceiling_4', payload '{"brightness":254,"color":{"x":0.459040904539081,"y":0.410413186446054},"color_temp":368,"state":"OFF","update":{"state":"idle"},"update_available":false}'
[36mzigbee2mqtt_1    |[0m [34mZigbee2MQTT:debug[39m 2021-01-13 21:40:20: Received MQTT message on 'zigbee2mqtt/ceiling_1/set' with data '{"state": "ON"}'
[36mzigbee2mqtt_1    |[0m [34mZigbee2MQTT:debug[39m 2021-01-13 21:40:20: Publishing 'set' 'state' to 'ceiling_1'
[36mzigbee2mqtt_1    |[0m [32mZigbee2MQTT:info [39m 2021-01-13 21:40:20: MQTT publish: topic 'zigbee2mqtt/ceiling_1', payload '{"brightness":254,"color":{"x":0.459040904539081,"y":0.410413186446054},"color_temp":368,"linkquality":47,"state":"ON","update":{"state":"idle"},"update_available":false}'
[36mzigbee2mqtt_1    |[0m [34mZigbee2MQTT:debug[39m 2021-01-13 21:40:21: Received MQTT message on 'zigbee2mqtt/ceiling_1/set' with data '{"state": "OFF"}'
[36mzigbee2mqtt_1    |[0m [34mZigbee2MQTT:debug[39m 2021-01-13 21:40:21: Publishing 'set' 'state' to 'ceiling_1'
[36mzigbee2mqtt_1    |[0m [32mZigbee2MQTT:info [39m 2021-01-13 21:40:21: MQTT publish: topic 'zigbee2mqtt/ceiling', payload '{"brightness":254,"color":{"x":0.459040904539081,"y":0.410413186446054},"color_temp":368,"state":"OFF"}'
[36mzigbee2mqtt_1    |[0m [32mZigbee2MQTT:info [39m 2021-01-13 21:40:21: MQTT publish: topic 'zigbee2mqtt/ceiling_1', payload '{"brightness":254,"color":{"x":0.459040904539081,"y":0.410413186446054},"color_temp":368,"linkquality":47,"state":"OFF","update":{"state":"idle"},"update_available":false}'
[36mzigbee2mqtt_1    |[0m [34mZigbee2MQTT:debug[39m 2021-01-13 21:40:22: Received MQTT message on 'zigbee2mqtt/ceiling_1/set' with data '{"state": "ON"}'
[36mzigbee2mqtt_1    |[0m [34mZigbee2MQTT:debug[39m 2021-01-13 21:40:22: Publishing 'set' 'state' to 'ceiling_1'
[36mzigbee2mqtt_1    |[0m [32mZigbee2MQTT:info [39m 2021-01-13 21:40:22: MQTT publish: topic 'zigbee2mqtt/ceiling', payload '{"brightness":254,"color":{"x":0.459040904539081,"y":0.410413186446054},"color_temp":368,"state":"ON"}'
[36mzigbee2mqtt_1    |[0m [32mZigbee2MQTT:info [39m 2021-01-13 21:40:22: MQTT publish: topic 'zigbee2mqtt/ceiling_1', payload '{"brightness":254,"color":{"x":0.459040904539081,"y":0.410413186446054},"color_temp":368,"linkquality":47,"state":"ON","update":{"state":"idle"},"update_available":false}'
```

Koenkk, 2021-01-15:
Should be fixed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

TebbeUbben, 2021-01-15:
Thanks, it's working now. 👍

closed 2021-01-15 by TebbeUbben