Koenkk/zigbee2mqtt #5728: Random "device announced itself" messages

passy05, 2021-01-13:
Recently I am getting some random "device announced itself" messages for various devices:

debug 2021-01-13 07:54:13: Device 'Office Light 3' announced itself                                                                                                                      
info  2021-01-13 07:54:13: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Office Light 3","ieee_address":"0x000d6ffffedc6092"},"type":"device_announce"}'
info  2021-01-13 07:54:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Office Light 3"},"type":"device_announced"}'

I have not found any pattern (any type of devices but most of my devices are not affected, anytime).  I tried to rebind the affected devices but the issue came back.

The annoying effect for lights is one flash when the light is on or worse, my OSRAM lights are turning on in the middle of the night.

Version: 1.17.0.1, CC2652R with latest firmware

Today last affected device:
{"id":59,"type":"Router","ieeeAddr":"0x000d6ffffedc6092","nwkAddr":59214,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb GU10 W 400lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb GU10 W 400lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1610517300327}
{"id":67,"type":"Group","groupID":2,"members":[{"deviceIeeeAddr":"0x000d6ffffed64919","endpointID":1},{"deviceIeeeAddr":"0x000d6ffffedc6092","endpointID":1},{"deviceIeeeAddr":"0x000d6ffffedc6139","endpointID":1}],"meta":{}}

sjorge, 2021-01-13:
Is that a new bulb? I had a few that would randomly crash, I just swapped them for new ones at IKEA.

passy05, 2021-01-13:
They are not new, several months old and I cannot complain about them. 
I have a Hive power socket with 21 "device_announce" messages in the last past 24 hours.

sjorge, 2021-01-13:
My other idea would be unstable power to the bulb.

passy05, 2021-01-13:
Interesting idea but I have seen the same "device_announce" messages for battery operated devices like the Sonoff SNZB-02 and Philips HUE switch.

sjorge, 2021-01-13:
That is interesting, any WiFi nearby on a channel that is causing interference?

~ sjorge

passy05, 2021-01-13:
I have not changed anything with Wifi, my Raspberry Pi 4 is connected via LAN and the CC2652R 2m apart from my Pi.

closed 2021-02-20 by github-actions[bot]

Morphy99, 2021-02-21:
I'm having the same issue recently with a Eurotronic Spirit TRV falling off the network and repeatedly spamming device_anounced messages. Ive noticed other devices are occasionally dropping off and then doing a device anounce but they're still working OK.

Morphy99, 2021-02-22:
I think I just found the culprit. Another system I've been testing something on has a CC2530 in it and even though the serial connection to the RPi GPIO wasn't enabled it was still powered and must have been causing interference! I've disconnected it now and removed and rejoined the TRV again so keeping fingers crossed it stays online now.

thrahlf, 2021-11-16:
I have the same problem. Any updates?

GSzabados, 2021-11-17:
Oddly, I ran into this problem yesterday with an IKEA filament bulb, an older on.
It has been turned off physically and on again what caused the issue to start. The bulb is continuously annoucing itself once at least a minute and it does not stop.
The other weird thing is the LQI value meanwhile it goes from 30-40 to 90 and back to 30-40. Like it has two routes assigned to the coordinator what it uses and one might drops messages or just thinks it is a different coordinator. I cannot find the source behind it. I considered the power issue as well, but the power on behavior is not triggered. Might be just a faulty zigbee chip or the power supply in the bulb went bad.
I've had an IKEA plug which did the same as well. That one was triggering power on behavior too. But it did this only on the Z2M network. When joined a SmartThings Zigbee network all was just fine.

thrahlf, 2021-12-28:
I still have the problem. Even a new teach-in of the affected devices only works temporarily.  Any updates?

bastir85, 2022-01-01:
I see a similar behaviour. First "device annouced" shows up in the logs and than the light switches on / flickers. I have OSRAM and ikea light bulbs but see the behavior mainly with OSRAM ones.

commander55, 2022-01-08:
Same here. My Osram devices powers on and off randomly for a few seconds.

Zigbee2MQTT:info  2022-01-08 06:37:42: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"OSRAM_POWER_1","ieee_address":"0x7cb03eaa0a06d68b"},"type":"device_announce"}'
Zigbee2MQTT:info  2022-01-08 06:37:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"OSRAM_POWER_1"},"type":"device_announced"}'
Zigbee2MQTT:info  2022-01-08 06:37:43: MQTT publish: topic 'zigbee2mqtt/OSRAM_POWER_1', payload '{"linkquality":13,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-01-08 06:37:49: MQTT publish: topic 'zigbee2mqtt/OSRAM_POWER_1', payload '{"linkquality":10,"state":"OFF","update":{"state":"idle"},"update_available":false}'

commander55, 2022-01-08:
> Same here. My Osram devices powers on and off randomly for a few seconds.
>
> Zigbee2MQTT:info 2022-01-08 06:37:42: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload [...]

As the Osram smart plug still goes on and off after unpairing and resetting the whole Zigbee network I suspect a hardware issue.

GSzabados, 2022-01-08:
> suspect a hardware issue

Try to turn off the Zigbee network, or just reset the plug. I saw similar behaviour on IKEA plugs and bulbs and I wouldn't be surprised if it is really not a serious bug in Z2M or in the coordinator firmware.

My IKEA devices started to do similar things after power outages, or when the network was destabilised by some restarting devices. The on/off is a reaction to some messages which is kinda power cycles the device, I would think. That is probably the reason of the announcement messages as well.

mckochan, 2022-02-11:
I don't know why, but my OSRAM/Sylvania Lightify lights were powering on and off when bound to channel 25. When I switched to channel 15 (which is what I was using with ZHA prior to Z2Mqtt), the issue went away. It is odd because I should have had less interference on 25 with my WiFi.

commander55, 2022-02-12:
I also suspected Wifi interferences, but changing Zigbee channels did not fix it. In the end, in my case, it was a hardware problem. A few days later my CC2531 adapter died. Since I switched to a CC2652P adapter the random behavior of my Osram device stopped.

Orochica, 2022-02-14:
I have the same problem with: TRADFRI ON/OFF switch E1743.
I get random Device 'IkeaSwitch2' announced itself for the first press and the action will not register, then the second press of the button works.

GSzabados, 2022-02-14:
> I get random Device 'IkeaSwitch2' announced itself for the first press and the action will not register, then the second press of the button works.

I guess this is a bit of a different issue. Probably the button leaves the network, or the coordinator considers it as left and once the device sends a message, the coordinator publishes an announcement first.

Just a guess, do you have availability turned on? Does this device report unavailable? And when was the device last seen when you have these announcements?

KidA001, 2023-01-30:
@passy05 did you ever resolved this? I have the same `device_announced` issue where my light randomly turns on and I don't understand what's causing it.

jempo, 2024-09-18:
Hi All, i wrote because i've the same issue here with some tradfri bulbs. I don't have understand what can be the problem. randomly i see the device annouce only for some bulbs (all for this this type https://www.zigbee2mqtt.io/devices/E1524_E1810.html#ikea-e1524%252Fe1810) and i see that they switch on and in the logs i see the device_annouce payload. Anyone have an idea about the problem?

GSzabados, 2024-09-18:
> Hi All, i wrote because i've the same issue here with some tradfri bulbs. I don't have understand what can be the problem. randomly i see the device annouce only for some bulbs (all for this this [...]

@jempo, could you please write your question in your native language, we will use Google translate as I don't understand even a word about what you are trying to ask, as that device is a remote not a bulb.

jempo, 2024-09-18:
Sorry @GSzabados, i've update the link shared because i'm seeing another device in my network. I think it's the same in my language, but if you want and can help i wrote also in italian. thanks for your help
`Ciao a tutti, scrivo perchè ho lo stesso problema con alcune lampadine Trafdri (ikea). Non ho capito quale possa essere il problema. In modo casuale vedo che le lampadine effettuano un device_annouce e si riaccendono senza una causa apparente, e succede solo per le lampadine di questo tipo (https://www.zigbee2mqtt.io/devices/LED1949C5.html#ikea-led1949c5). Qualcuno ha idea del perchè succeda? In aggiunta: ho notato che succede quando un device risulta non raggiungibile (perchè spento oppure, inspiegabilmente, esce dalla rete).`

GSzabados, 2024-09-18:
@jempo, those bubls have a really old Zigbee chip and they have had some serious issues as well in the firmware's code. They do not like when power cycled. I've got rid of most of the IKEA light bulbs due to their issues. Things that you can do to mitigate it, is to update their firmware to the latest version, make sure that your Zigbee coordinator is running its latest version. Have some different kind of Zigbee repeaters around these bulbs and connect them through those to the Zigbee network.

Once they start to do their flashing and announcing messages, I usually re-paired them to the zigbee network, which stopped the issue, but power cycling can trigger it any time. That's why it is better to get rid of them.

jempo, 2024-09-18:
@GSzabados Many thanks for your answer!