Koenkk/zigbee2mqtt #5729: SNZB-02 Sonoff Temperatura Sensor paired but not return data

srtxema, 2021-01-13:
Zigbee2mqtt paired snzb-02 after press for 5 seconds but 2 hours after then it don´t return data yet

[image]
[image]

passy05, 2021-01-13:
Mine is working fine, I can see the following message in the log:
info  2021-01-13 09:47:33: MQTT publish: topic 'zigbee2mqtt/Sonoff Temperature', payload '{"battery":100,"humidity":66.13,"linkquality":18,"temperature":23.13,"voltage":3200}'

srtxema, 2021-01-13:
I use conbee broker. I have snzb-03 and 04 that works correctly but this not return data

sjorge, 2021-01-13:
> I use conbee broker. I have snzb-03 and 04 that works correctly but this not return data

What does the reporting tab show?

srtxema, 2021-01-13:
i have removed for 3 times and aften then i obtain data
[image]
thanks!

closed 2021-01-13 by srtxema

sjorge, 2021-01-13:
That looks correct, make sure to force remove them, they don't seem to want to leave. 
After a force remove they usually pair OK for me.

srtxema, 2021-01-13:
Thanks sjorge!