Koenkk/zigbee2mqtt #5730: Removing device already removed results in "does not exist, skipping publish" loop

Morphy99, 2021-01-13:
## What happened
Whilst trying to remove a battery powered device and Z2M got stuck in a loop. I think as it was sending so many messages it wasn't removed in time and then reported failed. Pressing remove again and then Force Remove resulted in the error.

## What did you expect to happen
Device removed successfully.
## How to reproduce it (minimal and precise)
Pressed remove device in front end and kept moving device to keep awake:
```
info  2021-01-13 09:05:30: Removing device '0xd0cf5efffeea35cf' (block: false, force: false)
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"rotate_right","battery":100,"brightness":0,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"","battery":100,"brightness":0,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf/action', payload 'rotate_right'
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"rotate_right","battery":100,"brightness":6,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"","battery":100,"brightness":6,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf/action', payload 'rotate_right'
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"rotate_right","battery":100,"brightness":12,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"","battery":100,"brightness":12,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf/action', payload 'rotate_right'
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"rotate_right","battery":100,"brightness":18,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"","battery":100,"brightness":18,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf/action', payload 'rotate_right'
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"rotate_right","battery":100,"brightness":24,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"","battery":100,"brightness":24,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:31: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf/action', payload 'rotate_right'
info  2021-01-13 09:05:32: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"rotate_right","battery":100,"brightness":30,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:32: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"","battery":100,"brightness":30,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:32: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf/action', payload 'rotate_right'
info  2021-01-13 09:05:32: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"rotate_right","battery":100,"brightness":36,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:32: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"","battery":100,"brightness":36,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:32: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf/action', payload 'rotate_right'
info  2021-01-13 09:05:32: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"rotate_right","battery":100,"brightness":42,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:32: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"","battery":100,"brightness":42,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:32: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf/action', payload 'rotate_right'
info  2021-01-13 09:05:32: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"rotate_right","battery":100,"brightness":48,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:32: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"","battery":100,"brightness":48,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:32: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf/action', payload 'rotate_right'
[...]
info  2021-01-13 09:05:40: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"rotate_right","battery":100,"brightness":252,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:40: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"","battery":100,"brightness":252,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:40: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf/action', payload 'rotate_right'
info  2021-01-13 09:05:40: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"rotate_right","battery":100,"brightness":252,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:40: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"","battery":100,"brightness":252,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:40: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf/action', payload 'rotate_right'
info  2021-01-13 09:05:40: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"rotate_right","battery":100,"brightness":252,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:40: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf', payload '{"action":"","battery":100,"brightness":252,"last_seen":"2021-01-13T09:05:31+00:00","level":0,"linkquality":78,"rate":70,"update":{"state":"idle"},"update_available":false}'
info  2021-01-13 09:05:40: MQTT publish: topic 'zb2mqtt/0xd0cf5efffeea35cf/action', payload 'rotate_right'
error 2021-01-13 09:05:40: Request 'zb2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device '0xd0cf5efffeea35cf' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)'
```
As it didn't remove successfully I pressed remove again then Force Remove:

```
[... cut ...]
error 2021-01-13 09:05:56: '0xd0cf5efffeea35cf' does not exist, skipping publish
error 2021-01-13 09:05:56: '0xd0cf5efffeea35cf' does not exist, skipping publish
error 2021-01-13 09:05:56: '0xd0cf5efffeea35cf' does not exist, skipping publish
error 2021-01-13 09:05:56: '0xd0cf5efffeea35cf' does not exist, skipping publish
```

At which point I killed Z2M and restarted.

Device is IKEA Tradfri ICTC-G-1 Wireless Dimmer

## Debug info
Zigbee2MQTT version: 1.17.0-dev commit: 09d436f
Adapter hardware: CC2530
Adapter firmware version: 20201128

Koenkk, 2021-01-13:
Looks the device didn't respond to the leave request: `error 2021-01-13 09:05:40: Request 'zb2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device '0xd0cf5efffeea35cf' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)'`

Can you provide a sniff of this? Please share the pacpng + network key here: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

closed 2021-02-20 by github-actions[bot]

Morphy99, 2021-02-21:
I've been unable to recreate this and get a sniff sorry! I guess it was a freak incident.