Koenkk/zigbee2mqtt #5736: SmartThings GP-WOU019BBDWG fails to repeat/extend the network

pedrolamas, 2021-01-14:
## What happened

I've recently replaced all my smart plugs with [SmartThings GP-WOU019BBDWG](https://www.zigbee2mqtt.io/devices/GP-WOU019BBDWG.html) and I have noticed that battery powered device connected to them when I previously had a few...

One of the devices I noticed this was an Aqara temperature sensor, and it just stopped sending data completely unless I brought it closer to one other device (say, a Phillips Hue lamp).

To better illustrate here's part of my network graph: there are 2 plugs here, one from IKEA and another a Smartthings one, they are placed side-by-side and you can see the Aqara temperature sensor connected to the IKEA.

[image]

If I disconnect the IKEA plug, I was expecting the Aqara to jump into the Smartthings plug, but that never happens and so I never again receive data from the sensor...

## What did you expect to happen

That any mains powered smart plug would work as a zigbee network extender/repeater.

## How to reproduce it (minimal and precise)

Just get one of these plugs and try to put a battery powered device next to it, the device will either connect via another point or fail to send any data at all.

## Debug info
Zigbee2MQTT version: 1.17.0-dev commit: a822165
Adapter hardware: zzh
Adapter firmware version: 20201113

pedrolamas, 2021-01-14:
I do wonder if this is somehow related to https://github.com/Koenkk/zigbee2mqtt/issues/4694?

sjorge, 2021-01-14:
It's actually a known issue for Xiaomi/Aqara/Maiji devices :( The reason I also stopped using them.
https://github.com/Koenkk/zigbee2mqtt/issues/3941#issuecomment-665013283

You're find some more issues if you look for them.

pedrolamas, 2021-01-14:
Oh, I wasn't aware of this, I actually thought the issue was with the Smartthings plugs, not with the Awards sensors...

Still it is surprising that they do work with the IKEA plugs!

sjorge, 2021-01-14:
The simple explanation is... they stick to the router they were paired through.
You should be able to long press the button on the sensor to force it to pick a new route, but that doesn't seem to work for m y MiJia light sensor for example.

pedrolamas, 2021-01-14:
I did try long pressing the button on the Aqara sensor but that didn't seem to make any different... I even tried removing them from the network and adding them again but to no avail, it would sill stop communicating via the Samsung SmartThings plug after a few minutes.

Interestingly, if I keep the network "open" to new devices (permit join), it does seem to carry on working...

closed 2021-01-15 by pedrolamas

MattWestb, 2021-01-16:
IKEA router is pretty Xiaomi friendly (both ZLL and TB3 ones) but i think Smartthings is also using silabs chips but its depends how they have cooking the firmware.
The IKEA outlet is having one large condensator behind / in the antenna lobe that is grounding some radio signals and taking some of the antenna performance away :-).
Xiaomi sensors can being very tricky getting paired and dont like some routers to joining (also IKEA ones) but its happening with different version of them (Aqara wether old can all and the new model dont like ZB3 routers) but then they is in they is working OK. For tricking them i power of the router that is there parent and pushing the button and they is moving to one good in the near (also that is refused to pairing thru).
Also it can being problem for them to rejoining the network if they is leaving and trying rejoining, but its depends of the security setted in the trust center.

atv2016, 2022-02-06:
is this still an issue? Should i buy this zigbee smart plug or is it a dud..

pedrolamas, 2022-02-06:
@atv2016 The smart plug is great, I have 5 of them in my place! The problem was with the Aqara devices that don't implement the ZigBee protocol correctly... because of that, I still have 2 or 3 IKEA smart plugs that I use solely as routers, but I am considering replacing the Aqara devices with Tuya that seem to not have this kind of issues!

atv2016, 2022-02-06:
Good to know. I'm just exploring some smart plugs now, i'll keep an eye on the aqara end devices if they route properly. Current(ly) investigating:
Samsung one
Salus SP-600
Sonoff (no monitoring)

I looked at the IKEA signal repeater but i don't want to loose a socket.

pedrolamas, 2022-02-06:
I've had a Sales SP600 (IIRC, I did the initial Zigbee2MQTT integration) and those are quite good, but not very easy to find as i believe they have discontinued them.

atv2016, 2022-02-09:
@pedrolamas i better buy a few ones then, i have them in my basket saved for later. so far i've rounded it up to those 2.