Koenkk/zigbee2mqtt #5738: Brightness move don't work on Lonsonho dimmer

hakaneriksson, 2021-01-14:
# Bug Report
## What happened
Nothing happens to brightness on Lonsonho QS-Zigbee-D02-TRIAC-2C-LN dimmer when sending Brightness move command. Setting brightness to a value works fine.

## What did you expect to happen
Start dimming down.

## How to reproduce it (minimal and precise)
Send message, in this case channel 1:
zigbee2mqtt/0x5c0272fffe16b144/l1/set
{ "brightness_move": -40 }

Herdsman debug log: https://pastebin.com/B2Lt89Pz

## Debug Info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608

Koenkk, 2021-01-15:
Likely the firmware of this device doesn't support it.

MaZZie, 2021-02-07:
Seems like this dimmer is almost not dimming at all. 6% is almost full on. 5% is off. 
Ik will test the output of this dimmer later

closed 2021-03-17 by github-actions[bot]

Ape, 2021-04-30:
I also have this issue with brightness move and step.

Peterpc58, 2021-07-05:
I also have this issue with brightness move and Transition.

Daniel-WW, 2021-12-21:
I also have this issue with the QS-Zigbee-D02-TRIAC-LN. This is the "1 gang smart dimmer", instead of the above called "2 gang smart dimmer"