Koenkk/zigbee2mqtt #5740: Support Ledvance SMART+ LED CLASSIC E27 RGBW V2

Toon-nooT, 2021-01-14:
## Information about the device + link
Zigbee modelname is CLA60 RGBW II Z3. Not to be confused with other model added in #5385
Sold in package displaying "SMART+ Lighting Ledvance classic e27 multicolor", without mention of Osram.
[https://zigbeealliance.org/zigbee_products/cla60-rgbw-ii-z3/](url)

## data/database.db entry of the device
`{"id":7,"type":"Router","ieeeAddr":"0xf0d1b8000014cd06","nwkAddr":58419,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Mains (single phase)","modelId":"CLA60 RGBW II Z3","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,2821,4096,64527],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"CLA60 RGBW II Z3","manufacturerName":"LEDVANCE","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"Apr  1 2019\u0000\u0000\u0000\u0000\u0000","swBuildId":"00103101"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"Apr  1 2019\u0000\u0000\u0000\u0000\u0000","swBuildId":"00103101","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1610639596101}`

## My Attempt, based on other model (and seems to work)
> {
	zigbeeModel: ['CLA60 RGBW II Z3'],
        model: 'AC16381',
        vendor: 'LEDVANCE',
        description: 'SMART+ LED CLASSIC E27 RGBW V2',
        extend: preset.ledvance.light_onoff_brightness_colortemp_colorhs,
        ota: ota.ledvance,
    },

sjorge, 2021-01-14:
That looks good to me, can you create a PR against zhc yourself? If not, I can do one.

Toon-nooT, 2021-01-15:
Hi Jorge,
sorry i do not know what how to do that. I'm confused between zhc and devices.js. Could you add it with the above info?

Thanks (en groeten van Gent)

sjorge, 2021-01-15:
[Done](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2085), devices.js lives in a separate repo from this one.

zigbee2mqtt: the coordinator to mqtt application
zigbee-herdsman: the coordinator framework
zigbee-herdsman-converters: the device support used by zigbee-herdman

For simple stuff you can usually just navigate to the file https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js hit the edit button (pen icon) 

[image]

Then just update the file in the browser and then filling in:
[image]

And hit the big green button.

Toon-nooT, 2021-01-15:
Thanks Jorge, i'll close the issue.

closed 2021-01-15 by Toon-nooT