Koenkk/zigbee2mqtt #5743: Support for dresden ekeltronik FLS-A lp

vavo01, 2021-01-14:
Hi,

zigbee2mqtt already  supports two dresden elektronik devices. I tried the not yet support FLS-A lp ([datasheet](https://www.dresden-elektronik.de/produkt/fls-a-lp.html?file=files/dresden-elektronik/content/downloads/datenblaetter/wireless%20light%20control/ProdInfo_FLS-A%20lp-2016_600078.pdf)). But have been only partly successful. 

my entry in devices.js
`
    {
        zigbeeModel: ['FLS-A2 (1-10V)'],
        model: 'FLS-A',
        vendor: 'Dresden Elektronik',
        description: 'ZigBee Light Link wireless electronic ballast',
        extend: generic.light_onoff_brightness,
    },`

database.db entry of the device
`{"id":18,"type":"Router","ieeeAddr":"0x00212effff032ed6","nwkAddr":32011,"manufId":4405,"manufName":"dresden elektronik\u0000","powerSource":"Mains (single phase)","modelId":"FLS-A2 (1-10V)\u0000","epList":[11],"endpoints":{"11":{"profId":49246,"epId":11,"devId":256,"inClusterList":[0,4096,4,3,5,6,8,64718],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"FLS-A2 (1-10V)\u0000","manufacturerName":"dresden elektronik\u0000","powerSource":1,"zclVersion":1,"appVersion":2,"stackVersion":3,"hwVersion":1,"dateCode":"20170626\u0000","swBuildId":"0214.201000E9\u0000"}},"genLevelCtrl":{"attributes":{"currentLevel":100}}},"binds":[],"meta":{}}},"appVersion":2,"stackVersion":3,"hwVersion":1,"dateCode":"20170626\u0000","swBuildId":"0214.201000E9\u0000","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1610642540490}`

Actually, I got that way all I need for my setup. I can turn on/off and change the brightness for all four channels in parallel.

But as you can see in the datasheet, FLS-A lp can also control the four different channels indenpendently if you put the dip switch2 in position d. This setup don't work for me now. And I have now idea how to resolve it. May you support me to evaluate the full setup for this device?

l-mb, 2022-09-07:
Hi @vavo01, did you ever succeed with this? I'm now faced with the same scenario, I want to integrate two of the FLS-A2 devices.

vavo01, 2022-09-07:
Hi @l-mb ,
Unfortunately not. I'm still only able to turn on/off the device and change the brightness. But controll the four channels independently is not possible for me.

l-mb, 2022-09-27:
I wonder if I could shamelessly ask for this to be reopened? I've got two of these and they're the last devices that I have to keep the deconz bridge/raspberry around for at this point :-)

Koenkk, 2022-09-28:
Can you provide the data/database.db entry of this device when first force removing the device via the z2m frontend, restarting z2m and repairing the device?

l-mb, 2022-09-28:
Because I couldn't reopen this one I had since opened the other issue. Sorry! Let me know where you want to continue.

Koenkk, 2022-09-29:
No problem, lets continue in https://github.com/Koenkk/zigbee2mqtt/issues/14225

closed 2022-09-29 by Koenkk