Koenkk/zigbee2mqtt #5745: How should it be configured in multi-network？

CaptainNeil, 2021-01-15:
## What happened
When I set the PAN ID to 0xFFFF, the coordinator automatically assigned a PAN ID. But when I restarted the coordinator, the coordinator's PAN ID changed again, causing me to need to repair the device.
## What did you expect to happen
I want the coordinator to automatically assign the PAN ID on the first boot and remember that it won't change on the next boot

## Note
I found some solutions online.I need the firmware source code and recompile it.
[https://www.ti2k.com/39806.html](https://www.ti2k.com/39806.html)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

CaptainNeil, 2021-01-15:
I found a solution to this problem.
Add the ZDNwkMgr defined symbols  when compiling firmware.
[image]
How to resolve PAN ID conflicts is described in the official documentation.

[image]

CaptainNeil, 2021-01-15:
> I found a solution to this problem.
> Add the ZDNwkMgr defined symbols when compiling firmware.
> [image]
> How to resolve PAN ID conflicts is described in the official documentation.
> 
> [image]

This method is only valid for CC2530/1, not for CC26X2，I need your help

CaptainNeil, 2021-01-15:
This method is only valid for CC2530/1, not for CC26X2，I need your help @Koenkk

Koenkk, 2021-01-16:
The `0xFFFF` panID is currently not allowed. You have to define a static panID. I've modified zigbee-herdsman to throw an error in this case.

CaptainNeil, 2021-01-18:
When the PAN ID is set to 0xFFFF, the coordinator will generate a random PAN ID, which effectively avoids the problem of PAN ID conflicts.Why disable 0xffff. @Koenkk

Koenkk, 2021-01-18:
Because of:

> When I set the PAN ID to 0xFFFF, the coordinator automatically assigned a PAN ID. But when I restarted the coordinator, the coordinator's PAN ID changed again, causing me to need to repair the device.

and this is not something that can be fixed easily

sjorge, 2021-01-18:
> Because of:
> 
> > When I set the PAN ID to 0xFFFF, the coordinator automatically assigned a PAN ID. But when I restarted the coordinator, the coordinator's PAN ID changed again, causing me to need to repair the device.
> 
> and this is not something that can be fixed easily

I guess we could go the GENERATE route like for network_key, to get a similar effect... were we pick one at random and write it back to the configuration file.

Of course we could still hit a duplicate though, and I don't think there is an easy way to check/catch that.

CaptainNeil, 2021-01-18:
> I guess we could go the GENERATE route like for network_key, to get a similar effect... were we pick one at random and write it back to the configuration file.
>
> Of course we could still hit a [...]

This is exactly what I want. Because we need to deploy multiple coordinators in our office building. Assigning a static PAN ID to each coordinator is not very realistic. And this can easily cause PAN ID conflicts.

CaptainNeil, 2021-01-18:
In addition, I recompiled the CC2531 firmware, adding ZDNwkMgr and NV_RESTORE predefined at compile time. When I set pan_id to 0xffff, the coordinator randomly generates a PAN ID. And the PAN ID will not be regenerated the next time ZigBee2MQTT is started.

However, this approach does not work with CC2652, which baffles me.
[image]

Koenkk, 2021-01-18:
@sjorge the `GENERATE` is indeed a good idea.

Implemented this now:

```yaml
advanced:
  pan_id: GENERATE
```

CaptainNeil, 2021-01-19:
Thanks for your modification, is there any way to get the PAN ID of the surrounding network? Remove the duplicated PAN ID when generating the new PAN ID.

Koenkk, 2021-01-19:
@CaptainNeil this information is not passed from the adapter to z2m, I think you can only get it by sniffing.

Assuming this can be closed now.

closed 2021-01-19 by Koenkk

CaptainNeil, 2021-01-19:
> @CaptainNeil this information is not passed from the adapter to z2m, I think you can only get it by sniffing.
> 
> Assuming this can be closed now.

Thank you very much