Koenkk/zigbee2mqtt #5746: 'MAC no ack' error causes a obvious delay for other MQTT publish

zijings, 2021-01-15:
## What happened
I try to publish five 'set' messages at the same time to five different devices. At the beginning, everything was fine, but after I powered off two of them, some of the power-on devices showed a obvious delay on reaction to 'set' message.
## What did you expect to happen
After powering off some of the five devices, the left devices act to 'set' message normally
## How to reproduce it (minimal and precise)
Just in what happened
## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC2538
Adapter firmware version: ’20200211‘
Log(attention to the time around the 'mac no ack' error may help):
Jan 15 03:14:24 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:24: Received MQTT message on 'z2m_1010_1/0x00158d00047eba9c/set' with data '{"state":"OFF"}'
Jan 15 03:14:24 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:24: Publishing 'set' 'state' to '0x00158d00047eba9c'
Jan 15 03:14:24 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:24: Received MQTT message on 'z2m_1010_1/0x842e14fffe94b694/set' with data '{"state":"OFF"}'
Jan 15 03:14:24 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:24: Publishing 'set' 'state' to '0x842e14fffe94b694'
Jan 15 03:14:24 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:24: Received MQTT message on 'z2m_1010_1/0x00158d00045f0ee8/set' with data '{"state":"OFF"}'
Jan 15 03:14:24 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:24: Publishing 'set' 'state' to '0x00158d00045f0ee8'
Jan 15 03:14:24 syc-Default-string npm[11664]: Zigbee2MQTT:info  2021-01-15 03:14:24: MQTT publish: topic 'z2m_1010_1/0x00158d00047eba9c', payload '{"consumption":0.17,"energy":0.17,"linkquality":78,"power":0,"state":"OFF","temperature":29}'
Jan 15 03:14:24 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:24: Received MQTT message on 'z2m_1010_1/0xa4c138a594a8f09d/set' with data '{"state":"OFF"}'
Jan 15 03:14:24 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:24: Received MQTT message on 'z2m_1010_1/0x00158d000450dc3e/set' with data '{"state":"OFF"}'
Jan 15 03:14:24 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:24: Publishing 'set' 'state' to '0xa4c138a594a8f09d'
Jan 15 03:14:24 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:24: Publishing 'set' 'state' to '0x00158d000450dc3e'
Jan 15 03:14:24 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:24: Received Zigbee message from '0x00158d00047eba9c', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440525,"onOff":0}' from endpoint 1 with groupID 0
Jan 15 03:14:24 syc-Default-string npm[11664]: Zigbee2MQTT:info  2021-01-15 03:14:24: MQTT publish: topic 'z2m_1010_1/0x00158d00047eba9c', payload '{"consumption":0.17,"energy":0.17,"linkquality":78,"power":0,"state":"OFF","temperature":29}'
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:error 2021-01-15 03:14:42: Publish 'set' 'state' to '0x842e14fffe94b694' failed: 'Error: Command 0x842e14fffe94b694/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC no ack' (233))'
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:42: Error: Command 0x842e14fffe94b694/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC no ack' (233))
Jan 15 03:14:42 syc-Default-string npm[11664]:     at ZStackAdapter.<anonymous> (/opt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:339:27)
Jan 15 03:14:42 syc-Default-string npm[11664]:     at Generator.next (<anonymous>)
Jan 15 03:14:42 syc-Default-string npm[11664]:     at fulfilled (/opt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
Jan 15 03:14:42 syc-Default-string npm[11664]:     at processTicksAndRejections (internal/process/task_queues.js:97:5)
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:info  2021-01-15 03:14:42: MQTT publish: topic 'z2m_1010_1/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x842e14fffe94b694' failed: 'Error: Command 0x842e14fffe94b694/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"0x842e14fffe94b694"},"type":"zigbee_publish_error"}'
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:error 2021-01-15 03:14:42: Publish 'set' 'state' to '0x00158d00045f0ee8' failed: 'Error: Command 0x00158d00045f0ee8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC no ack' (233))'
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:42: Error: Command 0x00158d00045f0ee8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC no ack' (233))
Jan 15 03:14:42 syc-Default-string npm[11664]:     at ZStackAdapter.<anonymous> (/opt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:339:27)
Jan 15 03:14:42 syc-Default-string npm[11664]:     at Generator.next (<anonymous>)
Jan 15 03:14:42 syc-Default-string npm[11664]:     at fulfilled (/opt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
Jan 15 03:14:42 syc-Default-string npm[11664]:     at processTicksAndRejections (internal/process/task_queues.js:97:5)
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:info  2021-01-15 03:14:42: MQTT publish: topic 'z2m_1010_1/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x00158d00045f0ee8' failed: 'Error: Command 0x00158d00045f0ee8/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"0x00158d00045f0ee8"},"type":"zigbee_publish_error"}'
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:info  2021-01-15 03:14:42: MQTT publish: topic 'z2m_1010_1/0x00158d000450dc3e', payload '{"brightness":100,"color_temp":153,"linkquality":78,"state":"OFF"}'
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:42: Received Zigbee message from '0x00158d000450dc3e', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440529,"onOff":0}' from endpoint 1 with groupID 0
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:info  2021-01-15 03:14:42: MQTT publish: topic 'z2m_1010_1/0x00158d000450dc3e', payload '{"brightness":100,"color_temp":153,"linkquality":78,"state":"OFF"}'
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:42: Received Zigbee message from '0xa4c138a594a8f09d', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:info  2021-01-15 03:14:42: MQTT publish: topic 'z2m_1010_1/0xa4c138a594a8f09d', payload '{"linkquality":0,"state":"OFF"}'
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:info  2021-01-15 03:14:42: MQTT publish: topic 'z2m_1010_1/0xa4c138a594a8f09d', payload '{"linkquality":0,"state":"OFF"}'
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:42: Received Zigbee message from '0x00158d000450dc3e', type 'attributeReport', cluster 'genOnOff', data '{"61440":4214016,"onOff":0}' from endpoint 1 with groupID 0
[... cut ...]
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:info  2021-01-15 03:14:42: MQTT publish: topic 'z2m_1010_1/0x00158d000450dc3e', payload '{"brightness":100,"color_temp":153,"linkquality":73,"state":"OFF"}'
Jan 15 03:14:42 syc-Default-string npm[11664]: Zigbee2MQTT:info  2021-01-15 03:14:42: MQTT publish: topic 'z2m_1010_1/0x00158d000450dc3e', payload '{"brightness":100,"color_temp":153,"linkquality":73,"state":"OFF"}'
Jan 15 03:14:24 syc-Default-string npm[11664]: Zigbee2MQTT:debug 2021-01-15 03:14:24: Received MQTT message on 'z2m_1010_1/0x00158d00047eba9c/set' with data '{"state":"OFF"}'

Koenkk, 2021-01-15:
MAC no ack usually indicates interference, try to check wifi interference first and position the CC2538 away from your computer using an USB extension cable. https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html

zijings, 2021-01-16:
hello, Koenkk！
    I am afraid you misunderstood the scene I have described，I don't mean that 'mac no ack' is the abnormal issue because they are the two power-off devices. It seems these 'mac no ack' devices cause a obvious delay to other devices' command when I just publish five 'set' messages together to five devices as the logs show. And this is why I said attention to the time around the 'mac no ack' error may help.

Koenkk, 2021-01-16:
@zijings the default concurrency for this adapter is 2. You can increase it by adding the following to your `configuration.yaml`:

```yaml
advanced:
  adapter_concurrent: 10
```

zijings, 2021-01-16:
I will try it, thanks a lot for your remarkable job!

closed 2021-01-16 by zijings