Koenkk/zigbee2mqtt #5749: Auto-install external converter packages

svrooij, 2021-01-15:
## Is your feature request related to a problem?

I see a lot of people struggle to get new devices to work (by switching to dev, getting file access to `devices.js`, changing it and keeping it during updates), and the **external converters** option is a great way to [add support](https://svrooij.io/2020/12/31/new-device-for-zigbee2mqtt/) while we wait to get the device support into the latest version. It also supports loading the converters from a npm module 👍 

## Describe the solution you'd like

But you still have to do `npm install name-of-external-converter-package` before starting, which can be a pain to do when you're running docker (or even harder when you're running the home assistant plugin).

I think it would be great if the code below could be modified to install the external converter module if the module isn't installed yet. That would help the less tech savvy users to get bleeding edge device support by just adding the following to the configuration.

```yml
external_converters:
  - herdsman-support-lidl-lights # or whatever name scheme would be used
  - @svrooij/custom-device-support # if you want to publish them to npm under a namespace
```

## Describe alternatives you've considered

- Editing the docker startup script (which would require a new build of the container)
- Editing the `package.json` (which also needs an image rebuild)

## Additional context

This feature request is just to explorer the possibilities and to see what you guys think. I don't have any issue just creating the converter in a local file in the data folder, but I feel this feature isn't used enough, since it's way easier to use and keep updated compared to editing the `devices.js` file.

Source code where this would fit best (I think):
https://github.com/Koenkk/zigbee2mqtt/blob/07cdc9d549a9db4aa94a259388241b71aec4e12d/lib/util/utils.js#L153-L163

svrooij, 2021-02-15:
@Koenkk what do you think about this suggestions?

Koenkk, 2021-02-15:
@svrooij I like the suggestion, are you able to implement this?

svrooij, 2021-02-16:
What direction do you like most?

- Changing the docker startup file to parse the config and then just always execute `npm install xxx` for each line.
- Start an extra proces with `npm install xxx` in the described file (and then just waiting 30 seconds before continue).
- using [runtime-npm-install](https://www.npmjs.com/package/runtime-npm-install) or similar to install them right from the program and continue immediately after installing (or when already installed).

Koenkk, 2021-02-16:
https://www.npmjs.com/package/runtime-npm-install looks the cleanest to me. Can this be added to https://github.com/Koenkk/zigbee2mqtt/blob/f1a473d506cf9c32397f6fdc17cc278d35c93c0a/index.js#L19 ? (execute before z2m starts. Settings are available there)

closed 2021-02-24 by github-actions[bot]