Koenkk/zigbee2mqtt #5760: Lidl strip wrong colors, and E27 low brightness

noohi, 2021-01-15:
Hi,

changing colors on a Lidl strip (HG06104A) seems like it's of a wrong x/y coordinate.
For example choosing in Z2M frontend a yellow color, gives it clearly a greenish one.
Also to align it with Hue strip, in a certain purplish color "x":0.233,"y":0.144
on a lidl strip had to set "x":0.363,"y":0.217

I noticed that hue_correction exists, but is it really that the difference from the norm per each bulb/strip of this type isn't static, each one has a different shift?
I haven't noticed that randomness, so the point being maybe I can pin-point the shift for you and it gets hardcoded?
If so please give me clear instructions what would be good data to go on.

One more potentially related case while on the subject, the E27 Lidl color bulb (HG06106C), only when pure color is selected, the maximum brightness is too dim, like 20% of the white luminance.
Looking at the reviews and the bulb potential, it should be normal (~800 lumens).

Adapter hardware: CC26X2RB, latest dev and firmware.

maximklein, 2021-01-17:
I‘ve got the same Problem with the E27 Bulb

travelography, 2021-01-22:
The color LEDs of the E27 bulbs do not shine brighter. Not even with a LIDL gateway.

closed 2021-03-01 by github-actions[bot]