Koenkk/zigbee2mqtt #5761: Add new device Smart Power Strip from "Müller Licht"

zfrank2601, 2021-01-15:
Plz add new device from "Müller Licht"
https://www.amazon.de/gp/product/B08N6GZ5GG/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

{
    zigbeeModel: ['Smart Socket'], 
    model: '404017', 
    vendor: 'Müller Licht', 
    description: 'Smart Power Strip', 
    extend: preset.switch,
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
    const endpoint = device.getEndpoint(11);
        },

database-entry:
{"id":28,"type":"Router","ieeeAddr":"0xec1bbdfffe9d86fe","nwkAddr":9149,"manufId":4635,"manufName":"MLI","powerSource":"Mains (single phase)","modelId":"Smart Socket","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Smart Socket","manufacturerName":"MLI","powerSource":1,"zclVersion":3,"appVersion":40,"stackVersion":0,"hwVersion":0,"dateCode":"NULL","swBuildId":"2.5.3_r48"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca1c067","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":300,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":40,"stackVersion":0,"hwVersion":0,"dateCode":"NULL","swBuildId":"2.5.3_r48","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1,"reporting":1},"lastSeen":1610725121335}

Koenkk, 2021-01-16:
With this config, do you get feedback when manually clicking the button? Asking because the `configure` is misssing the following:

```js
{
    zigbeeModel: ['Smart Socket'],
    model: '404017',
    vendor: 'Müller Licht',
    description: 'Smart Power Strip',
    extend: preset.switch,
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(11);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
},
```

zfrank2601, 2021-01-16:
ah ok.. I´ see.. I got no feedback with manually clicking the button.
your config is perfect, tested it, thx

pairmode is 5 x click the button

Koenkk, 2021-01-16:
Added, thanks!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-16 by Koenkk