Koenkk/zigbee2mqtt #5767: Aqara Opple 3 band

smulle48, 2021-01-16:
Hi

https://www.zigbee2mqtt.io/devices/WXCJKG13LM.html

This switch is by default set to 33% dimming. Is there a way to change this so it behave like the IKEA dimmer?

MattWestb, 2021-01-16:
Yep writing one new firmware for it and flashing it to the devices.

smulle48, 2021-01-17:
> Yep writing one new firmware for it and flashing it to the devices.

Damn.... :-(

Guess i'll have to stick to deCONZ as they got it :-(

MattWestb, 2021-01-17:
In deCONZ is not using the real commands from the switch lal ZLL (Zigbee Light Link) for sending command to light groupes that is working then the coordinator is offline. 
The Opple switch is working in HA mode and not in ZLL mode in deCONZ that is making its unusable for direct controlling light.
Phoscon is one pure software hack that is running on the host system and is "converting" the light  commands from the switch to custom things (ala HA automations). 
If you like doing it on the host system then do it as one automation in HA but its not working if you HA system is offline (the sam for Phoscon). Its also possible setting the mode to HA in ZHA but i have not trying it out.

One recommendation use HA switches for HA things and ZLL switches for working with lights for getting less problem and getting it working then you system is offline (broken SD-card software update and so on).

The bad thing its not easy finding good real light switches and i think one of the best is one danich that have building wall switches of IKEA remotes for getting it working well with  good integration and autonomy.

smulle48, 2021-01-17:
> In deCONZ is not using the real commands from the switch lal ZLL (Zigbee Light Link) for sending command to light groupes that is working then the coordinator is offline.
>
> The Opple switch is [...]

Ahh. That why :-) now it makes sense :-)

I also think this system is way better and easier to implement new switches, etc then deCONZ :-)

If I'm not mistaking, the IKEA is using ZLL?

MattWestb, 2021-01-17:
All IKEA and Philips HUE is the old one ZLL (Zigbee Light Link) and the newer or updated is Zigbee 3 with lighting link capability (backward compatible with ZLL) = is working if the coordinator is offline.

In deCONZ adding one new devices must being made in 5 to 10 places in the code and its not may devs that can do that OK. And in the last time with many new tuya devices is being added from our community devs and is working but its not implanted in Phoscon as its closed source and looks very bad for new and normal users.

Z2M and ZHA is having making it much better way then adding new devices (ZHA many devices dont need quirks at all for working OK) and also for no code experience user like my can adding devices easily if one similar device is have being added before and i like that. And normally you can getting implanted in some days in the code and committed so all users can using it and dont need waiting 2 months with luck in deCONZ for getting it working at all.

The community is for the moment leading the race then getting good support of our zigbee devices.

smulle48, 2021-01-23:
I can see that some switches are capable of using simulated_brightness

Is that not an option with this? I know I'll most certainly loose that function if the system is down, but I'll guess that's the same with the other switches

closed 2021-03-03 by github-actions[bot]