Koenkk/zigbee2mqtt #5776: Web ui stopped loading devices / group page

fwelvering, 2021-01-16:
## What happened

After adding my new Eurotronic  Spirit Zigbee wireless heater thermostat (SPZB0001) the devices and groups page aren't loading anymore.

## What did you expect to happen

The devices page loads as normal and groups page is still working

## How to reproduce it (minimal and precise)

I joined the Thermostat (devices and groups page works)
After changing an value in HASSIO the devices/groups page isn't loading anymore.

`
{
	"battery": 100,
	"current_heating_setpoint": 16,
	"error_status": 0,
	"eurotronic_error_status": 0,
	"eurotronic_host_flags": {
		"boost": false,
		"child_protection": false,
		"mirror_display": false,
		"window_open": false
	},
	"eurotronic_system_mode": 1,
	"local_temperature": 15.06,
	"occupied_heating_setpoint": 16,
	"pi_heating_demand": 32,
	"system_mode": "auto",
	"unoccupied_heating_setpoint": 16
}

## Debug info
Zigbee2MQTT version: 1.17.0.1
Adapter hardware: CC2652RB
Adapter firmware version: 20201026

Koenkk, 2021-01-16:
I believe this already fixed in the dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

fwelvering, 2021-01-17:
@Koenkk thnx for your reply, i will wait for the next release than.
When do you expect it to be released?

Thnx

Koenkk, 2021-01-17:
1 february

fwelvering, 2021-02-02:
The release fixed the issue.

Thnx!

closed 2021-02-02 by fwelvering