Koenkk/zigbee2mqtt #5778: Tradfri motion sensor binding issue (only off works)

mvogt1, 2021-01-16:
## What happened
The motion sensor does not power on the LED1650R5 GU10 Lamp after 
successful binding to  OnOff.

## What did you expect to happen
The occupancy event should power on the Lamp

## How to reproduce it (minimal and precise)
- bind the Motion Sesonsor (e1525) successfully with the Tradfri GU10 (1650R5)
- motion is detected in the log files
- The led does not power on

## What I found out:

Power off works!

If I power on the LED, in the GUI, after the motion event appears in the log, but before the timer would expire, the lamp is switched off, after 60 seconds. 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5824898/log.txt)

So I would say, that the power off part of the bind works, but not the power on.

## Debug info
Zigbee2MQTT version: 1.17.0-dev (commit #5050ca0),
 zigbee-herdsman (0.13.55)
Adapter hardware: , CC26X2 (zzh)
Adapter firmware version:CC26X2R1_20201026

MattWestb, 2021-01-16:
IKEA controlling devices can only being bonded to groups and not to devices.
Making one groupe and putting light(s) in it and then bind the controlling devices to the group and its working. Depends of the firmware version it can being needy unbind the coordinator for getting it to working (see the docks).

mvogt1, 2021-01-16:
I can confirm that creating a group with the IKEA lamp, and then binding a 5-Button switch (Tradfri e1524) to it, works.
(dimming, on/off)
But the IKEA Motion sensor does not work, in a group setup.
Originally my setup / testing was with a group, but I made the report without it, because I thought its a simpler case.

In a group setup the behaviour is the same: 

-  The IKEA switch works, the IKEA motion sensor does not, both bind to the same group.
-  The motion sensor still powers off the lamp, after 60 seconds, but does not power on.

>Depends of the firmware version it can being needy unbind the coordinator for getting it to working (see the docks).
I have unbind the motion sensor from the default group, is this the "coordinator"?

jum, 2021-01-16:
I have a motion sensor that turns multiple IKEA GU10 lamps so it does work.

MattWestb, 2021-01-16:
I was upgrading my "IKEA Billy EZSP" yesterday and was by mistake deleting the NVR in the codiantor and loosing all link tokens to my zigbee 3 devises and was doing one compleet rejoining of my ZHA system and both the old and new IKEA motion sensor was gladly being bonded to one groupe and sending commands to it, as they always have being doing.

I can only thinking you is not walking the sensor up before sending the (un)binding commands and its sleeping and cant receiving it.

Setting up debug and looking in the log for the commands the motion sensor is sending and witch groupe they is being sent to (I think in Z2M is 0 (0x0000) if not being un bonded from the coordinator.

Z2M is zigbee as ZHA is and the devices is working the same way and is too unintelligent for doing any thing else than what there is made for to do.

mvogt1, 2021-01-17:
Today I reset my setup. 

- GU10 reset by switching it 6 times off/on, flicker confirm
- Motion/Switch by pressing button 4 times, until red light 
- stop zigbee2mqtt and deleted all yaml and db file

Then I used the WEB GUI only to add a group and renamed the devices.
The un/binding was done over the command line and to check if it was successfull,
I waited until the WEB Gui reports "Success" with a green popup window.

Then I found a second TRADFRI motion sensor, both the older version and did the test with both
Result:

- Both Motion sensor does not power on the lamp
- Tradfri switch works as expected

The only message which looks suspicious, for me, is during interview:

Zigbee2MQTT:debug 2021-01-17 14:46:25: Received Zigbee message from '0x000b57fffe958c52', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-01-17 14:46:25: No converter available for 'E1525/E1745' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}

Below the the sequence when I switch of the lamp and activate the motion1+2 sensor. Lamp is not powered up.

GU10 is on

Received Zigbee message from 'SSchalter1', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 1
info MQTT publish: topic 'zigbee2mqtt/SSchalter1', payload '{"action":"toggle","linkquality":72,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/SSchalter1', payload '{"action":"","linkquality":72,"update":{"state":"idle"},"update_available":false}'

GU10 is off

info MQTT publish: topic 'zigbee2mqtt/SSchalter1/action', payload 'toggle'
debug Received Zigbee message from 'SMotion1', type 'commandOnWithTimedOff', cluster 'genOnOff', data '{"ctrlbits":0,"offwaittime":0,"ontime":600}' from endpoint 1 with groupID 15644
info MQTT publish: topic 'zigbee2mqtt/SMotion1', payload '{"linkquality":159,"occupancy":true,"update":{"state":"idle"},"update_available":false}'
debug Received Zigbee message from 'SMotion2', type 'commandOnWithTimedOff', cluster 'genOnOff', data '{"ctrlbits":1,"offwaittime":0,"ontime":600}' from endpoint 1 with groupID 51490
debug Received Zigbee message from 'SMotion2', type 'commandOnWithTimedOff', cluster 'genOnOff', data '{"ctrlbits":1,"offwaittime":0,"ontime":600}' from endpoint 1 with groupID 51490
debug Received Zigbee message from 'SMotion1', type 'commandOnWithTimedOff', cluster 'genOnOff', data '{"ctrlbits":0,"offwaittime":0,"ontime":600}' from endpoint 1 with groupID 15644
info MQTT publish: topic 'zigbee2mqtt/SMotion1', payload '{"linkquality":153,"occupancy":true,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/SMotion1', payload '{"linkquality":153,"occupancy":false,"update":{"state":"idle"},"update_available":false}'

GU10 is still off

MattWestb, 2021-01-17:
The red LED on the front is one indicator that is being used then paring the device and not indicating motion is triggered.

First one is one toggle command from one remote to groupe 1.
```
Received Zigbee message from 'SSchalter1', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 1
```

What is the command received saying ?
```
debug Received Zigbee message from 'SMotion2', type 'commandOnWithTimedOff', cluster 'genOnOff', data '{"ctrlbits":1,"offwaittime":0,"ontime":600}' from endpoint 1 with groupID 51490
debug Received Zigbee message from 'SMotion1', type 'commandOnWithTimedOff', cluster 'genOnOff', data '{"ctrlbits":0,"offwaittime":0,"ontime":600}' from endpoint 1 with groupID 15644
```
The command is addressed to 0x006 (On/Off cluster) with command on with off time and with time is 600 of 1/10 seconds =  1 minute = one old motion sensor. And the command is sent to groupe ???

If you having the light in the same group as the motion sensor is bonded to (as in the log abow) and sending the commands its very likely you have setting the motion sensor to not triggering then its light around them (its the `"ctrlbits":X` in the log that is showing if its light or dark).

MattWestb, 2021-01-17:
If the problem is the last one (not the group binding) this can helping: https://www.ikea.com/at/de/customer-service/product-support/smart-lighting/bewegungsmelder-pubc0a7c571

MattWestb, 2021-01-17:
One trick with the old motion sensor that Z2M is not writing is that its one zigbee light link device (the new is zigbee 3) and is having one random groupe after reset. Instead of redoing the binding make one group in Z2M with the number the motion sensor is sending its commands and putting the light in it and you dont need redoing the binding in T2M. And if doing unbinding then you must using the random group its sending its commands to (51490/15644) or the unbinding is not working.

mvogt1, 2021-01-17:
Yes, that's it!
I created a group for id 51490 and put SLampe1 in:

mosquitto_pub [...] -t zigbee2mqtt/bridge/request/group/add -m '{"friendly_name": "test_51490", "id": "51490"}'
mosquitto_pub [...] -t zigbee2mqtt/bridge/request/device/bind -m '{"from": "SLampe1", "to": "test_51490"}'

this worked, the motion sensor powers on and off the lamp! 
Many thanks for your help!

MattWestb, 2021-01-17:
The light you dont need binding only adding it to the group and its working :-))

flatsiedatsie, 2021-01-30:
For my smart home controller I would just like to get the state of the motion sensor. I'll take care of it turning things on and off using rules later.

Is this possible? Or does this device only work with this `groups` funtionality?

I know that with the Conbee2 stick it's possible to have this motion sensor be handled as a stand-alone device.

With my CC2532 stick I am able to pair with the motion sensor, and see it through the exposure feature.

```
{
  "date_code": "20190308",
  "definition": {
    "description": "TRADFRI motion sensor",
    "exposes": [
      {
        "access": 1,
        "description": "Remaining battery in %",
        "name": "battery",
        "property": "battery",
        "type": "numeric",
        "unit": "%",
        "value_max": 100,
        "value_min": 0
      },
      {
        "access": 1,
        "description": "Indicates whether the device detected occupancy",
        "name": "occupancy",
        "property": "occupancy",
        "type": "binary",
        "value_off": false,
        "value_on": true
      },
      {
        "access": 1,
        "name": "requested_brightness_level",
        "property": "requested_brightness_level",
        "type": "numeric",
        "value_max": 254,
        "value_min": 76
      },
      {
        "access": 1,
        "name": "requested_brightness_percent",
        "property": "requested_brightness_percent",
        "type": "numeric",
        "value_max": 100,
        "value_min": 30
      },
      {
        "access": 1,
        "description": "Link quality (signal strength)",
        "name": "linkquality",
        "property": "linkquality",
        "type": "numeric",
        "unit": "lqi",
        "value_max": 255,
        "value_min": 0
      }
    ],
    "model": "E1525/E1745",
    "vendor": "IKEA"
  },
```

Once in a while I receive messages that say:
`{"battery":74,"linkquality":73,"update":{"state":"idle"},"update_available":false}`

But I never receive any messages that indicates motion has been detected?

mvogt1, 2021-01-30:
flatsiedatsie <notifications@github.com> schrieb am Sa., 30. Jan. 2021,
22:22:

> For my smart home controller I would just like to get the state of the
> motion sensor. I'll take care of it turning things on and off using rules
> later.
>

With AppDaemon I was able to trigger on the "occupancy : true" event with a
script, then I tried it with the new Blueprint feature, but both results
didn't convinced me, it was too slow for my case.
In your case, you should see the occupancy events in the log if you switch
on debugging or use the WebUI, and look the states of the Device this
should work. Then use the devicenames in HA in an automation.

>

flatsiedatsie, 2021-01-30:
Thanks. I'm not sure I follow though. I don't use HomeAssistant, I use the [WebThings Gateway](https://www.webthings.io), and am developing a Zigbee2MQTT addon for it.

Some confusion on my part:
- Appdeamon looks interesting. It's for Python, while Zigbee2MQTT is javascript?
- Not sure what the blueprint feature you're talking about is. Is that a feature of appdeamon?
- Is it true that not seeing this data from the motion sensor is 'normal'? Do you mean that if I set Zigbee2MQTT to a debugging mode it will be possible to see more messages that are flying around the network? If so, how do I enable debugging? I currently start Zigbee2MQTT by running a `node index.js` command.

mvogt1, 2021-01-30:
My fault, I thought it was a GUI question for HA. But as I see from my own log file above, the occupancy event is even i
reported on the info log level. 
I tested once a command line example and subscribed to all zigbee2mqtt messages on the mqtt bus, there I had never a problem that the occupancy state wasn't reported. 
So from my (HA setup) and limited experience this always worked in HA .

Maybe someone with more knowledge of zigbee2mqtt has an idea.
Why the occupancy event is not reported? Must this be setup somehow? 

Blueprint is a newer tempkate like feature in HA, for a complete scripting Appdaemon can be used.

flatsiedatsie, 2021-01-30:
> But as I see from my own log file above, the occupancy event is even i
reported on the info log level.

Glad to hear it. So in theory I should be able to use this device, and won't need to bind it to a group or device, correct?

flatsiedatsie, 2021-02-06:
(I got it working, my issue was with the CC2531 Zigbee Stick I was using, which was running the default Sonoff firmware)

closed 2021-03-16 by github-actions[bot]