Koenkk/zigbee2mqtt #5779: HS2WD-E via Home Assistant ?

kevincaradant, 2021-01-16:
Hi

I bought a HS2WD-E model (siren) but I don't understand how can we trigger the alarm ON / OFF on Home Assistant ? 

Is it a easy way to do it ? 

I saw only LQI and Battery level on it. But I didn't see nothing around actions ect like Xiaomi switches for example :)

Thanks 

Regards

Koenkk, 2021-01-16:
Add the warning to exposes, can you check if it works correctly?

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

kevincaradant, 2021-01-16:
OK I will look it in few hours :)

Thanks

BTW: I was surprised that you have nothing about melodies and the volume of this siren.

I found a thread and at the end of this one, people seems have the choise about melody and almost the volume of the siren.

Why we haven't this choice with Z2M ? :)

Thanks

Koenkk, 2021-01-17:
Good, can this issue be closed? The sound effect is controlled through the `mode`.

kevincaradant, 2021-01-17:
Mhh I thought we will close this issue when the sound level will be fixed ? 

[image]

Low level here is very very strong as I said yesterday, pretty sure it doesn't work ;)

Koenkk, 2021-01-17:
@kevincaradant I expect the sound level issue is in the firmware of the device (since the commands we send are according to the Zigbee spec) I think it doesn't support different sound levels, only way to be sure is to check it with the original gateway.

kevincaradant, 2021-01-17:
oh alright. If we can't do nothing because it's a firmware issue ( very annoying in my case to setup my use cases :P) 
By consequence, I close this issue :)

Thanks

closed 2021-01-17 by kevincaradant

kevincaradant, 2021-01-17:
One last thing @Koenkk 

Why I can't have the state ON / OFF on HA ? 

I removed the device from Z2M and pair it again with the latest version to be sure to have the state of the siren.

I have nothing in HA except battery and linkquality of the siren :/

[image]

Any idea why ?

Koenkk, 2021-01-18:
Because the alarm is not a simple on/off switch. You have to define a mode, sound level, duration, etc.

kevincaradant, 2021-01-18:
alright I will try to look into it using mqtt requets thank you

kovaga, 2021-02-15:
my 2 cents
Firmware: 2018.03.21

Setting the "level" values does make any difference, sounds the same.
Setting the "mode" values dpoesnt make any difference, sounds the same. (police_panic, fire_panic, emergency_panic are not accepted)

Has anyone actually verifed those with the original gateway?

sebvdk, 2021-02-17:
I have the same issue. When I send warning payload :

`"warning": {
    "mode": "burglar",
    "level": "low",
    "strobe": true,
    "duration": 2
  }`

All working fine, except mode and level. However, this attributes are specified in documentation : https://www.zigbee2mqtt.io/devices/HS2WD-E.html#warning-composite

Thank you,

Zigbee2MQTT : 1.17.1
Device firmware : 2018.03.21

orxaz, 2021-04-28:
Same here, mode and level have no effect.
Somebody know if it possible to upgrade siren firmware?

mUbsta, 2021-08-17:
> Somebody know if it possible to upgrade siren firmware?

I've just received one of these with device firmware reporting 2021.2.19.  Unfortunately both mode & level still have no effect.

10der, 2021-09-02:
mode & level still have no effect.

psy0rz, 2021-12-27:
same here..can anyone verify with the original gateway?

MilorES, 2022-01-08:
I've just received one of these with device firmware reporting 2019.6.23. Unfortunately both mode & level still have no effect.

I have no original gateway to test. :(

jhonny001, 2022-04-08:
Just received a new one and still none of the options work. It is just an ON/OFF alarm with the volume too high... Did any one find a solution?

ovizii, 2022-06-14:
I'm also trying to figure this out, would love to hear from anyone who made any progress and doesn't mind sharing.

MilorES, 2022-06-14:
No, still waiting for them to fix it.

Iceman248, 2022-11-18:
I just bought one of them and same thing, mode and volume level cannot be changed.

MilorES, 2022-12-02:
How could be done to fix it? (I guess some data is missing... or something)

calas80, 2022-12-25:
same problem. News?

baradetsky, 2023-03-10:
> @baradetsky can you please explain what works and what/where to insert this code you shared?

This is the code for setting the HS2WD-E siren as a switch in z2m home assistant. This is the usual yaml markup, for example, for a configuration file

baradetsky, 2023-03-10:
> is this just an extra config to create a switch from the device which was added already via Z2M?
exactly!! this is the code for configuratoin.yaml in HA, not for z2m config
just replace the entity names with your own

Iceman248, 2023-03-23:
> 

I did this, replacing the name in zigbee2mqtt with mine, but it just says not available. Even after restarting zigbee2mqtt, only the battery and link quality are reported through MQTT about my Siren.

solarmill, 2023-07-07:
same.  Apparently the payload to turn it in needs to include 
```
"strobe_duty_cycle": VALUE
``` 
without this I could not get it to fire, but with it I could send a manual MQTT payload to activate.  Still no level control, and only one sound effect.  But if you send an on payload followed quickly (120ms) by an off payload you can make it chirp.