Koenkk/zigbee2mqtt #5793: Neo NAS-AB02B0 huge spams

kevincaradant, 2021-01-17:
The siren reports sooooo many messages. Is there any way to keep calm it ? 

Every second or half-second the humidity or temperature is different so there is a new message. My coordinator begin to be laggy with all others devices which try to send messages.

What can I do ? 

Thank you

```
MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":66,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":66,"melody":1,"temperature":"22.2","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.2","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.2","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":66,"melody":1,"temperature":"22.2","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/switch_bed_light_bedroom', payload '{"battery":100,"linkquality":75,"voltage":3005}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/switch_couloir_livingroom', payload '{"battery":97,"linkquality":138,"voltage":2995}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.2","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":66,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":66,"melody":1,"temperature":"22.2","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"energy":74.2,"linkquality":51,"power":9.4,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":66,"melody":1,"temperature":"22.2","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":66,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.2","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"energy":74.2,"linkquality":51,"power":9.2,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":66,"melody":1,"temperature":"22.2","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":66,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.2","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"energy":74.2,"linkquality":48,"power":9.5,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":39,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":66,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":39,"humidity_alarm":false,"linkquality":66,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":38,"humidity_alarm":false,"linkquality":66,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Info 
```

Messages on 15 seconds only :/

Koenkk, 2021-01-17:
Can you provide the debug log of this?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

kevincaradant, 2021-01-17:
```
Debug Received Zigbee message from 'siren_alarm_bathroom', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,222],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":37,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.2","temperature_alarm":false,"volume":"low"}'
Debug Received Zigbee message from 'siren_alarm_bathroom', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,221],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":37,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Debug Received Zigbee message from 'siren_alarm_bathroom', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,36],"type":"Buffer"},"datatype":2,"dp":106,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":36,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Debug Received Zigbee message from 'siren_alarm_bathroom', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,222],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":36,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.2","temperature_alarm":false,"volume":"low"}'
Debug Received Zigbee message from 'motion_outdoor', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3005,"10":0,"100":0,"11":84,"3":10,"4":17320,"5":8,"6":[0,1]}}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/motion_outdoor', payload '{"battery":100,"illuminance":84,"illuminance_lux":84,"linkquality":60,"occupancy":false,"temperature":10,"voltage":3005}'
Debug Received Zigbee message from 'siren_alarm_bathroom', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,221],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":36,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Debug Saving state to file /app/data/state.json
Debug Received Zigbee message from 'siren_alarm_bathroom', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,222],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":36,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.2","temperature_alarm":false,"volume":"low"}'
Debug Received Zigbee message from 'thermometer_desktop', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":1987}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/thermometer_desktop', payload '{"battery":80,"humidity":46.18,"linkquality":126,"pressure":1008,"temperature":19.87,"voltage":2965}'
Debug Received Zigbee message from 'thermometer_desktop', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":4594}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/thermometer_desktop', payload '{"battery":80,"humidity":45.94,"linkquality":129,"pressure":1008,"temperature":19.87,"voltage":2965}'
Debug Received Zigbee message from 'thermometer_desktop', type 'attributeReport', cluster 'msPressureMeasurement', data '{"measuredValue":1008,"scale":-1,"scaledValue":10087}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/thermometer_desktop', payload '{"battery":80,"humidity":45.94,"linkquality":126,"pressure":1008,"temperature":19.87,"voltage":2965}'
Debug Received Zigbee message from 'siren_alarm_bathroom', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,221],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":36,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.1","temperature_alarm":false,"volume":"low"}'
Debug Received Zigbee message from 'siren_alarm_bathroom', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,222],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/siren_alarm_bathroom', payload '{"alarm":false,"humidity":36,"humidity_alarm":false,"linkquality":69,"melody":1,"temperature":"22.2","temperature_alarm":false,"volume":"low"}'
Debug Received Zigbee message from 'siren_alarm_bathroom', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,37],"type":"Buffer"},"datatype":2,"dp":106,"fn":0,"status":0,"transid":2}' from endpoint [... cut ...]
Debug Received Zigbee message from 'router_pc_desktop', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,742000],"currentSummReceived":[0,0],"instantaneousDemand":91}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"energy":74.2,"linkquality":117,"power":9.1,"state":"ON"}'
Debug Received Zigbee message from 'siren_alarm_bathroom', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,37],"type":"Buffer"},"datatype":2,"dp":106,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
Info 
```

Koenkk, 2021-01-18:
This device is sending a lot of zigbee messages, not sure how can prevent this. @kirovilya do you know this?

kirovilya, 2021-01-18:
I confirm this sensor sends a lot of data. the strange thing is that it did not always work that way, perhaps some settings affect, or you can adjust the sending frequency (I did not find how to do this). I didn’t begin to figure out whether it was possible to fix it - I sold it :)

kevincaradant, 2021-01-18:
So maybe with Z2M we can limited this ? 

Not sure it's really useful and I tried to put only one number as parameter to avoid digits for temp and humidity but it didn't work
Not sure to understand erally all parameters that I used to try to limited messages. I think it's useless

I tried this 

```
'0x5c0272fffecertde':
    friendly_name: siren_alarm_bathroom
    temperature_precision: 0
    humidity_precision: 0
    debounce: 1
    retain: false
```

kevincaradant, 2021-01-18:
Is there any way maybe I can filtered on humidity and temperature to avoid to get them into Z2M ( I don't use them)

And like that maybe you have a parameter which say if it's the same payload (message) I don't publish it ? 

It publish a lot of messages because temp and humidity change everytime. These params are useless in my case.

arduino43, 2021-01-19:
I have the same issue - it will spam and also notice 5-30 second delays for commands.

kevincaradant, 2021-01-26:
Yes always same behavior. I don't know if I should continue to keep it or not until the issue is solved.

Any plan to avoid to ddos the co-ordinator? 

It's for me a huge issue a device which spam like that every time :/

Thank you

roth-m, 2021-02-04:
After setting humidity/temperature_min/max values and temperature/humidity_precision to 0, mine is sending updates as often as my Xiaomi Temperature sensor.
It is also running on batteries but I don't have any battery update...

kevincaradant, 2021-02-06:
Hi @roth-m, did you set the `humidity_precision` to 0 inside the `configuration.yaml` file of Z2M ? 

Because I can set these different values: `humidity/temperature_min/max values` from the state of the device but for the precision, I just wrote something like that into the config file of Z2M:

```
'0x5c027xxxxxxxxxxxxa':
    friendly_name: siren_alarm_bathroom
    humidity_precision: 0
    temperature_precision: 0
```

Thank you to confirm if it was you did ?

kevincaradant, 2021-02-06:
not sure but I think I have less messages only when the device is on battery ...

As soon as I plug it on AC power, I have a lot of messages :/

I tried the precision of humidity and it's not better on AC power. 

As mentioned @roth-m , @Koenkk do you thing you have the possibility to add the battery information of the device ? :)

Thank you ;)

kevincaradant, 2021-02-06:
By the way @roth-m , do you also a kind of "sizzle" when it's running on battery ? Every minutes, I have a "chtic" sound very quick but I can hear it.

On AC I didn't have the sizzle sound but I got manyyyyy messages every seconds ... so it's good also.

roth-m, 2021-02-06:
> Hi @roth-m, did you set the `humidity_precision` to 0 inside the `configuration.yaml` file of Z2M ?
>
> Because I can set these different values: `humidity/temperature_min/max values` from the state of [...]

> By the way @roth-m , do you have also a kind of "sizzle" when it's running on battery ? Every minute, I have a "chtic" sound very quick but I can hear it.
> 
> On AC I didn't have the sizzle sound but I got manyyyyy messages every second ... so it's not good also.

Yes, this is it.

kevincaradant, 2021-02-07:
> > Hi @roth-m, did you set the `humidity_precision` to 0 inside the `configuration.yaml` file of Z2M ?
> > Because I can set these different values: `humidity/temperature_min/max values` from the state of the device but for the precision, I just wrote something like that into the config file of Z2M:
> > '0x5c027xxxxxxxxxxxxa':
> > friendly_name: siren_alarm_bathroom
> > humidity_precision: 0
> > temperature_precision: 0
> > ```
> > 
> > Thank you to confirm if it was you did ?
> > ```
> 
> Yes, this is it.
> 
> > By the way @roth-m , do you have also a kind of "sizzle" when it's running on battery ? Every minute, I have a "chtic" sound very quick but I can hear it.
> > On AC I didn't have the sizzle sound but I got manyyyyy messages every second ... so it's not good also.
> 
> Got some, now less frequent. It sounds like the speaker is powered on (without sound).

oh alright, you use it since the begin pour battery and it still works without doing many sizzles. Ok I will try to hide it somewhere where I can't hear it, 60 times in one hour, I will become crazy to be near of it :P

roth-m, 2021-02-07:
> > > Hi @roth-m, did you set the `humidity_precision` to 0 inside the `configuration.yaml` file of Z2M ?
> > > Because I can set these different values: `humidity/temperature_min/max values` from the state of the device but for the precision, I just wrote something like that into the config file of Z2M:
> > > '0x5c027xxxxxxxxxxxxa':
> > > friendly_name: siren_alarm_bathroom
> > > humidity_precision: 0
> > > temperature_precision: 0
> > > ```
> > > 
> > > Thank you to confirm if it was you did ?
> > > ```
> > 
> > 
> > Yes, this is it.
> > > By the way @roth-m , do you have also a kind of "sizzle" when it's running on battery ? Every minute, I have a "chtic" sound very quick but I can hear it.
> > > On AC I didn't have the sizzle sound but I got manyyyyy messages every second ... so it's not good also.
> > 
> > 
> > Got some, now less frequent. It sounds like the speaker is powered on (without sound).
> 
> oh alright, you use it since the begin on battery and it still works without doing many sizzles ?
> Ok I will try to hide it somewhere where I can't hear it, 60 times in one hour, I will become crazy to be near of it :P
> At least, on battery, I don't have 500 messages by hour. But now I have the sizzle :P

I heard them frequently after putting back the battery (like every 10 seconds).
I spent 10 minutes next to it but could not ear it once.

closed 2021-03-17 by github-actions[bot]