Koenkk/zigbee2mqtt #5794: How to use well Z2M groups ? Many spam in this case isn't it ?

kevincaradant, 2021-01-17:
Hi

I'm trying to use groups in Z2M, all my bulbs react in the same time contrary of the Home Assistant group.

Is it better to use Z2M group or HA group ? 

Sorry for the noobie question. I think I should have to use Z2M to avoid many send of messages but maybe HA could do the same ? 

However, I created `Z2M_alarm_lights` group in Z2M. I can see it on the logs but why I saw also all bulbs concerning by the group in the logs ? Is it a spam or a intense use for the coordinator ? 6 messages in the same time each time I trigger the group (every second in my case)

```
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":60,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Z2M_alarm_lights', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_2_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":117,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_3_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":84,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_4_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":66,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_5_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":72,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_couloir_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":69,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_2_couloir_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":36,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_bathroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":15,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_kitchen', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":78,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":60,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/Z2M_alarm_lights', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_2_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":117,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_3_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":84,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_4_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":66,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_5_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":72,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_couloir_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":69,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_2_couloir_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":36,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_bathroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":15,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_kitchen', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":78,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":60,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Z2M_alarm_lights', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_2_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":117,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_3_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":84,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_4_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":66,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_5_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":72,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_couloir_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":69,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_2_couloir_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":36,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_bathroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":15,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_kitchen', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":78,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":60,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/Z2M_alarm_lights', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_2_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":117,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_3_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":84,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_4_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":66,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_5_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":72,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_couloir_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":69,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_2_couloir_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":36,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_bathroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":15,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_kitchen', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":78,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":60,"state":"ON"}'
[... cut ...]
Info MQTT publish: topic 'zigbee2mqtt/Z2M_alarm_lights', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_2_livingroom', payload '{"brightness":254,"color":{"x":0.465681320891482,"y":0.411820371529656},"color_temp":380,"linkquality":117,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_3_livingroom', payload '{"brightness":254,"color"
```

[image]

Thank you to tell me if it's a bug ? a bas use of groups in Z2M ? Or something else ? :)

[image]

kevincaradant, 2021-01-17:
Oh is it linked to the endpoint value ? I saw all lights have the endpoints 11. It's like that we reduce the traffic on the coordinator ? 

We can see 100 lines but if it's the same end point, it's good ? :)

So new for me the scenes and groups :)

kevincaradant, 2021-01-26:
I think I use well as expected even if I didn't see any answer so I will close this issue.

If anyone have something to add, please do it in this issue

closed 2021-01-26 by kevincaradant

MattWestb, 2021-01-27:
One on command to one groupe of 55 lights is 1 command sent in the zigbee network.
One on command to one AH group of 55 lights is 55 individual commands sent in the zigbee network.
And if you have configure status reporting you is getting 55 or more status reports back from both variants (without transition time and with its 55 X status changes reported from etch light).

In zigbee you should working with light grupes then its possible for not spamming the network.

kevincaradant, 2021-01-27:
Thank you for this answer :).
Perfect, that confirm what I thought about groups in Z2M vs HA.