Koenkk/zigbee2mqtt #5795: CC2531 failed to set dtr/rts on  Raspberry Pi 3 B+

vicbap, 2021-01-17:
## What happened
The CC2531 dongle "stopped working" after a several months working on a Raspberry Pi 3 B+. After rebooting and unplug an plug the usb it works one or two hours and then crashes again. After that, if the usb is recconnected the, light start green and, when you start z2m it turns off and don not start until raspberry is reboot.

## What did you expect to happen
That Work properly

## How to reproduce it (minimal and precise)
Plug the usb dongle, start z2m and wait until message appears on dmesg.
I changed all possibilities, Raspberry Pi 3 B+, CC2351, Sd card, raspbian, power adapter, firmware and happens again after a few hours.

## Debug info
Zigbee2MQTT version: 1.17.0 (commit #07cdc9d)
Adapter hardware: CC2531
Adapter firmware version:  CC2531_DEFAULT_20190608.zip

dmesg before starting:
https://pastebin.com/Df5rwcFx

current log: 
https://pastebin.com/9v3Sax1E

When app stops dmesg shows "failed to set dtr/rts"

Koenkk, 2021-01-18:
Is your CC2531 connected through a usb extension cable? (if not do so). Have you also tried reflashing the CC2531?

vicbap, 2021-01-21:
Hi, thank you for answering.

I tried with and without extension cable. 
I've reflashed it and changed also with another one that I have.
Also I changed the Rpi3 and the sd too.

giuseppeg88, 2021-02-24:
I have the same problem ("failed to set dtr/rts" on Raspberry Pi 3 B+ & CC2531).

hoddi78, 2021-04-03:
I've had the same trouble for the past months. Going for conbeeII now.

closed 2021-05-12 by github-actions[bot]