Koenkk/zigbee2mqtt #5797: `debounce` missing for device options?

sjorge, 2021-01-17:
https://github.com/Koenkk/zigbee2mqtt/blob/2ef2c547023d29325616f72f5cefa49b68d96bac/lib/util/settings.schema.json#L464

I think debounce is missing here, but not sure. It certainly works when setting it per-device though.

nurikk, 2021-01-18:
Feel free to raise PR and add missing properties

sjorge, 2021-01-18:
> Feel free to raise PR and add missing properties

Assuming it is supposed to be there though, as I used it for a while but hit quite a few bugs with it over my time using it. So maybe it is missing for good reason.

timoline, 2021-01-18:
I think this is also missing?

frontend:
  port: 8080

nurikk, 2021-01-19:
> > Feel free to raise PR and add missing properties
> 
> Assuming it is supposed to be there though, as I used it for a while but hit quite a few bugs with it over my time using it. So maybe it is missing for good reason.

It was lacking in the original implementation. What is does is validates configuration.yaml on start, additionally, I use this definition to render UI

sjorge, 2021-01-19:
Done https://github.com/Koenkk/zigbee2mqtt/pull/5823

closed 2021-01-19 by sjorge

timoline, 2021-01-19:
> I think this is also missing?
> 
> frontend:
> port: 8080

I think something like this...

```
    "frontend": {
      "type": "object",
      "title": "Frontend settings",
      "properties": {
        "port": {
          "type": ["number"],
          "title": "Port",
          "description": "Frontend port",
          "default": "8080"
        },
        "host": {
          "type": "number",
          "title": "Hostname,
          "description": "Hostname of frontend",
          "default": "0.0.0.0"
        },
        "auth_token": {
          "type": "string",
          "title": "Authentication token",
          "Optional, enables authentication, disabled by default"
        }
      },
	  "required": ["port"]
    },
```

timoline, 2021-01-20:
syslog options are also missing...
https://www.zigbee2mqtt.io/information/configuration.html