Koenkk/zigbee2mqtt #5810: IKEA E1743 no battery info

smulle48, 2021-01-18:
Hi

Just installed a fresh Homebridge on my pi. Then I installed zigbee2mqtt and started adding devices.

The problem is that I get no battery information. I's been sitting next to the Conbee2 stick for a few hours now, but it still says N/A%

Is there anything I have done wrong?

If I include it in my deCONZ network, I get battery info right away

mvogt1, 2021-01-18:
I have seen the N/A power status too in the zigbee2mqtt webgui, but at that time I didn't care. The next time I looked the info was there, it took maybe 30 minutes, but it was not directly shown. (Tradfri switch and motion sensor)

sjorge, 2021-01-18:
Usually it shows up after ~24h, as it is reported once a day.

@Koenkk I was thinking about maybe adding a get converter, but it looks tricker as we get battery info in various ways :( not sure how to tackle this.

smulle48, 2021-01-18:
> Usually it shows up after ~24h, as it is reported once a day.
> 
> 
> 
> @Koenkk I was thinking about maybe adding a get converter, but it looks tricker as we get battery info in various ways :( not sure how to tackle this.

For fun, I just tried to add and tuya switch and that reported the battery right away.

Koenkk, 2021-01-19:
@sjorge I think we can read the voltage in the `configure`

sjorge, 2021-01-19:
> @sjorge I think we can read the voltage in the `configure`

That might be a better idea, we know if we need voltage or batterypercent too as that per-device.

Koenkk, 2021-01-19:
@sjorge that shouldn't be hard, as you should see either `reporting.batteryPercentageRemaining` or `reporting.batteryVoltage`

sjorge, 2021-01-19:
> @sjorge that shouldn't be hard, as you should see either `reporting.batteryPercentageRemaining` or `reporting.batteryVoltage`

Done, I've just added the read call to those reporting function, more motivation as to why I settled on that is in the PR.

smulle48, 2021-01-19:
> > @sjorge that shouldn't be hard, as you should see either `reporting.batteryPercentageRemaining` or `reporting.batteryVoltage`
> 
> 
> 
> Done, I've just added the read call to those reporting function, more motivation as to why I settled on that is in the PR.

Sweet :-)

I'll check it out tonight when I get home :-)

Thank you for your work :-)

smulle48, 2021-01-21:
Can confirm it's working, though I had to switch to the dev branch :-D

closed 2021-03-01 by github-actions[bot]