Koenkk/zigbee2mqtt #5811: Can't pass interview step for TS0601 actuator after update

sneakythr0ws, 2021-01-18:
## What happened
Can't pass interview step for TS0601 actuator after update. Have same passed interveiw in previous versions and it works fine.
Other zigbee devices connect and pass interview without problems. 
Tried to reinstall Zigbee2MQTT hassos plugin. Tried to turn off and reconnect usn stick.

## What did you expect to happen
Pass interview)
## How to reproduce it (minimal and precise)
Set in addon config
```
advanced:
  log_level: debug
zigbee_herdsman_debug: true
```
Debug log:
```
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":true},"status":"ok","transaction":"nqpt4-2"}'
INFO: Device '0x5c0272fffec0415a' joined
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffec0415a","ieee_address":"0x5c0272fffec0415a"},"type":"device_joined"}'
INFO: Starting interview of '0x5c0272fffec0415a'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffec0415a","ieee_address":"0x5c0272fffec0415a","status":"started"},"type":"device_interview"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x5c0272fffec0415a"},"type":"device_connected"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x5c0272fffec0415a"},"type":"pairing"}'
DEBUG: Received Zigbee message from 'Kitchen wall switch', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'Children's wall switch', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3300,"10":0,"100":1,"101":0,"110":1,"111":255,"153":0,"155":0,"3":28,"5":71,"8":9749}}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'Toilet/bathroom wall switch', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'Living room wall switch', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'Children's motion sensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Children's motion sensor', payload '{"battery":100,"battery_low":false,"linkquality":57,"occupancy":false,"tamper":false,"voltage":3000}'
DEBUG: Received Zigbee message from 'Children's motion sensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Children's motion sensor', payload '{"battery":100,"battery_low":false,"linkquality":57,"occupancy":true,"tamper":false,"voltage":3000}'
DEBUG: Received Zigbee message from 'Children's wall switch', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Living room motion sensor', payload '{"battery":100,"illuminance":4,"illuminance_lux":4,"linkquality":78,"occupancy":false,"voltage":3015}'
DEBUG: Received Zigbee message from 'Kitchen wall switch', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3300,"10":0,"100":1,"101":0,"110":1,"111":255,"153":0,"155":0,"3":24,"5":45,"8":9749}}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'Kitchen wall switch', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'Toilet/bathroom wall switch', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'Living room wall switch', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'Children's wall switch', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'Living room motion sensor', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":4}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Living room motion sensor', payload '{"battery":100,"illuminance":4,"illuminance_lux":4,"linkquality":76,"occupancy":true,"voltage":3015}'
DEBUG: Received Zigbee message from 'Living room motion sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Living room motion sensor', payload '{"battery":100,"illuminance":4,"illuminance_lux":4,"linkquality":73,"occupancy":true,"voltage":3015}'
DEBUG: Saving state to file /share/zigbee2mqtt/state.json
DEBUG: Received Zigbee message from 'Kitchen wall switch', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'Toilet/bathroom wall switch', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
ERROR: Failed to interview '0x5c0272fffec0415a', device has not successfully been paired
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffec0415a","ieee_address":"0x5c0272fffec0415a","status":"failed"},"type":"device_interview"}'
```
## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC2538
Adapter firmware version:    zStack30x
Home Assistant Add-on: Zigbee2mqtt: 1.17.0.1

sneakythr0ws, 2021-01-23:
Same with Sonoff ZBMINI :(
```
Zigbee2MQTT:error 2021-01-23 17:48:17: Failed to interview '0x847127fffe924ae0', device has not successfully been paired
```

Koenkk, 2021-01-23:
Make sure to check https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

sneakythr0ws, 2021-01-23:
@Koenkk 
Yes. I tried. 
Another devices join normally. But these hava problems in interview step.
Here is zigbee-herdsman debug log for ZBMINI
```
2021-01-23T15:33:35.563Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,10,0,196,10,1,1,0,105,0,74,209,32,0,0,5,16,55,0,0,0,196,10,29] - 59
2021-01-23T15:33:35.564Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":10,"srcaddr":2756,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":105,"securityuse":0,"timestamp":2150730,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[16,55,0,0,0]}}
2021-01-23T15:33:35.566Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":55,"manufacturerCode":null,"commandIdentifier":0},"Payload":[{"attrId":0}]},"address":2756,"endpoint":1,"linkquality":105,"groupID":0}'
2021-01-23T15:33:35.569Z zigbee-herdsman:controller:endpoint ReadResponse 0x00158d00024dc8ed/1 genTime({"time":664731216}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2021-01-23T15:33:35.570Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00158d00024dc8ed:2756/1 (0,0,1)
2021-01-23T15:33:35.572Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":2756,"destendpoint":1,"srcendpoint":1,"clusterid":10,"transid":106,"options":0,"radius":30,"len":11,"data":{"type":"Buffer","data":[24,55,1,0,0,0,226,80,254,158,39]}}
2021-01-23T15:33:35.573Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,21,36,1,196,10,1,1,10,0,106,0,30,11,24,55,1,0,0,0,226,80,254,158,39,80]
2021-01-23T15:33:35.574Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-01-23T15:33:35.588Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-01-23T15:33:35.588Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-01-23T15:33:35.589Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-01-23T15:33:35.589Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-01-23T15:33:35.590Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-01-23T15:33:35.986Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,106,172]
2021-01-23T15:33:35.987Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,106,172]
2021-01-23T15:33:35.988Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,106] - 172
2021-01-23T15:33:35.988Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":106}
2021-01-23T15:33:35.989Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-01-23T15:33:39.573Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,29,68,129,0,0,0,5,195,109,1,1,0,49,0,217,34,33,0,0,9,25,93,0,0,0,0,23,0,0,187,121,28,28]
2021-01-23T15:33:39.574Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,29,68,129,0,0,0,5,195,109,1,1,0,49,0,217,34,33,0,0,9,25,93,0,0,0,0,23,0,0,187,121,28,28]
2021-01-23T15:33:39.574Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 29 - 2 - 4 - 129 - [0,0,0,5,195,109,1,1,0,49,0,217,34,33,0,0,9,25,93,0,0,0,0,23,0,0,187,121,28] - 28
2021-01-23T15:33:39.576Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1280,"srcaddr":28099,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":49,"securityuse":0,"timestamp":2171609,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[25,93,0,0,0,0,23,0,0]}}
2021-01-23T15:33:39.579Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":93,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"zonestatus":0,"extendedstatus":0}},"address":28099,"endpoint":1,"linkquality":49,"groupID":0}'
2021-01-23T15:33:39.583Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2021-01-23 18:33:39: MQTT publish: topic 'zigbee2mqtt/Children's motion sensor', payload '{"battery":100,"battery_low":false,"linkquality":49,"occupancy":false,"tamper":false,"voltage":3100}'
2021-01-23T15:33:47.397Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,10,0,52,167,1,1,0,55,0,93,130,33,0,0,5,16,34,0,0,0,52,167,29,53]
2021-01-23T15:33:47.398Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,10,0,52,167,1,1,0,55,0,93,130,33,0,0,5,16,34,0,0,0,52,167,29,53]
2021-01-23T15:33:47.399Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,10,0,52,167,1,1,0,55,0,93,130,33,0,0,5,16,34,0,0,0,52,167,29] - 53
2021-01-23T15:33:47.400Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":10,"srcaddr":42804,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":55,"securityuse":0,"timestamp":2196061,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[16,34,0,0,0]}}
[... cut ...]
2021-01-23T15:36:12.962Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,193,154,193,154,35]
2021-01-23T15:36:12.966Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2021-01-23T15:36:12.967Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2021-01-23T15:36:12.967Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2021-01-23T15:36:12.968Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2021-01-23T15:36:12.969Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

Koenkk, 2021-01-23:
The device does not respond to any requests I see. Have you tried pairing it very close to the coordinator?

sneakythr0ws, 2021-01-23:
> The device does not respond to any requests I see. Have you tried pairing it very close to the coordinator?

I can't try it with zbmini
But I tried with TS0601 and situation is same(
```
Zigbee2MQTT:info  2021-01-23 19:12:59: Device '0x5c0272fffec0415a' joined
Zigbee2MQTT:info  2021-01-23 19:12:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffec0415a","ieee_address":"0x5c0272fffec0415a"},"type":"device_joined"}'
2021-01-23T16:12:59.374Z zigbee-herdsman:controller:log Interview '0x5c0272fffec0415a' start
Zigbee2MQTT:info  2021-01-23 19:12:59: Starting interview of '0x5c0272fffec0415a'
Zigbee2MQTT:info  2021-01-23 19:12:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffec0415a","ieee_address":"0x5c0272fffec0415a","status":"started"},"type":"device_interview"}'
2021-01-23T16:12:59.407Z zigbee-herdsman:controller:device:log Interview - start device '0x5c0272fffec0415a'
2021-01-23T16:12:59.417Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":48795,"nwkaddrofinterest":48795}
2021-01-23T16:12:59.417Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,155,190,155,190,35]
Zigbee2MQTT:info  2021-01-23 19:12:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x5c0272fffec0415a"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-01-23 19:12:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x5c0272fffec0415a"},"type":"pairing"}'
2021-01-23T16:12:59.446Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,155,190,90,65,192,254,255,114,2,92,0,0,80,254,1,101,2,0,102]
2021-01-23T16:12:59.447Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,155,190,90,65,192,254,255,114,2,92,0,0,80,254,1,101,2,0,102]
2021-01-23T16:12:59.447Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [155,190,90,65,192,254,255,114,2,92,0,0] - 80
2021-01-23T16:12:59.448Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":48795,"extaddr":"0x5c0272fffec0415a","parentaddr":0}
2021-01-23T16:12:59.449Z zigbee-herdsman:controller:log Device '0x5c0272fffec0415a' joined
2021-01-23T16:12:59.449Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2021-01-23T16:12:59.450Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2021-01-23T16:12:59.451Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2021-01-23T16:12:59.451Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-01-23T16:12:59.452Z zigbee-herdsman:controller:log Device '0x5c0272fffec0415a' accepted by handler
2021-01-23T16:12:59.452Z zigbee-herdsman:controller:log Not interviewing '0x5c0272fffec0415a', completed 'false', in progress 'true'
2021-01-23T16:13:00.290Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,240,1,8,62]
2021-01-23T16:13:00.291Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,240,1,8,62]
2021-01-23T16:13:00.291Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [240,1,8] - 62
2021-01-23T16:13:00.292Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":240,"endpoint":1,"transid":8}
2021-01-23T16:13:00.293Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-01-23T16:13:00.295Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,240,1,11,61,254,3,68,128,240,1,22,32]
2021-01-23T16:13:00.296Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,240,1,11,61,254,3,68,128,240,1,22,32]
2021-01-23T16:13:00.296Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [240,1,11] - 61
2021-01-23T16:13:00.297Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":240,"endpoint":1,"transid":11}
2021-01-23T16:13:00.298Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,240,1,22,32]
2021-01-23T16:13:00.298Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [240,1,22] - 32
2021-01-23T16:13:00.299Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":240,"endpoint":1,"transid":22}
2021-01-23T16:13:00.300Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-01-23T16:13:00.884Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,52,167,52,167,202,30,43,2,0,141,21,0,132,104]
2021-01-23T16:13:00.885Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,52,167,52,167,202,30,43,2,0,141,21,0,132,104]
2021-01-23T16:13:00.885Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [52,167,52,167,202,30,43,2,0,141,21,0,132] - 104
2021-01-23T16:13:00.886Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":42804,"nwkaddr":42804,"ieeeaddr":"0x00158d00022b1eca","capabilities":132}
2021-01-23T16:13:00.888Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 42804
2021-01-23T16:13:00.890Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":42804,"options":0,"radius":30}
2021-01-23T16:13:00.891Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,52,167,0,30,233]
2021-01-23T16:13:00.892Z zigbee-herdsman:controller:log Device announce '0x00158d00022b1eca'
[... cut ...]
2021-01-23T16:14:21.619Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,155,190,155,190,35]
2021-01-23T16:14:21.629Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2021-01-23T16:14:21.630Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2021-01-23T16:14:21.631Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2021-01-23T16:14:21.631Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2021-01-23T16:14:21.632Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

sneakythr0ws, 2021-01-23:
Right after that successfully attached the aqara button

sneakythr0ws, 2021-01-23:
Log after pairing ZBMINI close to the coordinator
```
2021-01-23T17:10:05.320Z zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '14348' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry
2021-01-23T17:10:05.321Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 14348
2021-01-23T17:10:05.322Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":14348,"options":0,"radius":30}
2021-01-23T17:10:05.323Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,12,56,0,30,78]
2021-01-23T17:10:05.330Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
2021-01-23T17:10:05.330Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
2021-01-23T17:10:05.330Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
2021-01-23T17:10:05.331Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
2021-01-23T17:10:05.331Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-01-23T17:10:08.341Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":14348,"nwkaddrofinterest":14348}
2021-01-23T17:10:08.342Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,12,56,12,56,35]
2021-01-23T17:10:08.346Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2021-01-23T17:10:08.347Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2021-01-23T17:10:08.347Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2021-01-23T17:10:08.348Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2021-01-23T17:10:08.348Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-01-23T17:10:09.903Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,10,0,20,116,1,1,0,60,0,197,24,182,0,0,5,16,37,0,0,0,187,121,28,15]
2021-01-23T17:10:09.903Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,10,0,20,116,1,1,0,60,0,197,24,182,0,0,5,16,37,0,0,0,187,121,28,15]
2021-01-23T17:10:09.904Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,10,0,20,116,1,1,0,60,0,197,24,182,0,0,5,16,37,0,0,0,187,121,28] - 15
2021-01-23T17:10:09.905Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":10,"srcaddr":29716,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":60,"securityuse":0,"timestamp":11933893,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[16,37,0,0,0]}}
2021-01-23T17:10:09.908Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":37,"manufacturerCode":null,"commandIdentifier":0},"Payload":[{"attrId":0}]},"address":29716,"endpoint":1,"linkquality":60,"groupID":0}'
2021-01-23T17:10:09.913Z zigbee-herdsman:controller:endpoint ReadResponse 0x00158d0002117e53/1 genTime({"time":664737010}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2021-01-23T17:10:09.914Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00158d0002117e53:29716/1 (0,0,2)
2021-01-23T17:10:09.916Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":29716,"destendpoint":1,"srcendpoint":1,"clusterid":10,"transid":148,"options":0,"radius":30,"len":11,"data":{"type":"Buffer","data":[24,37,1,0,0,0,226,242,20,159,39]}}
2021-01-23T17:10:09.917Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,21,36,1,20,116,1,1,10,0,148,0,30,11,24,37,1,0,0,0,226,242,20,159,39,91]
2021-01-23T17:10:09.918Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-01-23T17:10:09.923Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-01-23T17:10:09.923Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-01-23T17:10:09.923Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-01-23T17:10:09.924Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-01-23T17:10:09.924Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-01-23T17:10:09.929Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,148,82]
2021-01-23T17:10:09.929Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,148,82]
2021-01-23T17:10:09.930Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,148] - 82
2021-01-23T17:10:09.930Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":148}
2021-01-23T17:10:09.930Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-01-23T17:10:18.353Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x847127fffe924ae0', attempt 3
2021-01-23T17:10:18.356Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":14348,"nwkaddrofinterest":14348}
2021-01-23T17:10:18.356Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,12,56,12,56,35]
2021-01-23T17:10:18.361Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
[... cut ...]
2021-01-23T17:10:31.385Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,12,56,12,56,35]
2021-01-23T17:10:31.390Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2021-01-23T17:10:31.390Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2021-01-23T17:10:31.391Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2021-01-23T17:10:31.391Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2021-01-23T17:10:31.392Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

sneakythr0ws, 2021-01-23:
And this may be interesting 
```
2021-01-23T17:16:23.024Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x847127fffe924ae0', attempt 6
2021-01-23T17:16:23.026Z zigbee-herdsman:controller:device:log Interview - failed for device '0x847127fffe924ae0' with error 'Error: Interview failed because can not get node descriptor ('0x847127fffe924ae0')
    at Device.<anonymous> (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/controller/model/device.js:350:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/controller/model/device.js:25:65)'
2021-01-23T17:16:23.027Z zigbee-herdsman:controller:error Interview failed for '0x847127fffe924ae0 with error 'Error: DatabaseEntry with ID '26' does not exist'
Zigbee2MQTT:error 2021-01-23 20:16:23: Failed to interview '0x847127fffe924ae0', device has not successfully been paired
```

sneakythr0ws, 2021-01-23:
Is it possible to configure pair this timeout?
`nodeDescRsp after 10000ms`

Koenkk, 2021-01-24:
Increasing that timeout won't fix the issue, zigbee2mqtt tries 6 times with a timeout of 10000ms. To debug this further we would need to sniff the traffic when pairing this device: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

sneakythr0ws, 2021-01-26:
I set 20 channel in settings. And all devices are working now. Is it firmware/hardware issue?

Koenkk, 2021-01-27:
Could be that the firmware of the TS0601 only allows pairing on certain channel (e.g. 11, 15, 20, 25)

sneakythr0ws, 2021-01-27:
> Could be that the firmware of the TS0601 only allows pairing on certain channel (e.g. 11, 15, 20, 25)

Thank you very much for your help

closed 2021-01-27 by sneakythr0ws