Koenkk/zigbee2mqtt #5812: No log output when "log_level": "warn"

Mr-Groch, 2021-01-18:
## What happened
If I set `"log_level": "warn",` in configuration file and restart zigbee2mqtt, then there will be no log output to file/frontend. Thre will be no warn logs or even no error logs written. If I set `"log_level": "error"`, or `"log_level": "info",` then all is working as expected.

## What did you expect to happen
Logs at level warn and error should be written when `"log_level": "warn",` in configuration file

## How to reproduce it (minimal and precise)
1. Set `"log_level": "warn",` into `advanced:` section of configuration file
2. Restart addon and observe that no logs is wrtten to file or frontend logs tabs (even no error logs)

## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC26X2R1 zzh!
Adapter firmware version: 20201113

Koenkk, 2021-01-18:
Should be fixed now, can you confirm?

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Mr-Groch, 2021-01-18:
It's partially fixed. When I start z2m with `"log_level": "warn",`, logs at level warn and error will be written to file and console, but not to the frontend logs tab. Only after changing in frontend log level setting to something else (like error or info) and then to warn again it will be also outputed to frondend logs tab.

Koenkk, 2021-01-19:
Should also be fixed now.

Mr-Groch, 2021-01-19:
Its OK now, thanks!

closed 2021-01-19 by Mr-Groch