Koenkk/zigbee2mqtt #5814: Cannot pair Lidl TY0203 door/window switch

BogdanDIA, 2021-01-18:
## What happened
Cannot pair Lidl TY0203 door/window switch

## What did you expect to happen
To pair the device.

## How to reproduce it (minimal and precise)
I am using CC2652RB from Slaeh as a coordinator with Z-stack 3.x.0. I try to pair the device but 18/20 tries pairing fails. I am using HA on rpi 3B+.
My view is that the whole pairing/interviewing is too slow so that there is a timeout. See the logs I attach.
To verify my assumption, I did run the Z2M standalone on my desktop with Ubuntu 20.04 and it was obvious that the messages come faster than on rpi3 B+ so most of the times the pairing succeeds.
Interesting enough is that the If I am close to a light bulb like Livarno GU10 from Lidl, the pairing will be done through it and will succeed most of the times. Also in this case I can see that the messages come faster so I guess the timeout does not occur.

I should mention that this is happening with Lidl switches and motion sensors but  not with Livarno light bulbs which are  really fast and much more powerful than the CC2652RB with 3dBi antenna.

## Debug info
Zigbee2MQTT version: 5.1
Adapter hardware: CC2652RB 
Adapter firmware version: Z-stack 3.x.0

Here you can find the Z2M and heardsman relevant logs:

https://pastebin.com/raw/adP1gKr5

nikos1988, 2021-01-22:
I got the exact same problem. Tried zzh and slaesh stick and only got it to pair once in dozens of tries.

Koenkk, 2021-01-22:
This probably highly depends on where the device is paired. Does it succeed when pairing close to the coordinator?

nikos1988, 2021-01-22:
I put it as newr as 5cm to the antenna of the stick

BogdanDIA, 2021-01-22:
I did experiment with various distances from coordinator but it does not make difference if it is at 5cm from coordinator or 2m. Seems to matter the processing speed of the host (rpi or PC). But why it goes quicker when pairing through the bulb?

BTW, the datasheet from CC2652RB gives a setting for txPower up to 5dBm. Is the coordinator using maximum?

nikos1988, 2021-01-23:
Hi,
i just also tried to pair the LIDL motion sensor. It needed about 6 tries until it successfully paired.

I don't think the PC matters here. I am running z2m on a Xeon D-1527 in a VM that has enough power and it is also happening here.

Koenkk, 2021-01-23:
- Is the stick connected through a usb extension cable?
- Can you provide the herdsman debug log and a sniff when pairing fails? To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging ; https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

> BTW, the datasheet from CC2652RB gives a setting for txPower up to 5dBm. Is the coordinator using maximum?

With this firmware it is: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

BogdanDIA, 2021-01-23:
Hi Koenkk, 
- I use an USB cable for conecting the stick. I also disabled the wifi and BT on rpi boot.
- In initial post there is a log containing herdsman details for the moment when joining fails. I will create new logs when I will have a sniffer available. I expected to have a need ofr a sniffer and ordered one and will arrive in couple of days.

One question about herdsman log. Is there a way to save it in a file? I use hassio and docker so the log is only available in the add-on page but it is just last part of the full log since the add-on is started. I know herdsman is sending log on console.

BogdanDIA, 2021-01-27:
Hi Koenkk,

I finally got the sniffer and did a log for a failed pairing. Without the lightbulb I can get all the time failing pairing. Inside the archive are herdsman log, zigbee2mqtt log and a pcap from wireshark.

https://paste.c-net.org/CyclopsHedley

Koenkk, 2021-01-28:
The issue is that the coordinator does not respond to the "Node descriptor request", I currently have no clue why (maybe it's due to the high counter??).

[image]

Can you also provide a sniff when the pairing goes correctly? (so we can do a comparison)

BogdanDIA, 2021-01-28:
I attach a sniff with a successful join done through a router (light bulb). I cannot make a successful join directly with the coordinator. The Node Descriptor Request is successful in both directions.

https://paste.c-net.org/HomesMueller

Other observations:
1. if I get the device very close to the coordinator, like 5cm from the antenna, I get a lot of checksum errors. The ideal distance with minimum checksum errors seems to be around 1m
2. pairing through light bulb does not give checksum errors when placing the device at 5cm distance, The light bulb has tx power of max 13dBm from the manual.

I can do also sniffing when using the coordinator on my laptop, in which case the pairing works most of the time w/o router

BTW, what are the Ack messages w/o any source and destination address?

Koenkk, 2021-01-29:
@BogdanDIA it's indeed working in that case. Debugging this would require some serious stack debugging for which I need this device. Since you have a workaround and no other users seem to be affected yet that's overkill for now.

BogdanDIA, 2021-01-29:
@Koenkk 

Well, I did another sniff, this time with ZHA. With ZHA pairing works 10 out of 10 times and it is fast.
Then I switched to Z2M back by uninstalling ZHA and starting Z2M addon.
The first pairing with Z2M worked but subsequent ones did not.

Here are the pcap files:
ZHA
https://paste.c-net.org/BreezyBoulder
Z2M
https://paste.c-net.org/BryanDeadline

At a first glance I see a Bind response that I never saw in Z2M. Actually, in successful Z2M log there is a Bind timeout but the pairing still goes on.

90	19:46:36.796250	0x1164	0x0000	ZigBee ZDP	47	Bind Response, Status: Success

Maybe this can show some obvious problem.

Regards,
Bogdan

Koenkk, 2021-01-30:
Then the problems seems to be somewhere in the interview process. Both sniff you are provided are not complete since they don't contain messages send by the coordinator, E.g. in the ZHA sniff packet no. 4 you see an "Active endpoint response" but no "Active endpoint request" from the coordinator (which should be send before that). Probably the sniff was taken out of range from the coordinator.

hansgans007, 2021-01-30:
my Log https://pastebin.com/2nyuzcyf

Koenkk, 2021-01-30:
@hansgans007 can you try pairing it close to a bulb instead? If you already tried try pairing it very close to the coordinator. Also check: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

hansgans007, 2021-01-30:
these trys was make 30-50cm 
wait a sec 
ill try another time with 1 cm

hansgans007, 2021-01-30:
now 1cm 

https://pastebin.com/25h7FRQW

hansgans007, 2021-01-30:
zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '19480' failed (Error: SREQ '--> ZDO - nodeDescReq - {"dstaddr":19480,"nwkaddrofinterest":19480}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')), retry +3s

maybe ? 
ill start all 3 together :D

hansgans007, 2021-01-30:
ah i dont have any bulb only window sensors -.-

hansgans007, 2021-01-30:
this come after a while 

zcl data from unknown device

https://pastebin.com/iaizmfHQ

BogdanDIA, 2021-01-30:
> Then the problems seems to be somewhere in the interview process. Both sniff you are provided are not complete since they don't contain messages send by the coordinator, E.g. in the ZHA sniff packet [...]

@Koenkk Indeed there are some missing messages from coordinator. I observed that and that seemed to happen when I added a 40cm USB cable between laptop and sniffer. The distance to the coordinator is always less than 1m. Now I moved the sniffer a bit around and added a ferrite coil over the USB cable, same way that I did for coordinator. Seems to be better now.

But still what are those Ack messages with no Source ande Dest addr?

Here are ZHA and Z2M pcaps again. Like in the previous capture, the interview works with Z2M first time after ZHA.
ZHA
http://paste.c-net.org/MalakaiCasbah
Z2M
http://paste.c-net.org/CuratorSweatin

Koenkk, 2021-01-31:
Thanks, can you now also make a sniff of a failed attempt, looking at https://github.com/Koenkk/zigbee2mqtt/issues/5814#issuecomment-768559126 the coordinator requests are also missing there.

tetsuo55, 2021-01-31:
I can confirm that distance is not the issue, it won't pair when held against the antenna of the stick.
You do have to be within a few meters for pairing tho

Once connected to the hub once the reset behavior is slightly different and it pairs normally

hansgans007, 2021-01-31:
So we need a hub from lidl ?

tetsuo55, 2021-01-31:
> So we need a hub from lidl ?

Well the goal is to make it work without the hub i assume

BogdanDIA, 2021-01-31:
Ok, so here it is the new failing sniff. The sniff contains first the moment when I reset the device (in this case the Z2M takes nicely the device out - device_removed). It is obvious that "End Device Timeout Request" does not get the Response. I thinks I have seen it other times but also I have seen failed pairing when there is a response. But let's see.

http://paste.c-net.org/HotelsMarcie

Since I tried another pair after the failed one, this time it is a success. The "End Device Timeout Request" is still not having a response.

http://paste.c-net.org/PerfumeFives

Koenkk, 2021-02-01:
@BogdanDIA do you see a link between the following request being in the sniff? The device asks for an image but z2m does not answer yet because interview has no completed yet. I only see it in the failed sniff (http://paste.c-net.org/HotelsMarcie), it's not in the success sniff: http://paste.c-net.org/PerfumeFives

[image]

Koenkk, 2021-02-01:
@BogdanDIA I don't see where it goes wrong yet. Can you also try updating to the latest firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin . Otherwise I will check if my Lidl has this device on Sunday (going for groceries anyway)

BogdanDIA, 2021-02-01:
I checked the diff between ZHA and Z2M and here it is my observation:
...
Confirm Key SUCCESS
On ZHA:
ZC: Node Descriptor Request
ZED: Node Descriptor Response
ZC: Active Endpoint Request
ZED: Active Endpoint Response
**ZED: Permit Join Request**
ZC: Simple Descriptor Request
ZED: Simple Descriptor Response
...

On failed Z2M:
ZED: Node Descriptor Request
C: Simple Descriptor Request
ZED: Simple Descriptor Response
C: Node Descriptor Response
**Missing Permit Join Request** 
Transport Key
Verify Key - why after Confirm Key SUCCESS

On successful Z2M:
There is a Permit Join Request
 
So int the latest firmware we have 5dBm, that is good :)

BogdanDIA, 2021-02-02:
FYI, I did an update to the coordinator with the latest firmware. Nothing changed.

Koenkk, 2021-02-07:
Bought the door sensor and managed to reproduce the issue. Seems it doesn't happen with the CC2531, only CC2652. 

I've managed to fix it by adding a simple sleep before starting the interview. But since this affect the pairing of all devices I need to investigate it more. First I would like to check if this also fixes the pairing problems at your side @BogdanDIA .

Can you edit `node_modules/zigbee-herdsman/dist/controller/model/device.js` and change `interviewInternal() {` like this:

```js
    interviewInternal() {
        return __awaiter(this, void 0, void 0, function* () {
            const nodeDescriptorQuery = () => __awaiter(this, void 0, void 0, function* () {
                const nodeDescriptor = yield entity_1.default.adapter.nodeDescriptor(this.networkAddress);
                this._manufacturerID = nodeDescriptor.manufacturerCode;
                this._type = nodeDescriptor.type;
                this.save();
                debug.log(`Interview - got node descriptor for device '${this.ieeeAddr}'`);
            });
            yield utils_1.Wait(5000); // <--- ADD THIS LINE
            let gotNodeDescriptor = false;
            for (let attempt = 0; attempt < 6; attempt++) {
                try {
                    yield nodeDescriptorQuery();
                    gotNodeDescriptor = true;
                    break;
                }
```

Koenkk, 2021-02-07:
Forget my previous comment. Added a fix specifically for TuYa end devices. Both my Lidl motion and door sensor pair fine now on first try (tried multiple times).

Fix is in the dev branch now. Please confirm if it works correctly now @tetsuo55  @BogdanDIA 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

rientek, 2021-02-07:
I had the same/similar issues with my Lidl Smart Motion Sensor HG06335 connected to a Raspberry Pi and decided to set up a clean environment on my Raspberry Zero with latest software. After struggling the whole day and creating herdsman debug logging I noticed this was fixed a few hours ago. Installed the latest dev branch and all is working fine.
The Smart Home Socket HG06337 is working fine too.
Thanks Koen!

BogdanDIA, 2021-02-08:
@Koenkk I did an update of the HA add-on today and checked with 3x Lidl switches and 1x Motion sensor. All paired from the first try.
As a radio guy, I can tell that my gut feeling is that CC2652 has some radio issues, distorted signal, something like that.

Ok, so all good for now, good job !

Koenkk, 2021-02-08:
@BogdanDIA 

> As a radio guy, I can tell that my gut feeling is that CC2652 has some radio issues, distorted signal, something like that.

I don't think this is the case, I believe the CC2652 is responding a bit too fast for the lidl smart sensor and therefore the wait helps for the sensor to settle. But good to hear this is solved!

closed 2021-02-08 by Koenkk