Koenkk/zigbee2mqtt #5825: Unable to pair OSRAM AC03645 light bulb

pawelkw, 2021-01-19:
## What happened
I'm unable to pair my new [AC03645](https://www.zigbee2mqtt.io/devices/AC03645.html) bulbs. There is zero logs from the bulbs in debug log level. Only way i can get any exchange going is when using Touchlink. Device is found via Touchlink and I can perform a factory reset, after which i get a success log and the confirmation blinks on the bulb. Unfortunately the device never joins the network. Same after manual reset (via on-off cycles).
**Things i checked:**
- permit join is enabled
- tried two separate bulbs, so its not an issue with a single unit
- device is close to the dongle
- rebooted host PC
- unplugged dongle, restarted zigbee2mqtt, plugged back in
- followed pretty much all steps applicable to my setup from https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

I already have 6 other devices (all Xiaomi) on the network which work great.

## What did you expect to happen
Expected the bulb to join the network after a factory reset.

## How to reproduce it (minimal and precise)
Factory reset the bulb. Wait for it to join the network.

## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware:CC2530
Adapter firmware version:
```
Zigbee2MQTT:info  2021-01-18 21:26:12: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"1.17.0","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.17.0"}'
```

Koenkk, 2021-01-20:
Did you also reflash your CC2530? If that doesn't help could you sniff the traffic when trying to pair: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

nikos1988, 2021-01-21:
I am having a similar problem trying to pair a Lidl HG06336 or an Aqara Door/Window sensor.
I have a slaesh and zzh stick and tried both. permit_join is true but there are absolutely zero log entries when i put the sensors into pair mode. I am also using a powered hub.

PawelKwiecinski, 2021-01-31:
I have solved it. No hardware or software changes have been done to the Zigbee stick. I have set up a fresh zigbee2mqtt docker container. That alone did not solve it though. After that I've decided to change then `pan_id` to 6711 and `channel` to 15 (just to move away from the default values). After this change the bulbs joined my network without any trouble:

```
Zigbee2MQTT:info  2021-01-31 16:06:41: Zigbee: allowing new devices to join.

Zigbee2MQTT:info  2021-01-31 16:06:41: Connecting to MQTT server at mqtt://192.168.0.111:1883

Zigbee2MQTT:info  2021-01-31 16:06:41: Connected to MQTT server

Zigbee2MQTT:info  2021-01-31 16:06:41: MQTT publish: topic 'zigbee2mqtt_docker/bridge/state', payload 'online'

Zigbee2MQTT:info  2021-01-31 16:06:41: MQTT publish: topic 'zigbee2mqtt_docker/bridge/config', payload '{"commit":"1bccc5d","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":15,"extendedPanID":"0xdddddddddddddddd","panID":6711},"permit_join":true,"version":"1.15.0"}'

Zigbee2MQTT:info  2021-01-31 16:07:02: Device '0x7cb03eaa0a074780' joined

Zigbee2MQTT:info  2021-01-31 16:07:02: Starting interview of '0x7cb03eaa0a074780'

Zigbee2MQTT:info  2021-01-31 16:07:02: MQTT publish: topic 'zigbee2mqtt_docker/bridge/log', payload '{"message":{"friendly_name":"0x7cb03eaa0a074780"},"type":"device_connected"}'

Zigbee2MQTT:info  2021-01-31 16:07:02: MQTT publish: topic 'zigbee2mqtt_docker/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x7cb03eaa0a074780"},"type":"pairing"}'

Zigbee2MQTT:info  2021-01-31 16:07:02: Device '0x7cb03eaa0a060635' joined

Zigbee2MQTT:info  2021-01-31 16:07:02: Starting interview of '0x7cb03eaa0a060635'

Zigbee2MQTT:info  2021-01-31 16:07:02: MQTT publish: topic 'zigbee2mqtt_docker/bridge/log', payload '{"message":{"friendly_name":"0x7cb03eaa0a060635"},"type":"device_connected"}'

Zigbee2MQTT:info  2021-01-31 16:07:02: MQTT publish: topic 'zigbee2mqtt_docker/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x7cb03eaa0a060635"},"type":"pairing"}'

Zigbee2MQTT:info  2021-01-31 16:07:03: MQTT publish: topic 'zigbee2mqtt_docker/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x7cb03eaa0a060635"},"type":"device_announced"}'

Zigbee2MQTT:info  2021-01-31 16:07:03: MQTT publish: topic 'zigbee2mqtt_docker/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x7cb03eaa0a074780"},"type":"device_announced"}'

Zigbee2MQTT:info  2021-01-31 16:07:15: Successfully interviewed '0x7cb03eaa0a074780', device has successfully been paired

Zigbee2MQTT:info  2021-01-31 16:07:15: Device '0x7cb03eaa0a074780' is supported, identified as: OSRAM LIGHTIFY LED CLA60 E27 RGBW (AC03645)

Zigbee2MQTT:info  2021-01-31 16:07:15: MQTT publish: topic 'zigbee2mqtt_docker/bridge/log', payload '{"message":"interview_successful","meta":{"description":"LIGHTIFY LED CLA60 E27 RGBW","friendly_name":"0x7cb03eaa0a074780","model":"AC03645","supported":true,"vendor":"OSRAM"},"type":"pairing"}'

Zigbee2MQTT:info  2021-01-31 16:07:16: Successfully interviewed '0x7cb03eaa0a060635', device has successfully been paired

Zigbee2MQTT:info  2021-01-31 16:07:16: Device '0x7cb03eaa0a060635' is supported, identified as: OSRAM LIGHTIFY LED CLA60 E27 RGBW (AC03645)

Zigbee2MQTT:info  2021-01-31 16:07:16: MQTT publish: topic 'zigbee2mqtt_docker/bridge/log', payload '{"message":"interview_successful","meta":{"description":"LIGHTIFY LED CLA60 E27 RGBW","friendly_name":"0x7cb03eaa0a060635","model":"AC03645","supported":true,"vendor":"OSRAM"},"type":"pairing"}'```

closed 2021-03-11 by github-actions[bot]