Koenkk/zigbee2mqtt #5826: Question about "devices" schema: What are the different values of "access" and what do they mean?

tom-sherman, 2021-01-19:
I'm writing a GraphQL service in front of Zigbee2MQTT and this has slowed me down a little. By and large, the schema for the messages published in the `bridge/devices` topic are very intuitive, but `access` is quite opaque.

What are the possible values? What do they mean?

I've been able to guess at a couple, but I could be wrong here

- 1 = Read only
- 2 = Write only
- 7 = Read and write

sjorge, 2021-01-19:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/0676379ae7ca76e963fc4f13ce2d367d72cb26eb/lib/exposes.js#L365

Although I’m not sure what the difference between get/set and state_get/state_set are.

tom-sherman, 2021-01-19:
@sjorge Ah they're in a different repo, that's why I couldn't find them - thank you! Would you mind if we leave this open until we figure out the difference? Or would you prefer me to move this issue to that repo?

sjorge, 2021-01-20:
Leave it open here, this is the right place for general issues or questions.

sjorge, 2021-01-20:
https://github.com/Koenkk/zigbee2mqtt.io/blob/05fc32d8e09bf4cc1bccfc4acb8f618ce7ab3a22/docs/information/exposes.md

Thinks I got it

state -> device publish information, you cannot get or set it
set -> you can set the value only
get -> you can get the value only (aka device doesn't publish it on it's own)
state_get -> device publishes the value, but you can also request it
state_set -> device publishes the value, you can also set it (but cannot get it)
all -> device publishes the value, you can set it, you can also get it.

tom-sherman, 2021-01-20:
Thanks sooooo much for this! I hope to have time to add this to the docs tonight.

nurikk, 2021-01-21:
The access property is a 3-bit bitmask.

Bit 1: The property can be found in the published state of this device.
Bit 2: The property can be set with a /set command
Bit 3: The property can be retrieved with a /get command (when this bit is true, bit 1 will also be true)

closed 2021-03-01 by github-actions[bot]