Koenkk/zigbee2mqtt #5828: Support for Gledopto GL-C-007P

rlesniak, 2021-01-20:
## Is your feature request related to a problem? Please describe
Yes, I cant pair my RGBW controller because is unsupported. Model Gledopto GL-C-007P

## Describe the solution you'd like
I see many supported devices from Gledopto family, especially from 007 series, but not for GL-C-007P. I bet that its not that hard. Its typical RGBW controller.

## Additional context
[image]

Koenkk, 2021-01-20:
Added.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-20 by Koenkk

rlesniak, 2021-01-20:
I am afraid that it wont work :( 
I couldn't wait to update dev branch, so I created (I believe it works the same way as updated devices.js) `external_converters` with that content
```js
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');

const fz = zigbeeHerdsmanConverters.fromZigbeeConverters;
const tz = zigbeeHerdsmanConverters.toZigbeeConverters;
const exposes = zigbeeHerdsmanConverters.exposes;

const e = exposes.presets;

function replaceInArray(arr, oldElements, newElements) {
  const clone = [...arr];
  for (let i = 0; i < oldElements.length; i++) {
    const index = clone.indexOf(oldElements[i]);

    if (index !== -1) {
      clone[index] = newElements[i];
    } else {
      throw new Error('Element not in array');
    }
  }

  return clone;
}

const preset = {
  light_onoff_brightness_colorxy: () => ({
    exposes: [e.light_brightness_colorxy(), e.effect()],
    fromZigbee: [
      fz.color_colortemp,
      fz.on_off,
      fz.brightness,
      fz.level_config,
      fz.ignore_basic_report,
    ],
    toZigbee: [
      tz.light_onoff_brightness,
      tz.light_color,
      tz.ignore_transition,
      tz.ignore_rate,
      tz.effect,
      tz.light_brightness_move,
      tz.light_brightness_step,
      tz.level_config,
      tz.light_hue_saturation_move,
      tz.light_hue_saturation_step,
    ],
    meta: { configureKey: 2 },
    configure: async (device, coordinatorEndpoint, logger) => {
      console.log(device);
      for (const endpoint of device.endpoints.filter((e) =>
        e.supportsInputCluster('lightingColorCtrl')
      )) {
        try {
          await light.readColorCapabilities(endpoint);
        } catch (e) {
          /* Fails for some, e.g. https://github.com/Koenkk/zigbee2mqtt/issues/5717 */
        }
      }
    },
  }),
};
preset.gledopto = {
  light_onoff_brightness_colorxy: () => ({
    ...preset.light_onoff_brightness_colorxy(),
    toZigbee: replaceInArray(
      preset.light_onoff_brightness_colorxy().toZigbee,
      [tz.light_onoff_brightness, tz.light_color],
      [tz.gledopto_light_onoff_brightness, tz.gledopto_light_color]
    ),
  }),
};

const device = {
  zigbeeModel: ['GL-C-007P'],
  model: 'GL-C-007P',
  vendor: 'Gledopto',
  description: 'Zigbee LED controller RGBW',
  extend: preset.gledopto.light_onoff_brightness_colorxy(),
};

module.exports = device;
```

Device is correctly discovered 
[image]

BUT nothing shows up in Devices tab. 
I only see error message in logs:
```
Zigbee2MQTT:debug 2021-01-20 20:43:46: Received Zigbee message from '0x60a423fffe041dee', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":1,"imageType":0,"manufacturerCode":4098}' from endpoint 11 with groupID 0
Zigbee2MQTT:error 2021-01-20 20:43:46: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'filter' of undefined
    at Receive.onZigbeeEvent (/app/lib/extension/receive.js:116:65)
    at Controller.callExtensionMethod (/app/lib/controller.js:381:44))
```

Koenkk, 2021-01-20:
To be sure, please wait till this in the dev branch (try in 3 hours from now) and provide the log if it still fails with that.

rlesniak, 2021-01-21:
@Koenkk how things are going?

Koenkk, 2021-01-21:
it's in the dev branch now, can you test? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

rlesniak, 2021-01-21:
Are you sure it is on dev? last commit has nothing to do with this device (https://github.com/Koenkk/zigbee2mqtt/commit/e65d6602366de4f5a96b71e0079ceeb7ad326bca) 

And indeed, still `Unsupported` on `edge` version

rlesniak, 2021-01-22:
Okay, I have to admit that now it works! 😊 You can merge it to production. Thank you @Koenkk!

Koenkk, 2021-01-22:
Wil be merged to production with next release (1 feb)

blazejosl, 2021-02-14:
Hi Guys.
Few days ago I bought Gledopto GL-C-007P and try to connect it to my smart home installation:
1) CC2531 dongle connected to QNAP
2) zigbee2mqtt runs in a docker
3) mqtt broker runs in a docker
4) openhab 3.0 runs in a docker
Everything configured using text files and works fine (some xiaomi, aqara sensors, smart plugs, hue bulbs connected directly to the CC2531) so far.
Now I try to connect RGBW LED strip via Gledopto GL-C-007P but it is very resistant and does not want to connect.
I've installed latest-dev zigbee2mqtt branch but I don't have any entries to database.db, any entries to devices.yaml.
 Maybe someone can advise me what to do - the device is sort of invisible in my zigbee netowork.

tenkaibr, 2022-03-01:
I'm always getting this type of message even though I have disabled OTA.  Is it normal?

```
Zigbee2MQTT:debug 2022-03-01 17:03:53: Received Zigbee message from 'led_quarto', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":4,"imageType":0,"manufacturerCode":4098}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2022-03-01 17:03:56: Received Zigbee message from 'led_quarto', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":4,"imageType":0,"manufacturerCode":4098}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2022-03-01 17:03:59: Received Zigbee message from 'led_quarto', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":4,"imageType":0,"manufacturerCode":4098}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2022-03-01 17:04:02: Received Zigbee message from 'led_quarto', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":4,"imageType":0,"manufacturerCode":4098}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2022-03-01 17:04:05: Received Zigbee message from 'led_quarto', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":4,"imageType":0,"manufacturerCode":4098}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2022-03-01 17:04:30: Received Zigbee message from 'led_sala', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":4,"imageType":0,"manufacturerCode":4098}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2022-03-01 17:04:33: Received Zigbee message from 'led_sala', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":4,"imageType":0,"manufacturerCode":4098}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2022-03-01 17:04:36: Received Zigbee message from 'led_sala', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":4,"imageType":0,"manufacturerCode":4098}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2022-03-01 17:04:39: Received Zigbee message from 'led_sala', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":4,"imageType":0,"manufacturerCode":4098}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2022-03-01 17:04:42: Received Zigbee message from 'led_sala', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":4,"imageType":0,"manufacturerCode":4098}' from endpoint 11 with groupID 0
```

Koenkk, 2022-03-02:
Yes, disabling OTA will prevent z2m checking for an OTA, but will not prevent the device from requesting it.