Koenkk/zigbee2mqtt #5831: Allow server to be defined in secret.yaml

claha, 2021-01-20:
## Is your feature request related to a problem? Please describe
If I want to make my configuration public (e.g. store it on github) it would be nice to not have to expose the mqtt server (as with the the mqtt user/password).

## Describe the solution you'd like
To be able to write this in my coniguration.
```yaml
mqtt:
  server: '!secret server'
  user: '!secret user'
  password: '!secret password'
```

gsdevme, 2021-01-22:
This would also be very useful for the network_key

gsdevme, 2021-01-22:
Actually just went to look how to implement this and looks like its already there

https://github.com/Koenkk/zigbee2mqtt/blob/07cdc9d549a9db4aa94a259388241b71aec4e12d/lib/util/settings.js#L314-L325

https://www.zigbee2mqtt.io/information/configuration.html#specifying-mqtt-userpassword-and-network_key-in-a-different-file

closed 2021-03-01 by github-actions[bot]