Koenkk/zigbee2mqtt #5832: Pairing IKEA E1743 switch with CC2652RB adapter

americodias, 2021-01-20:
Hi!

I bought a CC2652RB development stick from you to use with zigbee2mqtt at my home. I have another zigbee2mqtt installation at my office (different location) with a CC2531 and never had any problems.

The first device that I tried to pair with your CC2652RB was a IKEA E1743 switch but it simply doesn’t work. I tried and tried and tried, but no success.

Log file:
`info  2021-01-20 13:30:51: Currently 0 devices are joined:
warn  2021-01-20 13:30:51: `permit_join` set to  `true` in configuration.yaml.
warn  2021-01-20 13:30:51: Allowing new devices to join.
warn  2021-01-20 13:30:51: Set `permit_join` to `false` once you joined all devices.
info  2021-01-20 13:30:51: Zigbee: allowing new devices to join.
info  2021-01-20 13:30:51: Connecting to MQTT server at mqtts://mqtt.office.americodias.com:8883
debug 2021-01-20 13:30:51: Using MQTT login with username: mosquitto
info  2021-01-20 13:30:51: Connected to MQTT server
info  2021-01-20 13:30:51: MQTT publish: topic 'z2m/bridge/state', payload 'online'
info  2021-01-20 13:30:51: MQTT publish: topic 'z2m/bridge/config', payload '{"commit":"04c15f7","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20201026,"transportrev":2},"type":"zStack3x0"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6755},"permit_join":true,"version":"1.16.2"}'
info  2021-01-20 13:31:23: Device '0x842e14fffe88716c' joined
info  2021-01-20 13:31:23: Starting interview of '0x842e14fffe88716c'
info  2021-01-20 13:31:23: MQTT publish: topic 'z2m/bridge/log', payload '{"message":{"friendly_name":"0x842e14fffe88716c"},"type":"device_connected"}'
info  2021-01-20 13:31:23: MQTT publish: topic 'z2m/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x842e14fffe88716c"},"type":"pairing"}'
info  2021-01-20 13:31:23: MQTT publish: topic 'z2m/0x842e14fffe88716c/availability', payload 'online'
debug 2021-01-20 13:31:25: Device '0x842e14fffe88716c' announced itself
info  2021-01-20 13:31:25: MQTT publish: topic 'z2m/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x842e14fffe88716c"},"type":"device_announced"}'
warn  2021-01-20 13:31:33: Device '0x842e14fffe88716c' left the network
info  2021-01-20 13:31:33: MQTT publish: topic 'z2m/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x842e14fffe88716c"},"type":"device_removed"}'
info  2021-01-20 13:32:46: Device '0x842e14fffe88716c' joined
info  2021-01-20 13:32:46: Starting interview of '0x842e14fffe88716c'
info  2021-01-20 13:32:46: MQTT publish: topic 'z2m/bridge/log', payload '{"message":{"friendly_name":"0x842e14fffe88716c"},"type":"device_connected"}'
info  2021-01-20 13:32:46: MQTT publish: topic 'z2m/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x842e14fffe88716c"},"type":"pairing"}'
debug 2021-01-20 13:32:48: Device '0x842e14fffe88716c' announced itself
info  2021-01-20 13:32:48: MQTT publish: topic 'z2m/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x842e14fffe88716c"},"type":"device_announced"}'
warn  2021-01-20 13:32:57: Device '0x842e14fffe88716c' left the network
info  2021-01-20 13:32:57: MQTT publish: topic 'z2m/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x842e14fffe88716c"},"type":"device_removed"}'
info  2021-01-20 13:33:28: Device '0x842e14fffe88716c' joined
info  2021-01-20 13:33:28: Starting interview of '0x842e14fffe88716c'
info  2021-01-20 13:33:28: MQTT publish: topic 'z2m/bridge/log', payload '{"message":{"friendly_name":"0x842e14fffe88716c"},"type":"device_connected"}'
info  2021-01-20 13:33:28: MQTT publish: topic 'z2m/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x842e14fffe88716c"},"type":"pairing"}'
debug 2021-01-20 13:33:30: Device '0x842e14fffe88716c' announced itself
info  2021-01-20 13:33:30: MQTT publish: topic 'z2m/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x842e14fffe88716c"},"type":"device_announced"}'
warn  2021-01-20 13:33:38: Device '0x842e14fffe88716c' left the network`

(and it continues... connecting and disconnecting.)

I tried to pair the same E1743 switch in my office setup and everything worked fine at first attempt. Then I removed the device, and switched the CC2531 by the CC2652RB adapter and it doesn't work (as shown in the log file).

The log file was obtained in my office install (1.16.2) but at home with the latest zigbee2mqtt version I get the same thing.

Yesterday I even flashed the stick firmware using the CC2652RB_20201026.zip.

Kind regards,
Américo

Koenkk, 2021-01-20:
Did you bring down your complete old network before starting the CC2652RB stick? Make sure to power off all routers and change the pan_id: https://www.zigbee2mqtt.io/information/FAQ.html#how-do-i-migrate-from-a-cc2531-to-a-more-powerful-coordinator-eg-zzh

americodias, 2021-01-20:
Hi Koenkk,

Thanks for your response. Yes, I did that.

Thank you,
Américo

Koenkk, 2021-01-20:
Did you also replace the battery of your E1743? Make sure to check everything of https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

If that doesn't help you have to sniff the traffic when trying to pair: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

americodias, 2021-01-22:
Hi again,

I have 4 different E1743 and I have the same problem with all of them. All batteries are new.

I did everything I could, but at the moment I don't have a spare usb zigbee stick to sniff the traffic (I ordered one, but it will probably take a while to arrive).

I start to think it is a problem in the coordinator (CC2652RB from slae.sh). Yesterday after dozen of tries, I decided to take of the antenna and finally was able to pair the E1743! The next time I tried it was not working again...

As I said, with the CC2531 I don't have any problem. I will try to send the CC2652RB back.

Thanks for your help,
Américo

slaesh, 2021-01-22:
Do you have other devices as well or just the IKEA stuff?

Main difference between those dongles is the ZigBee version:

CC2531: 3.0.X
CC2652: 3.X.X

americodias, 2021-01-22:
Hi Slaesh,

I also have some Xioumi sensors.
I will try to pair one of them - when I have time - and I will post the result here. Good idea!

Kind regards,
Americo

americodias, 2021-01-26:
Hi again,

Just a short update:

I have tried to use raspberrypi with a fresh OS install.
Now I can pair the E1743. It still leave the network 2 or 3 times during the interview but at the end the interview is successful.
I also tried to pair a Xioumi temperature sensor (WSDCGQ11LM) and it pairs almost immediately.
I will continue to debug this issue.

Kind regards,
Américo

MattWestb, 2021-01-26:
@americodias The 1743 is one zigbee 3 device and its doing one announcement then joining the network without having network or trustcenter key. Then getting the network key one new TC key and rebooting its doing one more device announcement and one verifying the new key is ok that z2m is inprenting as one device accounstment = zigbee 3.

Old ZHA devices is normally only getting on standard network key and dont do more security unpatings = Xiaomi devices that is not zigbee 3.

Good that you have installing one clean raspbian and not getting more problem in the future.

closed 2021-03-05 by github-actions[bot]

realalexandergeorgiev, 2021-11-04:
I can confirm that the E1473 still doesn't pair with a CC2652 chip (everything updated). It worked with the CC2531. In my network this E1473 is the only device not working, but most (all?) devices aren't Zigbee 3.0.

balarpan, 2021-11-24:
Bought yesterday E1743. After pairing procedure is over and interview completed my Z2M says that this is Unknown model of IKEA Manufacturer.
I was try many times to delete device and repair them again. Same result.
Z2M 1.22.0 with CC2652 as a coordinator. 
I already have E1743 in my network (paired about 3 months ago)  and this device works fine. Both displays the same Firmware build date and Firmware version

I don't know what else to try.
Could it be caused by some changes in z2m compared to previous versions?

tb205gti, 2021-12-25:
Same issue here, cannot pair an E1743 with the zig-a-zig-ah! as coordinator (CC2652)

boulete31, 2022-01-08:
Same issue here, cannot pair an E1743

tedzhouhk, 2022-10-01:
I am having random experience with Tubes zigbee hub (CC2562p). At first I was able to pair 3 E1743s normally without any issue. Several months later I decided to add some more E1743s, but I was not able to pair any of them following the normal pairing process, all saying device left several seconds after the interview start. Thankfully, the solution to remove antenna and put them in touch works.