Koenkk/zigbee2mqtt #5839: Lidl HG06492C Offline and not able to change state

kcvanderlinden, 2021-01-20:
## What happened
recently bought the Lidl HG06492C Led lamp, but I have not been able to get it working properly. It has been recognised fully.

Led lamp seems to go offline after some second of turning power on. It is online for a brief second and functions, because any mqtt message send from home assistant prior to powering the led lamp, is processed (lamp thus changes brightness, state and colour temprature). But after some seconds after powering, it goes offline. Hereby I can't change the state or any other parameter of the led lamp anymore. By turning on the availability_timeout in the advanced configuration, i can see that the device is offline when the errors appear, but the problem also appears without the availability_timeout.
`Jan 20 17:48:12 raspberrypi npm[16060]: Zigbee2MQTT:error 2021-01-20 17:48:12: Publish 'set' 'state' to '0x842e14fffe2a418c' failed: 'Error: Command 0x842e14fffe2a418c/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 38603 - 1 - 3 - 6 - 11 after 10000ms)'
Jan 20 17:48:12 raspberrypi npm[16060]: Zigbee2MQTT:info  2021-01-20 17:48:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x842e14fffe2a418c' failed: 'Error: Command 0x842e14fffe2a418c/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 38603 - 1 - 3 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x842e14fffe2a418c"},"type":"zigbee_publish_error"}'`

I have tried to use an extension cable for my cc2531 adapter, but it yielded no different result.

I have 1 other zigbee device connected (MiJia door & window contact sensor), but that one has been working fin for a couple of months now without any problems.

## Debug info
Zigbee2MQTT version: 1.17.0-dev (But also tried non-dev)
Adapter hardware: CC2531
Adapter firmware version: zStack12 ("revision":20190608,"transportrev":2)

BlackPantherSF, 2021-01-24:
I have the same problem. I bought 2 RGB bulbs recently (HG06106C - Lidl Livarno Lux E27 bulb RGB) and both have this issue. I can control light (on/off, brightness, RGB, color temp), but it takes approximate 10s for every command to execute. Here is the error from log for on/off command (get similar error for other types of command):

Jan 24 19:07:47 raspberrypi npm[4085]: Zigbee2MQTT:info  2021-01-24 19:07:47: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"07cdc9d","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20201026,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.17.0"}'
Jan 24 19:09:39 raspberrypi npm[4085]: Zigbee2MQTT:error 2021-01-24 19:09:39: Publish 'set' 'state' to 'livarnolux_bulb_01' failed: 'Error: Command 0x588e81fffeb96f89/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 18306 - 1 - 3 - 6 - 11 after 10000ms)'
Jan 24 19:09:39 raspberrypi npm[4085]: Zigbee2MQTT:info  2021-01-24 19:09:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'livarnolux_bulb_01' failed: 'Error: Command 0x588e81fffeb96f89/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 18306 - 1 - 3 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"livarnolux_bulb_01"},"type":"zigbee_publish_error"}'
Jan 24 19:10:02 raspberrypi npm[4085]: Zigbee2MQTT:error 2021-01-24 19:10:02: Publish 'set' 'state' to 'livarnolux_bulb_01' failed: 'Error: Command 0x588e81fffeb96f89/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 18306 - 1 - 4 - 6 - 11 after 10000ms)'
Jan 24 19:10:02 raspberrypi npm[4085]: Zigbee2MQTT:info  2021-01-24 19:10:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'livarnolux_bulb_01' failed: 'Error: Command 0x588e81fffeb96f89/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 18306 - 1 - 4 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"livarnolux_bulb_01"},"type":"zigbee_publish_error"}'

I use same Zigbee2MQTT version(1.17.0)(commit #07cdc9d) and different adapter zzh! - CC26X2R1.
I also tried with 3 m USB extension cable and result is the same.

gluck, 2021-02-03:
Same issue, using CC2530+CC2591:
`get` works fine:
```
Received MQTT message on 'zigbee2mqtt/led_white_2/get' with data '{"state":""}'
Publishing get 'get' 'state' to 'led_white_2'
Received Zigbee message from 'led_white_2', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
MQTT publish: topic 'zigbee2mqtt/led_white_2', payload '{"brightness":63,"color_temp":153,"linkquality":0,"state":"ON"}'
```
`set` commands all fail (first one update the bulb immediately, with mqtt failure after 10s, if you push more commands they get queued up and each apply after previous ones - each taking 10s timeout, not sure if that's clear)
```
Received MQTT message on 'zigbee2mqtt/led_white_2/set' with data '{"brightness":127.5}'
Publishing 'set' 'brightness' to 'led_white_2'
Publish 'set' 'brightness' to 'led_white_2' failed: 'Error: Command 0x588e81fffeb3c006/1 genLevelCtrl.moveToLevelWithOnOff({"level":127.5,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 20336 - 1 - 6 - 8 - 11 after 10000ms)'
Error: Command 0x588e81fffeb3c006/1 genLevelCtrl.moveToLevelWithOnOff({"level":127.5,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 20336 - 1 - 6 - 8 - 11 after 10000ms) at Timeout._onTimeout (/zigbee2mqtt-1.17.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35) at listOnTimeout (internal/timers.js:554:17) at processTimers (internal/timers.js:497:7)
```

Not sure how I can troubleshoot on my end (I don't have another zigbee stick to sniff, I don't have LIDL hub either).
Same behavior on 2 (HG06492C) bulbs so they're unlikely to be faulty.
I bought 2 RGB bulbs (HG06106C) and these work fine.

rmickael62, 2021-02-25:
Same problem with Lidl HG06492B
I have 48 other device on the zigbee network which works normally .

`
Error Publish 'set' 'state' to '0x588e81fffe9c5096' failed: 'Error: Command 0x588e81fffe9c5096/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 21024 - 1 - 95 - 6 - 11 after 10000ms)'
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x588e81fffe9c5096' failed: 'Error: Command 0x588e81fffe9c5096/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 21024 - 1 - 95 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x588e81fffe9c5096"},"type":"zigbee_publish_error"}'
`

I bought 6 RGB bulbs (HG06106A) and these work fine.

rmickael62, 2021-02-25:
sniff traffic zigbee : ("0x07de" following an "ON" command ) 
I am available for any other procedure that may solve this problem. 
[Wireshark_25_02_21.zip](https://github.com/Koenkk/zigbee2mqtt/files/6043359/Wireshark_25_02_21.zip)

closed 2021-04-05 by github-actions[bot]