Koenkk/zigbee2mqtt #5843: Please add device ZigBee 3 Gang Remote

electro-nic, 2021-01-20:
## Information about the device + link
Tuya Smart Life ZigBee Wireless Wall Switch 3 Gangs Remote
Model No. SS9300ZB
https://aliexpress.ru/item/1005001942130676.html
https://www.loratap.com
## data/database.db entry of the device
`{"id":5,"type":"EndDevice","ieeeAddr":"0x847127fffe0809cb","nwkAddr":54092,"manufId":4098,"epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,1,6],"outClusterList":[25,10],"clusters":{},"binds":[],"meta":{}},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"3":{"epId":3,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1611168375354}`
[image]
[image]
[image]

Can i help other information?

electro-nic, 2021-01-30:
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/devices', payload '[{"definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"11":{"bindings":[],"clusters":{"input":["ssIasAce"],"output":["ssIasZone","ssIasWd"]},"configured_reportings":[]},"110":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"12":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"13":{"bindings":[],"clusters":{"input":["genOta"],"output":[]},"configured_reportings":[]},"2":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"242":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"3":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"4":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"47":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"5":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"6":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"8":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]}},"friendly_name":"Coordinator","ieee_address":"0x00124b001949adfe","interview_completed":true,"interviewing":false,"network_address":0,"supported":false,"type":"Coordinator"},{"definition":{"description":"MiJia wireless switch","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["single","double","tripple","quadruple","hold","release"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"WXKG01LM","vendor":"Xiaomi"},"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]}},"friendly_name":"buttonmijia1","ieee_address":"0x00158d00024a41d4","interview_completed":true,"interviewing":false,"model_id":"lumi.sensor_switch","network_address":26461,"power_source":"Battery","supported":true,"type":"EndDevice"},{"date_code":"20161128","definition":{"description":"Aqara door & window contact sensor","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Indicates if the contact is closed (= true) or open (= false)","name":"contact","property":"contact","type":"binary","value_off":true,"value_on":false},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"MCCGQ11LM","vendor":"Xiaomi"},"endpoints":{"1":{"bindings":[],"clusters":{"input":["genBasic","genIdentify","genOnOff"],"output":["genBasic","genGroups"]},"configured_reportings":[]}},"friendly_name":"indoorlock","ieee_address":"0x00158d0002e32b0c","interview_completed":true,"interviewing":false,"model_id":"lumi.sensor_magnet.aq2","network_address":42704,"power_source":"Battery","software_build_id":"3000-0001","supported":true,"type":"EndDevice"},{"definition":{"description":"Aqara human body movement and illuminance sensor","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Indicates whether the device detected occupancy","name":"occupancy","property":"occupancy","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Measured illuminance in lux","name":"illuminance_lux","property":"illuminance","type":"numeric","unit":"lx"},{"access":1,"description":"Measured illuminance in lux","name":"illuminance","property":"illuminance","type":"numeric","unit":"lx"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"RTCGQ11LM","vendor":"Xiaomi"},"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]}},"friendly_name":"sensormovebody","ieee_address":"0x00158d0005cf50a4","interview_completed":true,"interviewing":false,"model_id":"lumi.sensor_motion.aq2","network_address":57445,"power_source":"Battery","supported":true,"type":"EndDevice"},{"date_code":"","definition":{"description":"Wireless switch with 3 buttons","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["1_single","1_double","1_hold","2_single","2_double","2_hold","3_single","3_double","3_hold"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0043","vendor":"TuYa"},"endpoints":{"1":{"bindings":[],"clusters":{"input":["genBasic","genPowerCfg","genOnOff"],"output":["genOta","genTime"]},"configured_reportings":[]},"2":{"bindings":[],"clusters":{"input":["genPowerCfg","genOnOff"],"output":[]},"configur [... cut ...] ":"zigbee2mqtt/WallSwitch3gang/action","type":"action"}'
DEBUG: Received MQTT message on 'homeassistant/device_automation/0x847127fffe0809cb/action_2_double/config' with data '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0x847127fffe0809cb"],"manufacturer":"TuYa","model":"Wireless switch with 3 buttons (TS0043)","name":"WallSwitch3gang","sw_version":"Zigbee2MQTT 1.17.0"},"payload":"2_double","subtype":"2_double","topic":"zigbee2mqtt/WallSwitch3gang/action","type":"action"}'

electro-nic, 2021-01-30:
no information about battery

johanvh83, 2021-02-12:
This 3-button remote control uses the same TYZS3 module
https://www.loratap.com/products/tuya-smart-life-zigbee-smart-home-wireless-switch-3-gangs-remote-tuya-zigbee-hub-required-no-limit-to-control-home-device
I could bind the remote/module and see the state of the buttons in the zigbee2mqtt logs but don't know how to proceed

closed 2021-03-23 by github-actions[bot]

Tontze, 2021-12-28:
https://www.becasmart.com/product/beca-bht-001-zigbee-voice-control-water-heating-water-boiler-gas-boiler-electric-heating-room-thermostat-support-online-purchase/

This device has also TYZS3 module