Koenkk/zigbee2mqtt #5844: fresh CC2652RB coordinator: app fails to start

dakky, 2021-01-20:
# Bug Report
## What happened
I have a brand new zigbee controller: [zigbee2mqtt doc](https://www.zigbee2mqtt.io/information/supported_adapters.html#slaeshs-cc2652rb-stick-recommended-for-most-users)
Zigbee2mqtt is running as docker container. The device is available in the container and writeable by the user. 

There was ne preexisting zignee network or anyhting like this. practically a green field ;)

Startup fails with:
```
zigbee2mqtt    | Zigbee2MQTT:error 2021-01-20 21:47:28: Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel
zigbee2mqtt    |     at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:172:23
zigbee2mqtt    |     at Generator.throw (<anonymous>)
zigbee2mqtt    |     at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:25:65)
zigbee2mqtt    | npm ERR! code ELIFECYCLE
zigbee2mqtt    | npm ERR! errno 1
zigbee2mqtt    | npm ERR! zigbee2mqtt@1.17.0 start: `node index.js`
zigbee2mqtt    | npm ERR! Exit status 1
```

I already changed the config to another pan_id. I also updated the firmware with this one: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC2652RB_20201026.zip

App running in debug mode:
https://pastebin.com/KeMCvVup

## What did you expect to happen
app starts without errors

## How to reproduce it (minimal and precise)
-

## Debug Info
Zigbee2MQTT version: 1.17.0 (commit #07cdc9d)
Adapter hardware: CC2652RB
Adapter firmware version: CC2652RB_20201026.hex

Koenkk, 2021-01-20:
- Can you try selecting a different channel?
- What system are you using, are there any other usb devices attached, and is the CC2652 connected through a usb extension cable?

dakky, 2021-01-20:
thanks for the fast feedback.

No, another channel does not change anything (channel: 15)

The host is a raspberrypi4 and yes, there is a ssd attached via usc2sata adapter.
Yes I use a usb extension, although its quite short (10cm)

Koenkk, 2021-01-20:
> The host is a raspberrypi4 and yes, there is a ssd attached via usb2sata adapter.

SSD could be the issue here (have seen this multiple times). Can you try on a different system or when the SSD is connected through a powered USB hub?

dakky, 2021-01-20:
hurm not so very easy. The ssd is the boot device. I will have to think about how to test this without destroying everything.
The power supply itself should be stable enought. It's 3A and there a no low voltage messages in the systemlog/dmesg

Is it a alternative (at least for testing purposes) to plug the CC2652  via powered usb hub to the pi?

dakky, 2021-01-20:
and here we are:
powering the stick via an active usb hub fixes the problem. maybe something for the documentation? at least the CC2652  seems to draw too much power from the usb hub of the raspberry pi4 at least when there is a ssd connected.

I will validate the result against a longer usb extension in order to be sure, that this is not the issue (because at the moment with the powered hub i'm about 1m away from the raspberry)

i'll report the results here

m0wlheld, 2021-01-21:
Is the SSD attached to USB3? I have an RPI4B and had an USB 3.1 pen drive connected to USB 3 port. Never managed to get the Raspbee II module running properly. I was given the hint, that [USB3 has a severe negative effect on 2,4GHz networks](https://www.intel.com/content/www/us/en/products/docs/io/universal-serial-bus/usb3-frequency-interference-paper.html) at close range. After switching back to SD-Card, everything was fine.

So I'd guess it's the distance, not the power consumption.

Koenkk, 2021-01-21:
Good that you fixed this, added it to the FAQ

closed 2021-02-21 by Koenkk

Mega-Touk, 2021-03-30:
I know this is closed, thanks for the fix.

I have an SSD as my boot drive too. and i had the same exact problem when migrating from CC2531 to Slaesh CC2652RB 

Instead of buying expensive usb hub, i ordered a cheap "dual power usb cable" from Aliexpress.
[image]

One usb on the raspberry (power+data) and one on my synology nas (power only). this saved me space, money and one power outlet
I's working great