Koenkk/zigbee2mqtt #5854: Error while starting zigbee-herdsman

benl84, 2021-01-21:
Hi,

can some1 help me please? I'm quite at a loss and I've read all the forum posts and FAQs on this and still it doesnt get better.
I'm running raspien and trying to get zigbee2mqtt to work. Somehow when starting the service I always get errors.
See install errors: [https://pastebin.com/embed_js/Fh6cQFY5](url)
Also i have the log here : [https://pastebin.com/ZhMKmynZ](url)

# Bug Report
## What happened
cant start zigbee2mqtt
## What did you expect to happen
should start?
## How to reproduce it (minimal and precise)
npm start command
## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608

deiger, 2021-01-22:
This has recently started happening to me as well, with repeated crashing:
```log
Zigbee2MQTT:info  2021-01-22 20:21:12: Logging to console and directory: '/share/zigbee2mqtt/log/2021-01-22.20-21-10' filename: log.txt
Zigbee2MQTT:info  2021-01-22 20:21:12: Starting Zigbee2MQTT version 1.17.0 (commit #1.17.0)
Zigbee2MQTT:info  2021-01-22 20:21:12: Starting zigbee-herdsman (0.13.46)
Zigbee2MQTT:error 2021-01-22 20:21:52: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-01-22 20:21:52: Failed to start zigbee
Zigbee2MQTT:error 2021-01-22 20:21:52: Exiting...
Zigbee2MQTT:error 2021-01-22 20:21:52: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 
ERR!
 code ELIFECYCLE
npm
 ERR! errno 1
npm
 ERR! zigbee2mqtt@1.17.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.0 start script.
npm ERR!
 This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm 
ERR!     /root/.npm/_logs/2021-01-22T18_21_57_403Z-debug.log
2021-01-22T20:21:59: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2021-01-22T20:21:59: PM2 log: App [npm:0] starting in -fork mode-
2021-01-22T20:21:59: PM2 log: App [npm:0] online
> zigbee2mqtt@1.17.0 start /zigbee2mqtt-1.17.0
> node index.js
```

Koenkk, 2021-01-22:
> This has recently started happening to me as well, with repeated crashing:

Probably your adapter crashed, try relfashing it

OptionHenrik, 2021-01-22:
Or move it to another USB port ... preferably one that delivers more power. I had the same problem but moving adapter from USB2 to USB3 did the trick. USB3 delivers more power on my raspberry4B.

deiger, 2021-01-22:
Reflashing didn't help, and it was already connected to USB3. I think the adapter might be toast...

m0wlheld, 2021-01-22:
I have exactly the same issue with slaesh's CCRB that came fresh out of the mail. It does not work when using /dev/ttyAMA0 as device / port. It **does** work, if using the /dev/serial/by-id path or the (autocreated) link /dev/ttyUSB0

```bash
$ ls -la /dev/serial/by-id/usb-Silicon_Labs_slae.sh_...-port0
lrwxrwxrwx 1 root root 13 Jan 22 22:44 /dev/serial/by-id/usb-Silicon_Labs_slae.sh_...-port0 -> ../../ttyUSB0
$ ls -la /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 Jan 22 22:44 /dev/ttyUSB0
$ ls -la /dev/ttyAMA0
crw--w---- 1 root tty 204, 64 Dec 25 12:17 /dev/ttyAMA0
```

insipiens, 2021-01-23:
cc2531 can only really handle 10 devices, are you trying to connect more [battery operated devices] than this?

deiger, 2021-01-23:
@insipiens I have more than 10 battery-based devices, but also 5-6 devices that can act as repeaters, so it shouldn't really be the issue. It used to work, up until a couple of days ago.
In any case, I already ordered a CC2652R, as the CC2531 was always flaky.

m0wlheld, 2021-01-23:
> cc2531 can only really handle 10 devices, are you trying to connect more than this? Worth checking just in case.

How is that related to the bug?

m0wlheld, 2021-01-23:
> Reflashing didn't help, and it was already connected to USB3. I think the adapter might be toast...

@deiger Did you try the /dev/serial/by-id path yet?

fbijma2, 2021-01-23:
Had the same issue, after restarting HASSIO. Reflased the CC2652, but that did not help. Then I decided to uninstall zigbee2mqtt saving the configuration and reinstalling it again. This solved my problem. Everything works fine now!

fbijma2, 2021-01-23:
Strange, restarted zigbee2mqtt and now I have the same issue again.

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2021-01-23T16:41:06: PM2 log: Launching in no daemon mode
2021-01-23T16:41:06: PM2 log: App [npm:0] starting in -fork mode-
2021-01-23T16:41:06: PM2 log: App [npm:0] online
> zigbee2mqtt@1.17.0 start /zigbee2mqtt-1.17.0
> node index.js
Zigbee2MQTT:info  2021-01-23 16:41:11: Logging to console and directory: '/share/zigbee2mqtt/log/2021-01-23.16-41-09' filename: log.txt
Zigbee2MQTT:info  2021-01-23 16:41:11: Starting Zigbee2MQTT version 1.17.0 (commit #1.17.0)
Zigbee2MQTT:info  2021-01-23 16:41:11: Starting zigbee-herdsman (0.13.46)
Zigbee2MQTT:error 2021-01-23 16:41:31: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-01-23 16:41:31: Failed to start zigbee
Zigbee2MQTT:error 2021-01-23 16:41:31: Exiting...
Zigbee2MQTT:error 2021-01-23 16:41:31: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 
ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR!
 zigbee2mqtt@1.17.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-01-23T15_41_31_529Z-debug.log
2021-01-23T16:41:34: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2021-01-23T16:41:34: PM2 log: App [npm:0] starting in -fork mode-
2021-01-23T16:41:34: PM2 log: App [npm:0] online
> zigbee2mqtt@1.17.0 start /zigbee2mqtt-1.17.0
> node index.js
Zigbee2MQTT:info  2021-01-23 16:41:38: Logging to console and directory: '/share/zigbee2mqtt/log/2021-01-23.16-41-37' filename: log.txt
Zigbee2MQTT:info  2021-01-23 16:41:38: Starting Zigbee2MQTT version 1.17.0 (commit #1.17.0)
Zigbee2MQTT:info  2021-01-23 16:41:38: Starting zigbee-herdsman (0.13.46)
Zigbee2MQTT:error 2021-01-23 16:41:58: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-01-23 16:41:58: Failed to start zigbee
Zigbee2MQTT:error 2021-01-23 16:41:58: Exiting...
Zigbee2MQTT:error 2021-01-23 16:41:58: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE

fbijma2, 2021-01-23:
OK found the problem. The 3 USB serial devices I have had been mixed up. That means the P1 and RFXTFX and also the Zigbee stick all changed. The P1 was using the same USB device as the zigbee2mqtt. To others, check your USB devices. IN ESPHome you can find a great list of described USB serial devices if you have it installed.

aviogit, 2021-01-25:
Had exactly the same problem until one minute ago and solved it simply loading the `cdc-acm` kernel module with `sudo modprobe cdc-acm`. This brought up `/dev/ttyACM0` as before the last RPI4 kernel upgrade.

joermke, 2021-01-26:
Same thing here, but neiter `modprobe cdc-acm` nor using `/dev/by-id/...`solved this. Occured after updating the raspberry pi a couple of days ago.

deiger, 2021-01-26:
> > Reflashing didn't help, and it was already connected to USB3. I think the adapter might be toast...
> 
> @deiger Did you try the /dev/serial/by-id path yet?

It was already by-id. Eventually I just switched to a new CC2652R, and everything works splendid.

Andreasschweiger, 2021-01-29:
I had the same Problem. Linking the /dev/serial/by-id path solved it.

joermke, 2021-02-02:
Now solved for me, too. I had to do several things, though:

- Update to latest firmware (CC2531)
- Change device to `/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0014D97638-if00` in both, config file and my docker startup command `docker run ... --device=...`

insipiens, 2021-02-03:
So is the thinking here that a Pi kernel update might include an update to the usb chip and that could have caused the problem? If so then I have a drawer full of perfectly good cc2531 which I thought were toast 😖

flatsiedatsie, 2021-02-03:
Strange, for me it's the opposite. On Raspbian I can get Zigbee2MQTT to connect if I use something like `/dev/ttyAMA0` but not if I use `/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2173476-if00`. Then I get the error:

```
Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
2021-02-03 11:52:24.052 INFO   : zigbee2mqtt-adapter:     at ZStackAdapter.<anonymous> (/home/pi/.webthings/data/zigbee2mqtt-adapter/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
2021-02-03 11:52:24.053 INFO   : zigbee2mqtt-adapter:     at Generator.throw (<anonymous>)
2021-02-03 11:52:24.054 INFO   : zigbee2mqtt-adapter:     at rejected (/home/pi/.webthings/data/zigbee2mqtt-adapter/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
```

arnaud-gouder, 2021-02-28:
I have the same problem right now as well. Here is the log:

```
Zigbee2MQTT:info  2021-02-28 19:44:41: Logging to console and directory: '/app/data/log/2021-02-28.19-44-41' filename: log.txt,
Zigbee2MQTT:debug 2021-02-28 19:44:41: Removing old log directory '/app/data/log/2021-02-28.19-31-19',
Zigbee2MQTT:info  2021-02-28 19:44:41: Starting Zigbee2MQTT version 1.17.1 (commit #99274c7),
Zigbee2MQTT:info  2021-02-28 19:44:41: Starting zigbee-herdsman (0.13.59),
Zigbee2MQTT:debug 2021-02-28 19:44:41: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6757},"serialPort":{"path":"/dev/zigbee2mqtt"}}',
2021-02-28T18:44:41.721Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[182,210,165,214,37,2,179,116,214,90,240,168,82,145,32,66],"panID":6757,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/zigbee2mqtt"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}',
2021-02-28T18:44:41.722Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/zigbee2mqtt and {"baudRate":115200,"rtscts":false,"autoOpen":false},
2021-02-28T18:44:41.724Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened,
2021-02-28T18:44:41.733Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0],
2021-02-28T18:44:41.733Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0],
2021-02-28T18:44:42.037Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0],
2021-02-28T18:44:42.038Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0,0],
2021-02-28T18:44:42.179Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload,
2021-02-28T18:44:42.180Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239],
2021-02-28T18:44:43.181Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1},
2021-02-28T18:44:43.182Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32],
2021-02-28T18:44:49.186Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1},
2021-02-28T18:44:49.186Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32],
2021-02-28T18:44:55.190Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1},
2021-02-28T18:44:55.190Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32],
Zigbee2MQTT:error 2021-02-28 19:45:01: Error while starting zigbee-herdsman,
Zigbee2MQTT:error 2021-02-28 19:45:01: Failed to start zigbee,
Zigbee2MQTT:error 2021-02-28 19:45:01: Exiting...,
Zigbee2MQTT:error 2021-02-28 19:45:01: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms),
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31),
    at Generator.throw (<anonymous>),
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65),
npm ERR! code ELIFECYCLE,
npm ERR! errno 1,
npm ERR! zigbee2mqtt@1.17.1 start: `node index.js`,
npm ERR! Exit status 1,
```

I am afraid my Slaesh CC2652RB USB stick might have died... I did check, and I'm 100% sure I'm connecting to the correct device! The CP210x is detected perfectly by my kernel, and the /dev/ttyUSB0 device pops up correctly (using Ubuntu 20.04).
This is waht I see in my kernel messages when I plug in the stick:
```
usb 1-6: new full-speed USB device number 10 using xhci_hcd
[10471.399469] usb 1-6: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[10471.399474] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10471.399478] usb 1-6: Product: slae.sh cc2652rb stick - slaesh's iot stuff
[10471.399480] usb 1-6: Manufacturer: Silicon Labs
[10471.399482] usb 1-6: SerialNumber: 00_12_4B_00_21_CC_3F_21
[10471.403370] cp210x 1-6:1.0: cp210x converter detected
[10471.406238] usb 1-6: cp210x converter now attached to ttyUSB0
```

fbijma2, 2021-02-28:
Most of the times this happens after a reboot where USB serial devices got
mixed up. Check the names and links of your devices in /dev/serial/by-??
You probably have a new order in your devices.

arnaud-gouder, 2021-02-28:
I really don't think so. As you can see from the kernel log, the ttyUSB0 device is the device that pops up when I plug in the USB stick. This is definitely the only cp210x device I have.

arnaud-gouder, 2021-02-28:
I tried again by reflashing both with the firmware from zigbee2mqtt.io and with slaesh firmware with the 2 leds:

When I connect to the device using miniterm (so 100% the correct device) and I press 'Enter', I see the orange led toggle.

Once I start zigbee2mqtt, both the blue and the orange leds go off.

fbijma2, 2021-02-28:
> at Generator.throw (), at rejected

Sorry, I was too quick answering.

m0wlheld, 2021-03-01:
> I am afraid my Slaesh CC2652RB USB stick might have died...

I am seeing this error when updating to 1.18.0 (see #6517), but going back to 1.17.1 avoids this error without reboot or reflash. Could have been something introduced between 1.17.1 and 1.18.0 (so: in February 2021). I also do use Slae'sh CC2652RB stick.

arnaud-gouder, 2021-03-02:
> I am seeing this error when updating to 1.18.0 (see #6517), but going back to 1.17.1 avoids this error without reboot or reflash. Could have been something introduced between 1.17.1 and 1.18.0 (so: [...]

Hi, Going back to 1.17.1 fixed it for me as well!

Thanks a lot!

Best regards,

Arnaud

oivindoh, 2021-03-24:
FWIW @ Conbee users arriving here; this problem went away for me when I explicitly specified adapter: deconz under serial in configuration.yaml.

denics, 2021-04-13:
> FWIW @ Conbee users arriving here; this problem went away for me when I explicitly specified adapter: deconz under serial in configuration.yaml.

Thanks @oivindoh I think this should be added in the docs!

closed 2021-05-22 by github-actions[bot]

theopoon, 2021-09-02:
I had the same issue just this hour, am running zigbee2mqtt directly on a Pi3b with ethernet.

Rebooted a couple of time, no luck, turns out the CC2531 was having issue, it does not show up even on lsusb

hard unplug power, unplug usb, waited 30s and boot up, lsusb shows:
Bus 001 Device 004: ID 0451:16a8 Texas Instruments, Inc.

zigbee2mqtt works again.

I have been running it for a while, all doing well, first time seen this issue.

simfun, 2021-10-31:
Got the same problem after time change because of winter settings automatically!!!! I checked the logs and after 04:00am the HA was unresponsive and ZigBeetoMQTT does not start at all giving the above message.

DennisJohnsen, 2021-11-29:
I first had an error on my Raspbee II saying that there was nothing on the serial port, I simply ran `npm start` again.
Then I got the error mention in this issue, I simply ran `npm start` again.

 **Now this 3rd time running** `npm start` **it started without any issues and I could see a payload from zigbee2mqtt**.
 
 Just posting this here for future me, or someone else in the same position as me.

johngo7470, 2022-03-26:
I found another thread that talked about running 'npm ci' to refresh the install, followed by 'npm run build' (obviously, these need to be run in the ./zigbee2mqtt directory). 

Happy to report that my zigbee sensors are back in business (no more herdsman error).

Raizelian, 2022-06-23:
> I found another thread that talked about running 'npm ci' to refresh the install, followed by 'npm run build' (obviously, these need to be run in the ./zigbee2mqtt directory).
> 
> I had a second Raspi that was still in a failed (herdsman error) state - happy to report that after I ran npm ci and npm run build, my zigbee sensors are back in business (no more herdsman error).

Running "npm ci" in the zigbee2mqtt directory worked for me on RPi4. Thanks for sharing!

outlying, 2022-08-15:
@Raizelian are you running on Home Assistant Operating System? I got into the container but when I try to run `npm ci` I end up with a bunch of build errors

Raizelian, 2022-09-13:
> @Raizelian are you running on Home Assistant Operating System? I got into the container but when I try to run `npm ci` I end up with a bunch of build errors

I'm running on Home Assistant Core, so I'm not sure how much help I can be. Maybe you could share the build errors and someone can shed light on that?

pakosel, 2022-09-27:
Had exactly the same issue and `npm ci` didn't work for me. I had to re-flash the USB dongle to make my `zigbee2mqtt` service start again

wizardofozzie, 2023-07-25:
> > Reflashing didn't help, and it was already connected to USB3. I think the adapter might be toast...
> 
> @deiger Did you try the /dev/serial/by-id path yet?

How do we do this? Mine only shows up under /dev/serial/by-path

deiger, 2023-07-25:
> > > Reflashing didn't help, and it was already connected to USB3. I think the adapter might be toast...
> > 
> > 
> > @deiger Did you try the /dev/serial/by-id path yet?
> 
> How do we do this? Mine only shows up under /dev/serial/by-path

Every device exists under both `/dev/serial/by-id` and `/dev/serial/by-path`, as symlinks to `/dev/ttyUSBX`. `by-path` depends on the USB chaining tree, so it's far more fragile.

wizardofozzie, 2023-07-25:
> Every device exists under both `/dev/serial/by-id` and `/dev/serial/by-path`, as symlinks to `/dev/ttyUSBX`. `by-path` depends on the USB chaining tree, so it's far more fragile.

The tutorial I was following used by-id, which was my intention, but I can't get it to show the by-id path. How can I do this? I'm running a raspberry pi 4 4Gb and I've just flashed the Sonoff Dongle-P with latest firmware