Koenkk/zigbee2mqtt #5855: Why do groups not give access to color and brightness?

kjlisby, 2021-01-21:
When I group a bunch of bulbs, the group always seems to become only a simple on/off switch.

I would like the group to behave as the most capable member of the group, or more precisely as the combination of all abilities any member of the group might expose, such as RGB, dimming etc.

m0wlheld, 2021-01-22:
Using Zigbee(2MQTT) groups as single light entity in Home Assistant, at least brightness is supported. Cannot say anything to color or temperature, as I don't have this type of bulbs. Just dimmable, soft-white.

kjlisby, 2021-01-22:
Hmmm.....  That is strange. I have grouped a number of IKEA LED1837R5, which are dimmable GU10 bulbs. 
I am not using Home Assistant. I am using Domoticz. And there the group shows up as a simple switch only.
So I should probably complain to the Domoticz plugin project instead?
How do I see in the Zigbe2MQTT frontend web page which abilities the group exposes? There is a really good way to not only see but also exercise the abilities of the individual bulbs. But I cannot find the same for groups.

kjlisby, 2021-01-22:
I now see some interesting information in the README of the Domoticz plugin. I guess that solves my case, once I start doing as I am told:
Plugin supports zigbee groups. It is impossible to automatically identify what device should represent the group (OnOff switch, dimmer, etc.), so plugin relies on suffixes in zigbee group name.

kjlisby, 2021-01-22:
Thanks for the input @m0wlheld

closed 2021-01-22 by kjlisby