Koenkk/zigbee2mqtt #5867: Having issues following the how to support new devices page.

WCP12345, 2021-01-23:
Just want to start out by saying I am very new to Zigbee2hqtt and also HA in general.

the device I am trying to make supported is: (there is a similar device in the list here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/cf864be3828ad542cbfcbee2c73d4687e4e43cad/devices.js)

    {
        zigbeeModel: ['ZBT-OnOffPlug-D0011', 'ZBT-OnOffPlug-D0010'],
        model: 'ZS190000138',
        vendor: 'Linkind',
        description: 'Control outlet',
        extend: preset.switch(),
    },

I have pared the unsupported device, and I have added the "advanced: log_level: debug" in the configuration on zigbee2mqtt and restarted, when I go to the log section and trigger the device nothing happens (i click refresh), so I assume I have to follow the "device requires additional configuration" steps, this is where I get very confused! what am I suppose to do with the information on (https://github.com/Koenkk/zigbee-herdsman-converters/blob/cf864be3828ad542cbfcbee2c73d4687e4e43cad/devices.js) 

On step 2. Adding your device (https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html)
It's telling me to add an entry of your device to _node_modules/zigbee-herdsman-converters/devices.js_ **this is nowhere on my HA, and if I do have to add the files from** (https://github.com/Koenkk/zigbee-herdsman-converters/tree/cf864be3828ad542cbfcbee2c73d4687e4e43cad) **where do I add it?**

I have searched high and low for the files it's telling me to add to but I can't find anything on my HA, any guidance to push me in the right direction would be great! 

Thanks,
Will

pvermeij, 2021-01-23:
Did you chech the bottom of the page on [https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html)
and [https://github.com/danielwelch/hassio-zigbee2mqtt#adding-support-for-new-devices](https://github.com/danielwelch/hassio-zigbee2mqtt#adding-support-for-new-devices)

WCP12345, 2021-01-23:
@pvermeij, I didn't see this at the bottom of the page but when I look into it, it opens more confusion! I have Zigbee2hqtt addon installed on my HA (i have 2 devices connected to it successfully).

I'm now following the link (https://github.com/danielwelch/hassio-zigbee2mqtt#adding-support-for-new-devices), I have added zigbee_shepherd_devices: true in the configuration section on zigbee2hqtt and restarted. (i have no idea what this does) I am on version 1.17.0.1 and clicked through to (https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.17.0) now it's telling me to "3. Click on the commit hash (ex. f8066e8) and then browse files button" I have no idea where this is? i am really confused at this point and feel like im going completely off cause from the original instructions (https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html)

I believe my issue is with some of the terminologies, it seems like these guides are here for people that already know what to do, basically, a programmer than is emptying his brain and completely understands the process, then when us newbies come along we can't understand it, as we don't know the terminologies or the process. 

Basically what I'm saying is these guides aren't guides for newbies, or I'm just completely retarded. I think its the latter of the two.

Sorry for the rant! if you still believe you can push me in the right direction any instructions would be great.

BiggsSkeedR, 2021-01-24:
It's not just you, I work in IT, admittedly on hardware and network rather than software development, but even I'm struggling to get anywhere with adding a new device. My Hive heating receiver is coming up as an SLR1 rather than an SLR1b, but I've yet to get the Zigbee2MQTT add-on to recognise my custom device file.

@danielwelch can you help clarify the procedure on Home Assistant please?

batamire, 2021-02-19:
I am a webdev with 18+ years of experience trying to add an unsupported [Philips Hue Phoenix Pendant Light](https://www.philips-hue.com/en-gb/p/hue-white-ambiance-phoenix-suspension-light/3115231PH) and am completely LOST 🤯

I also have HA running on PI3 and have no idea where to find `devices.js`. When I edit over Samba `/share/zigbee2mqtt/configuration.yaml` (which is in JSON btw) it gets overwritten on each zigbee2mqtt restart.

Is `zigbee_shepherd_devices` still relevant? Is `external_converters` the way to go? Placing new device in `/share/zigbee2mqtt/data/new-device.js`? How do I import `presets` from `devices.js`? Why is zigbee's `configuration.yaml` overwritten on each addon restart?

As far as I understand it should be as easy as adding these lines to `devices.js`:

```
{
    zigbeeModel: ["LLM011"],
    model: "3115231PH",
    vendor: "Philips",
    description: "Phoenix pendant light",
    meta: { turnsOffAtBrightness1: true },
    extend: preset.hue.light_onoff_brightness_colortemp(),
    ota: ota.zigbeeOTA,
}
```

@danielwelch could you please clarify the procedure, simple, for dummies?

HA ecosystem is fantastic but is seriously lacking good documentation.
Thank you for all the work and time you've devoted to us 😉

batamire, 2021-02-19:
Just taking time to formulate and say the question out loud gives you clues hehe.

Dummy steps:

1. be sure to edit `configuration.yaml` only in `Supervisor -> Zigbee2mqtt -> Configuration` and set `log_level: debug`
2. set `external_converters` to include `new-device.js`
3. place `new-device.js` in `/share/zigbe2mqtt/new-device.js`, not in `data/` as docs say
4. start with https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/externalConvertersExample/light.js (broken link in docs)
5. watch your logs in `Supervisor -> Zigbee2mqtt -> Logs`
6. good luck, you will need it 😜

@danielwelch how can I test this with `devices.js` so I can contribute to this project?

I am getting `Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x0017880102327a21!` any tips?

closed 2021-03-30 by github-actions[bot]