Koenkk/zigbee2mqtt #5868: Mirgration/pairing issue Xiaomi Sensors

nielssteyaert, 2021-01-23:
I recently updated my zigbee adapter from the CC2531 to the slaesh’s CC2652RB.
Migrating al my devices (Ikea, Philips Hue, Sonoff, Xiaomi MiJia wireless switch, ect..) to the new adapter went well, except for the Xiaomi Sensors:
- Xiaomi WSDCGQ01LM MiJia temperature & humidity sensor
- Xiaomi WSDCGQ11LM Aqara temperature, humidity and pressure sensor

I own a fiew of them and none of the sensors can be paired again.
There is even nothing to be seen on the debug panel.

The things I tried:
- Removing the database.db
- Removing all old devices from devices.yaml
- Rebooting HA/powering off-on
- USB-extension cord
- Removing battery from sensor before reset

I'm getting realy desperate atm.
Is there something I'm missing?

Thanks is advance!

## Debug info
Zigbee2MQTT version: 1.17.0.1
Adapter hardware: CC2652RB
Adapter firmware version: zStack3x0

Fridolin34, 2021-01-23:
I had a similar issue in the past and the following worked for me:

- shutdown the system
- remove the stick from the device
- wait like one minute
- restore everything and reboot the system
- try to pair the sensor

I was about to throw away my sensor, because it was the only one not working anymore, but this workflow somehow solved the issue and since today it works fine.

nielssteyaert, 2021-01-23:
Thanks Fridolin34,

I did what you sugested, but:
I had to replace the battery... It seems that there was just not enough power left to successfully interviewing with the coordinator.

Issues solved.

closed 2021-01-23 by nielssteyaert